linrunner/TLP #711: Enable battery threshold only

loshz, 2023-09-27:
**Is your feature request related to a problem? Please describe.**

Sorry if this isn't the correct forum to ask this question. I've searched the docs and man pages, and didn't seem to get a concrete answer.

Is is possible to configure TLP to _only_ set battery thresholds?

My config only consists of the following 2 lines:
```
START_CHARGE_THRESH_BAT0=75
STOP_CHARGE_THRESH_BAT0=80
```

Does this mean everything else is "off/disabled" by default?

**Describe the solution you'd like**

A clear and concise description of:

* I'd like TLP to only handle battery charge thresholds.

**Describe alternatives you've considered**

* Right now I have my own hacked together bash script that does the job.

**Additional context**

N/A

linrunner, 2023-09-28:
Maybe this can be done by deactivating all parameters that have an intrinsic default via `PARAM=""`. However, this does not represent a supported use case and I do not want to provide any further assistance for it. Thanks for your understanding and good luck!

loshz, 2023-09-28:
No worries! I will give the param thing a try and see how that works out. I fully understand that this isn't a supported use case - thanks again!