linrunner/TLP #713: Nvidia Optimus with proprietary driver won't power down on AC

heinthanth, 2023-11-01:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I have Nvidia Optimus setup with proprietary driver. It works well when TLP is stopped or is on BAT mode.
I've read TLP FAQ and enable SOUND_POWER_SAVE_ON_AC=1. But Nvidia is not power down when laptop is connected to AC.

**Expected behavior**

Nvidia dGPU should be power down even on AC.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Only happens on AC
2. Setup Nvidia Optimus with PRIME
3. Plug AC adapter
4. https://gist.github.com/heinthanth/ff443ef23952dc559055fbfac2deb32c

heinthanth, 2023-11-02:
[image]

I found `RUNTIME_PM_ON_AC` is disabled somehow. Now, it works! Thank you!

closed 2023-11-02 by heinthanth

linrunner, 2023-11-02:
Thanks for your feedback. Glad you were able to figure it out for yourself. I couldn't find `RUNTIME_PM_ON_AC=auto` in your output, so the default `on` must have been effective.