linrunner/TLP #714: tlp-stat -b shows wrong (unavailable) battery, thresholds not working

H-Sachse, 2023-11-04:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

tlp-stat shows values for unavailable battery BAT1
configured thresholds are not applied

Environment:
 - Lenovo Edge 11 (type 2545-A12)
 - Debian 12 (Bookworm) with kernel 6.1.0-13-amd64
 - tlp version 1.5.0-2
 - tp-smapi version 0.43 or 0.44

It used to work correctly with Debian 11 (Bullseye)

**Expected behavior**

correct battery is shown, thresholds are applied correctly

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
  - both on battery and AC 
2. Actions to reproduce the behaviour
 - use tlp-stat -b on described environment
3. Shell commands entered and their output
 - 'tlp-stat -b'
 - 'grep '
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
[on battery](https://gist.github.com/H-Sachse/51c6f4cbf259f483c1eca98ae46d49eb)
[on AC](https://gist.github.com/H-Sachse/f1d84961023511fb36eae78ebf8f7106)
**Additional context**
[dump of BAT0 parameters](https://gist.github.com/H-Sachse/ae1425004a47c2d5bceb2ec0fab0116f)
[dump of BAT1 parameters](https://gist.github.com/H-Sachse/0bed54f5bfe8b6cd81a5cbce032cbef9)

Add any other context about the problem here.

linrunner, 2023-11-05:
Hi, looks interesting. Do you remember the TLP version you used with Bullseye:

- the ancient 1.3.1 from the regular repository - or
- 1.5.0 from backports?

I also need the output of

    grep . /sys/class/power_supply/BAT?/*

via Gist.

H-Sachse, 2023-11-05:
Hi,

I'm pretty sure I used the regular version, not the backports one.
[Here 's the dump of /sys/class/power_supply/BAT?/*](https://gist.github.com/H-Sachse/0ac8758b0ad5043bded08de1f8df6f6e)
It doesn't contain any BAT0 entries, but the data matches smapi BAT0.

and here's the content of ´/sys/class/power_supply`
```
ACAD -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/ACAD
BAT1 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT1
```

linrunner, 2023-11-05:
Seems to be a kernel regression. Until now, the battery names for ACPI and tp_smapi always matched, i.e. /sys/class/power_supply/BAT0 == /sys/devices/platform/smapi/BAT0. However, the Debian kernel 6.1 now turns this into /sys/class/power_supply/**BAT1** != /sys/devices/platform/smapi/BAT0.

My X200 doesn't show the problem, but the early Edge series in particular is known for its "special curls".

I'll have to take a closer look, may take a while.

H-Sachse, 2023-11-05:
Thanks for looking into it. It's not time critical for me since the Edge 11 is only used for testing things that might break my main laptop (x230).

linrunner, 2023-11-05:
All right, my X200 is also only warmed up for testing.

linrunner, 2023-11-07:
It was easier than I first thought. So here we go:

1. Please grab packages from [here](https://download.linrunner.de/packages/), remember to keep your configuration when installing them with dpkg.
2. Do some testing and show the output:
```
sudo tlp-stat -s -b
sudo tlp fullcharge
sudo tlp setcharge
sudo tlp discharge # interrupt with Ctrl+C once discharging commences
```

H-Sachse, 2023-11-11:
Thanks for the quick fix.

Here's the output of `tlp-stat -s -b`:
```
--- TLP 1.7.0-alpha.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad Edge 2545A12
BIOS           = 87ET44WW (1.18 )
EC Firmware    = 1.10
OS Release     = Debian GNU/Linux 12 (bookworm)
Kernel         = 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64
/proc/cmdline  = BOOT_IMAGE=/ROOT/debian-buster@/boot/vmlinuz-6.1.0-13-amd64 root=ZFS=rpool/ROOT/debian-buster ro boot=zfs quiet splash radeon.runpm=0
Init system    = systemd v252 (252.17-1~deb12u1)
Boot mode      = BIOS (CSM, Legacy)
Suspend mode   = s2idle [deep]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = unknown
Mode           = unknown
Power source   = AC

+++ Battery Care
Plugin: thinkpad-legacy
Supported features: charge thresholds, recalibration
Driver usage:
* tp-smapi (tp_smapi) = active (status, charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  2..96(default)
* STOP_CHARGE_THRESH_BAT0/1:   6..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/devices/platform/smapi/BAT0/manufacturer               = LGC
/sys/devices/platform/smapi/BAT0/model                      = 42T4897
/sys/devices/platform/smapi/BAT0/manufacture_date           = 2010-09-12
/sys/devices/platform/smapi/BAT0/first_use_date             = 2010-12-30
/sys/devices/platform/smapi/BAT0/cycle_count                =    319
/sys/devices/platform/smapi/BAT0/temperature                =     24 [°C]
/sys/devices/platform/smapi/BAT0/design_capacity            =  56160 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =  43420 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =  34140 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =     79 [%]
/sys/devices/platform/smapi/BAT0/remaining_running_time_now =      0 [min]
/sys/devices/platform/smapi/BAT0/remaining_charging_time    =    102 [min]
/sys/devices/platform/smapi/BAT0/power_now                  =  39905 [mW]
/sys/devices/platform/smapi/BAT0/power_avg                  =  38720 [mW]
/sys/devices/platform/smapi/BAT0/state                      = charging

/sys/devices/platform/smapi/BAT0/start_charge_thresh        =     96 [%]
/sys/devices/platform/smapi/BAT0/stop_charge_thresh         =     95 [%]
/sys/devices/platform/smapi/BAT0/force_discharge            =      0

Charge                                                      =   78.6 [%]
Capacity                                                    =   77.3 [%]
```
`tlp fullcharge `
```
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96 (no change)
Charging starts now, keep AC connected.
```
`tlp setcharge`
```
Setting temporary charge thresholds for BAT0:
  start =  75
  stop  =  80
```
`tlp discharge`
```
Currently discharging battery BAT0:
voltage            =  11621 [mV]
remaining capacity =  35990 [mWh]
remaining percent  =     83 [%]
remaining time     =      0 [min]
power              = -11946 [mW]
state              = discharging
force discharge    = 1
Press Ctrl+C to cancel.
```
`tlp-stat -s -b`
```
--- TLP 1.7.0-alpha.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad Edge 2545A12
BIOS           = 87ET44WW (1.18 )
EC Firmware    = 1.10
OS Release     = Debian GNU/Linux 12 (bookworm)
Kernel         = 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64
/proc/cmdline  = BOOT_IMAGE=/ROOT/debian-buster@/boot/vmlinuz-6.1.0-13-amd64 root=ZFS=rpool/ROOT/debian-buster ro boot=zfs quiet splash radeon.runpm=0
Init system    = systemd v252 (252.17-1~deb12u1)
Boot mode      = BIOS (CSM, Legacy)
Suspend mode   = s2idle [deep]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 07:20:21, 139 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: thinkpad-legacy
Supported features: charge thresholds, recalibration
Driver usage:
* tp-smapi (tp_smapi) = active (status, charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  2..96(default)
* STOP_CHARGE_THRESH_BAT0/1:   6..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/devices/platform/smapi/BAT0/manufacturer               = LGC
/sys/devices/platform/smapi/BAT0/model                      = 42T4897
/sys/devices/platform/smapi/BAT0/manufacture_date           = 2010-09-12
/sys/devices/platform/smapi/BAT0/first_use_date             = 2010-12-30
/sys/devices/platform/smapi/BAT0/cycle_count                =    319
/sys/devices/platform/smapi/BAT0/temperature                =     26 [°C]
/sys/devices/platform/smapi/BAT0/design_capacity            =  56160 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =  43420 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =  35970 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =     83 [%]
/sys/devices/platform/smapi/BAT0/remaining_running_time_now =      0 [min]
/sys/devices/platform/smapi/BAT0/remaining_charging_time    = not_charging [min]
/sys/devices/platform/smapi/BAT0/power_now                  =      0 [mW]
/sys/devices/platform/smapi/BAT0/power_avg                  =  -9343 [mW]
/sys/devices/platform/smapi/BAT0/state                      = idle

/sys/devices/platform/smapi/BAT0/start_charge_thresh        =     81 [%]
/sys/devices/platform/smapi/BAT0/stop_charge_thresh         =     80 [%]
/sys/devices/platform/smapi/BAT0/force_discharge            =      0

Charge                                                      =   82.8 [%]
Capacity                                                    =   77.3 [%]
```

linrunner, 2023-11-11:
Great. Output and function are 100% as expected, including the threshold readout quirk that is so typical of the Edge/E series.

Thanks for testing!

linrunner, 2023-11-11:
Btw: Do you happen to use Coreboot on your X230? I have something else that needs to be tested: https://github.com/linrunner/TLP/issues/657

H-Sachse, 2023-11-11:
I use Vanilla BIOS with grub-pc, so I can't help you with that, sorry.

linrunner, 2023-11-21:
I see.

closed 2023-11-21 by linrunner

linrunner, 2024-09-05:
Hi @H-Sachse : TLP 1.7 Beta 1 is out with the corresponding fix. Please verify -> https://github.com/linrunner/TLP/issues/760

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1: