linrunner/TLP #717: [Lenovo non-ThinkPad series] Battery conservation mode threshold is different on some machines

Root-Core, 2023-11-21:
TLP references the Battery Conservation Mode to have a fixed threshold of 60%.
Newer laptops - or some - have 80% as their target. At least my does [and some others](https://gist.github.com/linrunner/4a6876648765fac5e141f15d0582a945?permalink_comment_id=4718582#gistcomment-4718582).

It seems there is no way to read out the actual target.
Therefore TLP should avoid mentioning the 60% - or any - threshold, as some hardware configurations have different thresholds.

I had the idea to map the battery or laptop models to a threshold, but that can get complex quiet quickly.
Maybe there is another hint that could be used like the capacity etc. but this might be out of scope for TLP.

CC https://github.com/johnfanv2/LenovoLegionLinux/issues/136

linrunner, 2023-11-21:
I also think it's a pity that there is no way to read out the actual threshold. I'll add a note on the website first and remove the percentage from the `tlp-stat` output with version 1.7.

linrunner, 2023-11-23:
Done -> https://linrunner.de/tlp/settings/bc-vendors.html#lenovo-non-thinkpad-series

closed 2023-11-23 by linrunner

Root-Core, 2023-11-24:
@linrunner You might add to the comment in the docs, that the BCM does not discharge the battery to the fixed threshold automatically. You have to unplug until it's at/under the threshold.

linrunner, 2023-11-25:
@linrunner it's already in the FAQ:

https://linrunner.de/tlp/faq/battery.html#what-is-battery-care
> **What charge thresholds do not do**
> -  In particular, with AC connected, a battery with a charge level higher than the stop charge threshold will not be discharged to the stop charge threshold, nor will there be a (cyclic) discharge down to the start charge threshold

As well as https://linrunner.de/tlp/faq/battery.html#why-does-the-battery-not-begin-to-discharge-when-the-stop-threshold-is-reached

Root-Core, 2023-11-25:
@linrunner My bad, I confused the TLP documentation with some other. I should have checked again, sorry for the inconvenience.
Thank you for the quick adoption and response.

linrunner, 2024-09-05:
Hi @Root-Core  : TLP 1.7 Beta 1 is out with the corresponding change to the output. Please verify -> https://github.com/linrunner/TLP/issues/760

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1: