linrunner/TLP #719: DEVICES_TO_DISABLE_ON_STARTUP doesn't work for wifi

vrgovinda, 2023-11-30:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I have enabled the  DEVICES_TO_DISABLE_ON_STARTUP="bluetooth nfc wifi wwan" line in /etc/tlp.conf

The Bluetooth gets disabled but the WiFi doesn't.

**Expected behavior**

Both should get disabled at startup. 

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
battery

2. Actions to reproduce the behaviour

- open sudo nano /etc/tlp.conf
- uncomment  DEVICES_TO_DISABLE_ON_STARTUP="bluetooth nfc wifi wwan"
- save and exit.

4. Shell commands entered and their output
5. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0029: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
/etc/tlp.conf L0152: SCHED_POWERSAVE_ON_AC="1"
/etc/tlp.conf L0153: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
[... 188 lines cut ...]
Plugin: generic
Supported features: none available

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = DELL MGCM52A
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   3645 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   3645 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   2238 [mAh]
/sys/class/power_supply/BAT0/current_now                    =    806 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold = (not available) 
/sys/class/power_supply/BAT0/charge_control_end_threshold   = (not available) 

Charge                                                      =   61.4 [%]
Capacity                                                    =  100.0 [%]

+++ Recommendations
* Install smartmontools for disk drive health info

**Additional context**

Add any other context about the problem here.

linrunner, 2023-12-01:
Does

    wifi off

work?

vrgovinda, 2023-12-03:
> Does
> 
> ```
> wifi off
> ```
> 
> work?

Do you mean ` DEVICES_TO_DISABLE_ON_STARTUP="bluetooth nfc wifi off wwan"`

linrunner, 2023-12-03:
Also please show the output of another terminal command:

    sudo service tlp status

linrunner, 2023-12-18:
Due to inactivity I'll close here. Nevertheless, you are welcome to post new findings about your problem here.

closed 2023-12-18 by linrunner