linrunner/TLP #720: Idea: Support power-profiles-daemon API

ryanabx, 2023-12-15:
**Is your feature request related to a problem? Please describe.**

The problem I face is wanting to have the power savings of tlp with the nice interface that DEs provide with power-profiles-daemon.

**Describe the solution you'd like**

An API could be implemented that takes calls to change the power profile in ppd and changes to a relevant profile in tlp instead.
Since ppd is already recommended to be uninstalled, this would be non-intrusive to a user using tlp, and would allow tlp to hook into the UI components created by GNOME and KDE to control power profiles.

**Describe alternatives you've considered**

There is a project called TuneD which provides custom configuration profiles similar to tlp, and is implementing the power-profiles-daemon API due to a proposal to replace it with ppd in Fedora, see: 
https://discussion.fedoraproject.org/t/f40-change-proposal-tuned-replaces-power-profiles-daemon-self-contained/94995
Also see relevant PR: https://github.com/redhat-performance/tuned/pull/579

linrunner, 2023-12-15:
Hi, thanks for your suggestion. 

However, there is a major obstacle: TLP is _not_ implemented as a _daemon_ [1], but in my opinion such a daemon would be required for a DBUS API. 

[1] https://linrunner.de/tlp/developers/architecture.html

ryanabx, 2023-12-17:
> Hi, thanks for your suggestion.
> 
> However, there is a major obstacle: TLP is _not_ implemented as a _daemon_ [1], but in my opinion such a daemon would be required for a DBUS API.
> 
> [1] https://linrunner.de/tlp/developers/architecture.html

Thank you for your response! I think you're right that it's probable that a daemon would be required, since that's how ppd is currently implemented. As such it's probably out of scope for this project. I appreciate your work with tlp by the way!

tbodt, 2025-01-11:
Indeed, a daemon of some kind is required to implement the API. But it would make sense to implement as a separate, mostly independent, systemd service, to be enabled only if needed.

tbodt, 2025-01-12:
@linrunner Are you open to adding a python script to the project? If not...well, bash is one of the few languages you can't write a dbus service in, but I can rewrite it in pretty much any other language.

tbodt, 2025-01-14:
Re tracking all system state: Is it not enough to watch for changes to /run/tlp/last_pwr?

Re root privilege: Yeah, seems like it has to become a security boundary. But it's not different from the way power-profiles-daemon itself operates, and if they have it working TLP can too.

Re three power profiles, the PPD APi is able to dynamically report that there are only two profiles. There is also the option of repurposing "balanced" as automatic selection.

linrunner, 2025-01-24:
I'm currently quite busy with the 1.8 beta test. I'll get back to you as soon as I have time for a proper answer.

linrunner, 2025-03-01:
@tbodt sorry for the long delay. 

I would say start with your suggestion. Too many users want it. Here's another one: https://github.com/linrunner/TLP/issues/787

However, I would like to start it separately from the main project in its own repository, working title **TLP-PD**. I hope that's OK for you.

I assume that we need three profiles. I'll think about that separately and start a user survey to design it.

tbodt, 2025-03-01:
Separate repo does make sense. In the end it likely should become a separate package in distros anyway.

Are you planning to do it, or would you like me to try? I have a bunch of half finished python code which I haven't worked on since my first comment here.

linrunner, 2025-03-01:
I don't speak Python yet, but I've been wanting to start learning it for years. So you have to start ...

tbodt, 2025-03-02:
I went and pushed the code I've been writing in the TLP repo so far to a branch over here: https://github.com/tbodt/TLP/commit/fd67ee6d2575fef9ec4de6963558f5c025138337. It's more of a stub, it doesn't talk to TLP yet, but KDE's widget does connect to it at least. TODO move it out to its own repo

linrunner, 2025-03-23:
@tbodt Is it intentional that **tlp-ppd.policy** is installed twice under its original name and modified by sed each time? Shouldn't this happen in the loop as **$(BUS_NAME).policy**? 

https://github.com/tbodt/TLP/blob/ppd/Makefile#L215

tbodt, 2025-03-23:
Yes, thanks for pointing that out. But I should also check if it actually has to be installed twice. power-profilers-daemon just installs once.

linrunner, 2025-03-23:
Uninstall is of .policy, .conf, .service is missing as well.

tbodt, 2025-03-23:
Updated. It looks like it's fine to just install one polkit file, that's not an external interface and is just a way for tlp-ppd to ask polkit whether the request comes from an interactive user.

tbodt, 2025-03-24:
Relating to the number of profiles, the protocol is in theory highly flexible and would support any set of profiles with any name, but unfortunately KDE Powerdevil (at least) behaves somewhat strangely when the returned set of profiles is different from the 3 that power-profiles-daemon returns.

linrunner, 2025-03-24:
@tbodt I would prefer that we call the daemon **tlp-pd** (“TLP Profiles Daemon”) instead of **tlp-ppd**. I'd like to avoid having the competition in the name. Please also rename the repo to **TLP-PD**, otherwise I can't fork it and send you changes ... ;-)

linrunner, 2025-03-24:
> Relating to the number of profiles, the protocol is in theory highly flexible and would support any set of profiles with any name, but unfortunately KDE Powerdevil (at least) behaves somewhat strangely when the returned set of profiles is different from the 3 that power-profiles-daemon returns.

I think the users would insist on three profiles anyway, simply because they are used to it. We might as well call them “power-saver”, “balanced” and “performance” as ppd does.

My (tricky) task is to figure out how to map this to AC/BAT plus “something” in TLP.

tbodt, 2025-03-24:
> My (tricky) task is to figure out how to map this to AC/BAT plus “something” in TLP.

When I thought about this last time I imagined that you could change 0/1 function arguments to be AC/BAT, expand those directly into the variable name, and now it's possible to make as many profiles as you want by changing the last word of the settings variable. But talk is cheap.

tbodt, 2025-03-24:
> [@tbodt](https://github.com/tbodt?rgh-link-date=2025-03-24T14%3A33%3A20.000Z) I would prefer that we call the daemon **tlp-pd** (“TLP Profiles Daemon”) instead of **tlp-ppd**. I'd like to avoid having the competition in the name. Please also rename your repo to **TLP-PD**, otherwise I can't fork it and send you changes ... ;-)

I've renamed the repo and also added you as a committer.

linrunner, 2025-03-24:
@tbodt GitHub doesn't let me fork your repository, perhaps because it is forked from TLP.

I think it can't be avoided that you set up only your contribution, i.e. what is necessary for TLP-PD as a *new* repository without the whole history of TLP. Delete the TLP-PD repository on GitHub beforehand.

So in the first approach LICENSE (GPL2), COPYING (you as author) and Makefile plus your stuff. 

You can leave the Makefile as it is for now, and I'll strip it down.

tbodt, 2025-03-24:
I'll try that this evening.

tbodt, 2025-03-27:
Just got around to it :)

linrunner, 2025-03-30:
Well. I just sat down trying to understand the **Makefile** and the _BUS_NAME_ replacements there and I don't get it (yet):

1. There is a loop for *BUS_NAME* over *org.freedesktop.UPower.PowerProfiles net.hadess.PowerProfiles* creating two renamed copies each of **tlp-pd.dbus.conf** and **tlp-pd.dbus.service** and also doing the replacements for *@BUS_NAME@*.
2. There is an instance of *@BUS_NAME@* in **tlp-pd.service.in** which is replaced by the _$(INFILES)_ target.
3. **But** **tlp-pd** contains three instances of *@BUS_NAME@* that are replaced nowhere - is this intended?
4. Do we really need **both** interfaces _org.freedesktop.UPower.PowerProfiles_ **and** _net.hadess.PowerProfiles_?

tbodt, 2025-03-30:
> 3. **But** **tlp-pd** contains three instances of _@BUS_NAME@_ that are replaced nowhere - is this intended?
It is replaced at the bottom in python code before bus.publish.

> 4. Do we really need **both** interfaces _org.freedesktop.UPower.PowerProfiles_ **and** _net.hadess.PowerProfiles_?
Sadly, yes...because the PowerProfiles project changed ownership and changed the interface name along with that. But some clients still use the old name, so they have to keep both working.

linrunner, 2025-04-18:
@tbodt I have been thinking quite some time about how to map profiles. My proposal is that we implement four profiles:

| Profile | Command | Config Used | Remarks |
| :---- | :---- | :---- | :---- |
| performance | `tlp performance` *) |  **_AC** | Activates manual mode, i.e. deactivates automatic switching, same as tlp ac |
| balanced | `tlp balanced` *) | **_BAT** | Activates manual mode, i.e. deactivates automatic switching, same as tlp bat |
| power-saver | `tlp power-saver` | **_SAV**, **_BAT** **) | Activates manual mode, i.e. deactivates automatic switching |
| auto ***) | `tlp auto` | **_AC**, **_BAT** | Exits manual mode and reactivates automatic switching, same as tlp start  |

*) I deliberately do not employ `tlp ac/bat`so that the behavior  mya be changed later if necessary.

**) In the first step I do not intend to create a full set of _**SAV** parameters, only the small subset ppd does atm -> https://linrunner.de/tlp/faq/ppd.html plus maybe a few more. For the rest, power-saver will use **_BAT**.

***) default on startup.

What do you think?

linrunner, 2025-04-18:
@tbodt smart-ass question (sorry) from someone with no Python experience about the dbus library used: [pydbus](https://github.com/LEW21/pydbus) has not been developed on for 7 years. Of the alternatives listed [here](https://wiki.python.org/moin/DbusExamples), only [python-sdbus](https://github.com/python-sdbus/python-sdbus) seems to be alive.

I'm unable to decipher what [tuned-ppd ](https://github.com/redhat-performance/tuned/blob/master/tuned-ppd.py) uses.

linrunner, 2025-04-18:
Seems like GNOME 48 does not support arbitrary profiles, only performance, balanced, power-saver  ... :-(

tbodt, 2025-04-18:
> Seems like GNOME 48 does not support arbitrary profiles, only performance, balanced, power-saver ... :-(
> 
> So how do to disable manual mode then?

Yes, KDE is the same. :( Not sure of a way. ppd doesn't have an automatic mode, the design is that the DE watches for whether you're on battery and does all the switching manually.

> [@tbodt](https://github.com/tbodt) smart-ass question (sorry) from someone with no Python experience about the D-BUS library used: [...]

Yes, I was disappointed by this too... tuned-ppd uses [dbus-python](https://dbus.freedesktop.org/doc/dbus-python/). Perhaps it is sensible to simply copy what it did as they have it figured out already - I figured using pydbus would be less code to write, and it is, up until you have to interface with `polkit` :P

linrunner, 2025-04-18:
Then we'll put the auto profile topic on hold. May I assume that you agree with the rest of the proposal?

ATM in GNOME I can click on a profile but it is stuck on "Power-Saver". Is this due to the incomplete implementation of the daemon?

tbodt, 2025-04-18:
> Then we'll put the auto profile topic on hold. May I assume that you agree with the rest of the proposal?

Looks good

> ATM in GNOME I can click on a profile but it is stuck on "Power-Saver". Is this due to the incomplete implementation of the daemon?

Yes, the setter function for the profile just prints what profile you asked for and does nothing. It should update what the getter returns, and also send a property change notification.

linrunner, 2025-04-18:
Well, then I consider the TLP-side interface for the profiles to be defined. 

You may continue if you like. I will soon start implementing the new commands in a separate branch.

Mylinde, 2025-10-15:
Testing on Asus Vivobook with Ubuntu Budgie 24.04.3, Power Profiles are under Setttings > Energie present, switching is working.

[image]

[stat AC](https://gist.github.com/Mylinde/a51dc7d025bdffe629fa0302914d570e)

[stat BAT](https://gist.github.com/Mylinde/cca10dbc9cefca796d1ff83bb707d700)

Mylinde, 2025-10-16:
@linrunner 
Are there any differences in handling tlp settings/values to version 1.8? Have taken over the configschema and config for tlpui from version 1.8  to 1.9 to get tlpui working.

linrunner, 2025-10-21:
@Mylinde so your point is: using TLP_DISABLE_DEFAULTS=1 leads to PLATFORM_PROFILE_ON_AC/BAT/SAV being disabled despite explicitly configured. Actually, there is already #786 for that. Thanks for reporting, I'll look into it of course.

linrunner, 2025-10-23:
@Mylinde Oh, my! Please try again.

Mylinde, 2025-10-23:
Working on BAT and AC, defaults disabled.

[stat](https://gist.github.com/Mylinde/6d3add842818459593b4da6b9864345e#file-fixed_pp_dd)

Mylinde, 2025-10-24:
Default values are usually decided by the creator and may not match the user's preferences, or they may correlate with other adjustments made by the user. I like to decide for myself which settings to change, as this makes it easier for me to identify the cause of a malfunction or poor performance without having to search for a long time.