linrunner/TLP #725: TLP doesn't affect charging while off

PenelopeFudd, 2024-02-10:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

It looks like the charging threshold are not observed while the laptop is plugged in but turned off.  I'd set the stop_charge_thresh_bat0 to 70 but when I turned on the laptop in the morning it was at 100%. 

**Expected behavior**

Expected it to be at 70%.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?  On AC
2. Actions to reproduce the behaviour: Set start+stop charge threshold to anything
3. Shell commands entered and their output: n/a
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)  https://gist.github.com/PenelopeFudd/ef88911b58e6a18ee427a419c47275fe.js 

**Additional context**

Add any other context about the problem here.

PenelopeFudd, 2024-02-10:
Note, I've been fiddling with the charge thresholds, so they may not be consistent in the output of `tlp-stat`.  Sorry about that.

linrunner, 2024-02-10:
Your link leads to a page that I cannot decipher. Please correct it.

PenelopeFudd, 2024-02-10:
Fixed: https://gist.githubusercontent.com/PenelopeFudd/ef88911b58e6a18ee427a419c47275fe/raw

linrunner, 2024-02-11:
The output proves that TLP has set the thresholds:

```
/sys/class/power_supply/BAT0/charge_control_start_threshold =     50 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     70 [%]

```
But in your case, the hardware is not enforcing them. It is definitely not a TLP problem. 

Your ThinkPad should have an emergency reset hole on the underside, try that first. Otherwise, I would update the BIOS and EC firmware to the latest version.

https://linrunner.de/tlp/faq/battery.html#various-thinkpad-battery-malfunctions

PenelopeFudd, 2024-02-11:
What's the fastest method of testing this to see if the emergency reset fixed it?

linrunner, 2024-02-17:
@PenelopeFudd You set start and stop thresholds a few percent above the current charge level, make sure that the Thinkpad is starting to charge and shut it down. Then wait long enough.

closed 2024-02-17 by linrunner