linrunner/TLP #727: Conflicting Lenovo conservation mode tresholds

cankurttekin, 2024-02-17:
tlp-stat output:

`+++ Battery Care
Plugin: lenovo
Supported features: charge threshold
Driver usage:
* vendor (ideapad_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 0(off), 1(on) -- battery conservation mode

/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode = 1 (60%)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = Sunwoda
/sys/class/power_supply/BAT0/model_name                     = L22D4PA1
/sys/class/power_supply/BAT0/cycle_count                    =     12
/sys/class/power_supply/BAT0/energy_full_design             =  71000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  69980 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  63890 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   4411 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

Charge                                                      =   91.3 [%]
Capacity                                                    =   98.6 [%]
`

Hello, i dont know if it is the right place to report my issue. before making any changes on my new non-thinkpad lenovo laptop it was set default to 75-80 tresholds. I made some configurations but it did not applied and it charged to 100 so i wanted to enable conservation mode again but now its set to 60 percent. It shows 75-80 on windows Lenovo vantage app, but i cant set it to 80 with tlp and now it ignores everything and charges to 100 whatever i do.

linrunner, 2024-02-19:
Hi, on Lenovo laptops, the threshold is stored in the hardware and will affect the other system in dual boot setups. In your case I recommend to adjust the threshold under Windows only.

To do so, first remove the threshold from /etc/tlp.conf by commenting the setting. The line should look like this:

    #STOP_CHARGE_THRESH_BAT0=1

Then, delete it from the hardware using

    sudo tlp fullcharge

Last, reboot to Windows, remove the threshold there and set it again.

ps. the 60% threshold claimed by TLP is based on an incorrect assumption made during development. In reality, the threshold is set by the Lenovo firmware, and some models, including yours, have a threshold of 80% instead of 60%.

cankurttekin, 2024-02-27:
I deleted Windows and set treshold again with tlp, it resolved.

closed 2024-02-27 by cankurttekin

linrunner, 2024-03-01:
Good solution. Thanks for your feedback.