linrunner/TLP #732: `energy_performance_preference` is not set with `CPU_ENERGY_PERF_POLICY`

0xdeafbeef, 2024-03-16:
**Describe the bug**

```
/sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference
```
is not set with 
```
CPU_ENERGY_PERF_POLICY_ON_BAT=balance_power
```

I can set it manualy with 
```bash
for file in /sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference; do
    echo "balance_power" | sudo tee "$file"
done
```

https://gist.github.com/0xdeafbeef/e4a79dc51d3c7a9d8165cf98e1c36000

 power-profiles-daemon is removed

linrunner, 2024-03-17:
For AMD you need at least TLP version 1.6.

closed 2024-03-17 by linrunner

0xdeafbeef, 2024-03-18:
@linrunner, sorry, do you maintain the Fedora package? If so, could you release the 1.6.0 version?

linrunner, 2024-03-19:
@0xdeafbeef Sorry, I do not maintain any distribution packages.