linrunner/TLP #735: [Possible bug] Thresholds for  Slice 19+ on X220 not honored

samveen, 2024-03-24:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

The `STOP_CHARGE_THRESH_BAT1`  threshold for BAT1 (`Lenovo Slice 19+`) are not honored  when charging, and the charging continues till 100%

**Expected behavior**

The `STOP_CHARGE_THRESH_BAT1`  threshold for BAT1 (`Lenovo Slice 19+`)  should be honored  when charging, and the charging should stop at the configured percentage.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Let the battery discharge below the `STOP_CHARGE_THRESH_BAT1`.
2. Connect  the charger back to the Slice 19+ and start charging.
3.  Let charging complete.

** `tlp-stat -b` while discharging

```
samveen@samveen-x220:~$ sudo tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
* tp-smapi (tp_smapi)     = readonly (status)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/devices/platform/smapi/BAT0/manufacturer               = LGC
/sys/devices/platform/smapi/BAT0/model                      = 45N1029
/sys/devices/platform/smapi/BAT0/manufacture_date           = 2016-03-22
/sys/devices/platform/smapi/BAT0/first_use_date             = 2016-10-04
/sys/devices/platform/smapi/BAT0/cycle_count                =    219
/sys/devices/platform/smapi/BAT0/temperature                =     37 [°C]
/sys/devices/platform/smapi/BAT0/design_capacity            =  86580 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =   4970 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =   4090 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =     82 [%]
/sys/devices/platform/smapi/BAT0/remaining_running_time_now = not_discharging [min]
/sys/devices/platform/smapi/BAT0/remaining_charging_time    = not_charging [min]
/sys/devices/platform/smapi/BAT0/power_now                  =      0 [mW]
/sys/devices/platform/smapi/BAT0/power_avg                  =      0 [mW]
/sys/devices/platform/smapi/BAT0/state                      = idle

/sys/class/power_supply/BAT0/charge_control_start_threshold =     75 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     85 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   82.3 [%]
Capacity                                                    =    5.7 [%]

+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/devices/platform/smapi/BAT1/manufacturer               = SONY
/sys/devices/platform/smapi/BAT1/model                      = 42T4967
/sys/devices/platform/smapi/BAT1/manufacture_date           = 2015-06-12
/sys/devices/platform/smapi/BAT1/first_use_date             = 2020-02-21
/sys/devices/platform/smapi/BAT1/cycle_count                =     19
/sys/devices/platform/smapi/BAT1/temperature                =     37 [°C]
/sys/devices/platform/smapi/BAT1/design_capacity            =  64380 [mWh]
/sys/devices/platform/smapi/BAT1/last_full_capacity         =  61330 [mWh]
/sys/devices/platform/smapi/BAT1/remaining_capacity         =  48340 [mWh]
/sys/devices/platform/smapi/BAT1/remaining_percent          =     79 [%]
[... 168 lines cut ...]
```

** Non-default configuration
```
samveen@samveen-x220:~$ sudo tlp-stat --cdiff
--- TLP 1.6.1 --------------------------------------------

+++ Configured Settings (only differences to defaults):
/etc/tlp.d/03-cpu-restrictions.conf L0007: CPU_ENERGY_PERF_POLICY_ON_AC="balance_power"
/etc/tlp.d/01-wlan-powersave-off.conf L0002: WIFI_PWR_ON_BAT="off"
/etc/tlp.d/02-battery-thresholds.conf L0005: START_CHARGE_THRESH_BAT0="75"
/etc/tlp.d/02-battery-thresholds.conf L0006: STOP_CHARGE_THRESH_BAT0="85"
/etc/tlp.d/02-battery-thresholds.conf L0008: START_CHARGE_THRESH_BAT1="75"
/etc/tlp.d/02-battery-thresholds.conf L0009: STOP_CHARGE_THRESH_BAT1="85"
/etc/tlp.d/03-cpu-restrictions.conf L0001: CPU_DRIVER_OPMODE_ON_AC="active"
/etc/tlp.d/03-cpu-restrictions.conf L0002: CPU_DRIVER_OPMODE_ON_BAT="active"
/etc/tlp.d/03-cpu-restrictions.conf L0004: CPU_SCALING_GOVERNOR_ON_AC="powersave"
/etc/tlp.d/03-cpu-restrictions.conf L0005: CPU_SCALING_GOVERNOR_ON_BAT="powersave"
/etc/tlp.d/03-cpu-restrictions.conf L0010: CPU_MIN_PERF_ON_AC="0"
/etc/tlp.d/03-cpu-restrictions.conf L0011: CPU_MAX_PERF_ON_AC="30"
/etc/tlp.d/03-cpu-restrictions.conf L0012: CPU_MIN_PERF_ON_BAT="0"
/etc/tlp.d/03-cpu-restrictions.conf L0013: CPU_MAX_PERF_ON_BAT="30"
/etc/tlp.d/03-cpu-restrictions.conf L0015: CPU_BOOST_ON_AC="0"
/etc/tlp.d/03-cpu-restrictions.conf L0016: CPU_BOOST_ON_BAT="0"
```

linrunner, 2024-03-24:
Your output clearly shows that the thresholds are set correctly:

> +++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
> [...]
> /sys/class/power_supply/BAT1/charge_control_start_threshold =     75 [%]
> /sys/class/power_supply/BAT1/charge_control_end_threshold   =     85 [%]

So the behavior is due to your ThinkPad, not TLP. I wouldn't expect everything to still work perfectly on hardware that is about 12 years old.

closed 2024-03-24 by linrunner

samveen, 2024-03-24:
@linrunner Would you be able to point me towards where I might check for the underlying issue?

linrunner, 2024-03-24:
If I did that for every charge threshold problem on ThinkPads, the issue tracker would become a support forum and I wouldn't have time to develop anymore.

Just this much: the standard workaround for older models is to reset the EC - remove all batteries, disconnect the power supply, hold the power button for 30 seconds.

samveen, 2024-03-25:
Thank you !!! 

_And apologies. It's very easy for forget that helpful people are taking time out from their priorities to make life easier for us, and to not take more of their time. Thank you for making my life easier for the last 4 thinkpads I've had  :bow: _