linrunner/TLP #739: Lenovo not charging

CcHhDdOo, 2024-04-07:
Hi guys,

I just tried tlp for my Lenovo, but now it's not charging anymore. I guess I have the wrong driver. I could use some help please.

<pre>--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = LENOVO Lenovo U41-70 80JV
BIOS           = BDCN61WW
OS Release     = LMDE 6 (faye)
Kernel         = 6.6.13+bpo-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.13-1~bpo12+1 (2024-02-15) x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-6.6.13+bpo-amd64 root=UUID=38978c56-5b1f-425d-8434-15dab2e9b9d3 ro quiet splash
Init system    = systemd v252 (252.22-1~deb12u1)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 08:12:13,   2295 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: asus
Supported features: none available
Driver usage:
* natacpi (asus_wmi) = inactive (laptop not supported)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = L14M2P21
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  30000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  22870 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  12630 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Not charging

Charge                                                      =   55.2 [%]
Capacity                                                    =   76.2 [%]

</pre>

linrunner, 2024-04-07:
Uninstall TLP. Shut down the laptop. Switch it on and start. Is it now charging the battery?

CcHhDdOo, 2024-04-07:
I did all the mentioned things already.The thing I didn't do was letting drain the battery as I hoped for a battery reset or so. In fact at under 30 % the cpu automatically reduced so I plugged in again and it's charging now. Let's see how far.

CcHhDdOo, 2024-04-07:
Battery stops charging at 60 % like it did before. So the problem still is that I can't go beyond that value (battery charge thresholds not available). Maybe before I switched to Linux, the Windows App Lenovo Vantage had battery protection as active and is now keeping this setting although having Linux now.

linrunner, 2024-04-07:
Since "laptop not supported" is displayed, it is very unlikely that TLP has set the charge threshold. 

Apparently Mint has loaded the wrong vendor specific driver,  `asus_wmi` instead of `ideapad_laptop`. Let's fix it manually. Please show the output of:

    sudo modprobe -rv asus_wmi
    sudo modprobe -v ideapad_laptop
    sudo tlp-stat --cdiff -s -b

CcHhDdOo, 2024-04-07:
I've found the solution: There is that Vantage App also for Linux https://github.com/niizam/vantage

There I can turn off the conservation mode and now it's charging beyond 60 % again, yeah!

Anyway, thank you for the support.

linrunner, 2024-04-07:
I am (on behalf of other Lenovo users) very interested to know if setting the threshold also works with TLP when the correct kernel module is loaded. So I would be grateful if you could show the outputs.

CcHhDdOo, 2024-04-07:
It looks like there is only on (1) or off (0) for the threshold, which is preset to 60% : 

<pre>--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings (only differences to defaults):
/etc/tlp.conf L0038: TLP_WARN_LEVEL=&quot;&quot;
/etc/tlp.conf L0514: RESTORE_THRESHOLDS_ON_BAT=&quot;1&quot;
/etc/tlp.conf L0057: TLP_PS_IGNORE=&quot;BAT&quot;

+++ System Info
System         = LENOVO Lenovo U41-70 80JV
BIOS           = BDCN61WW
OS Release     = LMDE 6 (faye)
Kernel         = 6.6.13+bpo-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.13-1~bpo12+1 (2024-02-15) x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-6.6.13+bpo-amd64 root=UUID=38978c56-5b1f-425d-8434-15dab2e9b9d3 ro quiet splash
Init system    = systemd v252 (252.22-1~deb12u1)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 18:20:04,   2830 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: lenovo
Supported features: charge threshold
Driver usage:
* vendor (ideapad_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 0(off), 1(on) -- battery conservation mode

/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode = 1 (60%)

</pre>

linrunner, 2024-04-07:
The output of the two modprobe commands would have been important, do you still have them?

CcHhDdOo, 2024-04-07:
<pre><span style="color:#8AE234"><b>lenovo@Lenovo</b></span>:<span style="color:#729FCF"><b>~</b></span>$ sudo modprobe -rv asus_wmi
rmmod asus_wmi
rmmod battery
<span style="color:#8AE234"><b>lenovo@Lenovo</b></span>:<span style="color:#729FCF"><b>~</b></span>$ sudo modprobe -v ideapad_laptop
<span style="color:#8AE234"><b>lenovo@Lenovo</b></span>:<span style="color:#729FCF"><b>~</b></span>$ sudo tlp-stat --cdiff -s -b
</pre>

linrunner, 2024-04-07:
So you have to blacklist `asus_wmi`. I will write this in the FAQ. Thanks.

linrunner, 2024-04-08:
I changed the vendor detection sequence. Should work without blacklisting now.

linrunner, 2024-08-26:
@Bryan2333 https://linrunner.de/tlp/settings/bc-vendors.html#lenovo-non-thinkpad-series done.

linrunner, 2024-08-26:
@CcHhDdOo add to the FAQ https://linrunner.de/tlp/faq/battery.html#mis-identification-as-asus-laptop

closed 2024-08-26 by linrunner

linrunner, 2024-09-05:
Hi @Bryan2333 @CcHhDdOo : TLP 1.7 Beta 1 is out with the corresponding fix. Please test -> https://github.com/linrunner/TLP/issues/760

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1: