linrunner/TLP #741: Connecting or disconnecting the charger while laptop is sleeping causes it to no longer be able to wake up (Lenovo Ideapad 5 Pro 14ARH7)

YoriMirus, 2024-04-21:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

When the laptop is sleeping and I plug or unplug the charger, then the laptop has a high chance of no longer being able to wake up.

**Expected behavior**

Laptop should wake up normally when the lid is opened or a keyboard key is pressed.

**To Reproduce**

Steps to reproduce the unexpected behavior:

- Put the laptop into s2idle sleep mode
- Unplug the charger, or plug it in if the laptop isn't charging.
- Laptop will sometimes no longer be able to wake up from sleep mode

1. Does the problem occur on battery or AC or both?
- Both. The change in charging state seems to cause this.
2. Shell commands entered and their output
- No commands need to be entered to cause this bug
3. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

https://gist.github.com/YoriMirus/3f0b9b01eb9b3c0d1dc7fc3488d96e4c

**Additional context**

The mentioned tlp configuration should be the defaults as set by TLP.

This bug does not always happen but occurs pretty often. When this bug occurs, nothing seems to visually change. The laptop starts producing heat, but the power LED still indicates sleep mode. If this bug occurs while disconnecting the charger, the keyboard backlight briefly turns on and then off (for around 100ms).

The issue temporarily goes away by reinstalling tlp, but will come back after the laptop falls asleep a few times.

One strange thing I found though is that setting TLP_ENABLE=0 does not fix this. I need to completely uninstall tlp for this issue to disappear. tlp-stat says that tlp is not running, so I have no idea why tlp would still be causing this even if it's turned off. However, if tlp is disabled (but still installed), when this bug occurs, the fans spin up.

There do not seem to be any errors logged in journalctl when this occurs. The last line in journalctl only mentioned that the laptop successfully fell asleep. Maybe there are errors but they are not logged as the laptop needs to be forcefully shut down in order to be useable again. Kernel also stops listening to the SysRq key, so a safe shutdown seems to be impossible.

linrunner, 2024-04-25:
Usually wake-up problems are caused by the kernel. A deactivated TLP really doesn't touch any settings, I think it's just a coincidence. 

You can try to find the causal setting by means of the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html).

Good luck.

YoriMirus, 2024-04-25:
I do not know how, but after reinstalling TLP again, the issue went away. Closing issue.

closed 2024-04-25 by YoriMirus