linrunner/TLP #742: AMD cpu with amd_pstate=active using amd-pstate-epp driver doesn't switch energy_performance_preference

zyf0330, 2024-04-28:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I use AMD cpu with kernel option `amd_pstate=active` and amd-pstate-epp driver.
The `energy_performance_preference` config doesn't switch respecting tlp.conf.

**Expected behavior**
The `energy_performance_preference` config switch.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? 
both
3. Actions to reproduce the behaviour
always exist

7. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
https://gist.github.com/zyf0330/1b5d014b089b8cfefd0fa8ec913cdddc
**Additional context**

Add any other context about the problem here.

linrunner, 2024-04-28:
EPP for AMD was only added with TLP version 1.6.

Please read the documentation _before_ opening issues:

- https://linrunner.de/tlp/settings/processor.html#cpu-energy-perf-policy-on-ac-bat
- https://github.com/linrunner/TLP/blob/main/changelog#L106

Thanks for your attention.

closed 2024-04-28 by linrunner

zyf0330, 2024-04-28:
Sorry, just tried google search and has no result.

linrunner, 2024-04-30:
You are also welcome to use the modest search function on the TLP website: https://linrunner.de/tlp/search.html?q=amd-pstate&check_keywords=yes&area=default

zyf0330, 2024-04-30:
> You are also welcome to use the modest search function on the TLP website: https://linrunner.de/tlp/search.html?q=amd-pstate&check_keywords=yes&area=default

Thanks, I did that after then.

nlgranger, 2024-06-01:
I'm actually hitting this bug with 1.6.1. energy_performance_preference remains in performance mode.

Kernel boot parameter: `amd_pstate=active`
tlp-stat and tlp.conf: https://gist.github.com/nlgranger/cf375a9b7b74eeecc0bf72fb05aea7e0

Is `/sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference` supposed to be writable? echoing the desired mode (`balance_power`) returns `write: Unknown error 524`.