linrunner/TLP #743: Ubuntu PPA uses weak algorithm (rsa1024) warning, it's not a problem but still needs to be improved.

iHarryPotter178, 2024-06-02:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
In Ubuntu Noble, 24.04 lts, it shows a warning that the TLP PPA uses weak algorithom, though it's not a problem, seeing that everytime is annoying and and it definitly needs to be improved.

**Expected behavior**

It should use stronger encryption so that warning doesn't appear.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Simply add the PPA in Ubuntu noble and do an sudo apt update, the warning should be at the end.

linrunner, 2024-06-03:
Without knowing the actual message, I can't say much about it.  Can you please copy the message from the terminal here?

iHarryPotter178, 2024-06-04:
As theres nothing to do, I'll close this as complete

closed 2024-06-04 by iHarryPotter178

zyf0330, 2024-10-18:
After remove and re-add, this problem is solved.
```
sudo add-apt-repository -r ppa:linrunner/tlp
sudo add-apt-repository ppa:linrunner/tlp
```

linrunner, 2024-10-19:
Launchpad fixed it.