linrunner/TLP #744: DISK_APM_LEVEL_ON_**= floods log irrespective of its value

KAGEYAM4, 2024-06-04:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
Using default config values for DISK_APM_LEVEL_ON_** and using AHCI_RUNTIME_PM_ON_**=auto results in `sda` flooding the logs for Starting,Synchronization,Stopping.
```
Jun 04 13:26:14 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:26:14 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:26:40 ArchLinux kernel: sd 5:0:0:0: [sda] Synchronizing SCSI cache
Jun 04 13:26:40 ArchLinux kernel: sd 5:0:0:0: [sda] Stopping disk
Jun 04 13:26:41 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:26:41 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:26:41 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:27:07 ArchLinux kernel: sd 5:0:0:0: [sda] Synchronizing SCSI cache
Jun 04 13:27:07 ArchLinux kernel: sd 5:0:0:0: [sda] Stopping disk
Jun 04 13:27:13 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:27:13 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:27:13 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:27:39 ArchLinux kernel: sd 5:0:0:0: [sda] Synchronizing SCSI cache
Jun 04 13:27:39 ArchLinux kernel: sd 5:0:0:0: [sda] Stopping disk
Jun 04 13:27:42 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:27:42 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
Jun 04 13:27:42 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
```

**Expected behavior**
No flooding of logs for default values. The log also get flooded when using DISK_APM_LEVEL_ON_BAT="254 254".

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. **Does the problem occur on battery or AC or both?**  Yes it happens in both with default values. I explictitly tested with `tlp bat` & `tlp ac`
2.** Actions to reproduce the behaviour** Using default config values for this expect forcing `BAT` profile on both `AC` and `BAT`
3. Shell commands entered and their output
4. **Full output of `tlp-stat` https://gist.github.com/KAGEYAM4/88d81f1e18e9e0fca2ca64f4190ca415

**Additional context**
FAQs states that -> `Unfortunately drive vendors interpret the APM levels quite spaciously to save power.  Solution: try values > 128. `

But the problem is that when i disable runtime_power_management (AHCI_RUNTIME_PM_ON_**=on),  and check the value of my ssd ( i also restarted my laptop ) -> 
```
[phoenix@ArchLinux ~]$ sudo hdparm -I /dev/sda | grep 'Advanced power management level'
	Advanced power management level: 128
```
It shows the same value that tlp uses, but with this logs do not get flooded. 

But logs get flooded when i eable runtime_power_management  (AHCI_RUNTIME_PM_ON_**=auto) and the value of DISK_APM_LEVEL_ON_BAT="254 254". 

Are there anyother factors that affect this behaviour?

linrunner, 2024-06-10:
I have no idea. Personally, I would do without APM in this case.

There is an older FAQ entry where a Crucial SSD does not like APM: https://linrunner.de/tlp/faq/disks.html#why-is-my-crucial-m4-ssd-so-slow-on-battery

linrunner, 2024-06-28:
I'm closing due to inactivity. If you have any new information, please continue here.

closed 2024-06-28 by linrunner