linrunner/TLP #745: Slowing down the system

Massimo-B, 2024-06-20:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Hi, I installed tlp-1.6.1 on Gentoo, as replacement for older laptop-mode-tools. For that I temporarily disabled the laptop-mode-tools service.

Now doing a `tlp start` makes the laptop slow and not usable anymore. Connected via ssh I can type 1 character per second. None of `tlp ac` or `tlp bat` solves it. Only disconnecting and reconnecting AC power solves it.

Doing a `rc-service tlp` start does not show that issue.

The init script does a `tlp init start`, doing that manually also does not show the issue. But this `init` is not documented anywhere.

Gentoo Linux 23.0
tlp-1.6.1
openrc-0.54 (No systemd)

In the syslog I've seen some

`Jun 20 13:29:12 [kernel] [13146.366090] NOHZ tick-stop error: local softirq work is pending, handler #08!!!`

linrunner, 2024-06-21:
Hi. Generally I expect that people provide actionable information in their issue in order to enable analysis; the issue template should help. 

In your case this amounts to the full output of

    sudo tlp-stat 

1. After boot (I presume that the service is active) 
2. After the command causing the problem: `tlp start`

Please do _not_ simply _copy_ the long output here. Instead, use https://gist.github.com/. Thanks.

Massimo-B, 2024-07-01:
Ok, I compared the tlp-stat before and after running `tlp start`:
```diff
$ diff tlp-stat.before tlp-stat.after 
...
212c212
< Bus 002 Device 009 ID 0bda:8153 control = on,   autosuspend_delay_ms =    0 -- Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter (r8152)
---
> Bus 002 Device 009 ID 0bda:8153 control = auto, autosuspend_delay_ms =    0 -- Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter (r8152)
...
```
Performance loss is only on the SSH connection via network going via a USB-C dock. Locally the system feels fast as expected.

linrunner, 2024-07-05:
Looks as if enabling autosuspend for the USB ethernet adapter causes the slow down of the incoming SSH session. I suggest to exclude the adapter via TLP's configuration:

    USB_DENYLIST="0bda:8153"

Massimo-B, 2024-07-10:
That solved it, thanks.
Is that a bug of autosuspend on the Realtek kernel module?

Massimo-B, 2024-07-10:
Question beside... I wonder what daemon is running to track power changes. When starting tlp via `/etc/init.d/tlp` it does a `tlp init start`. There is no daemon running after that. `man tlp` does not describe the `init` command. How does tlp recognize a switch from ac to bat or do I need to configure something else for that?

linrunner, 2024-07-10:
> Is that a bug of autosuspend on the Realtek kernel module?

Exactly. Realtek adapters often seem to have problems with power saving. On my laptops with USB and PCIe variants.

As for your question of understanding, this is explained in the docs:

- https://linrunner.de/tlp/introduction.html#event-driven-architecture
- https://linrunner.de/tlp/developers/architecture.html

closed 2024-07-10 by linrunner