linrunner/TLP #749: TLP will not start

JayH1119, 2024-07-09:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

tlp.service does not start successfully.

Output from systemctl status:
Jul 09 09:08:57 jLaptop systemd[1]: Starting tlp.service - TLP system startup/shutdown...
Jul 09 09:08:57 jLaptop tlp[1712]: Applying power save settings...done.
Jul 09 09:08:57 jLaptop tlp[1712]: Setting battery charge thresholds.../usr/sbin/tlp: 14: readonly: BATDRV_LG_BCL: is read only
Jul 09 09:08:57 jLaptop systemd[1]: tlp.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jul 09 09:08:57 jLaptop systemd[1]: tlp.service: Failed with result 'exit-code'.
Jul 09 09:08:57 jLaptop systemd[1]: Failed to start tlp.service - TLP system startup/shutdown.

Occurs at startup on A/C or battery. Also occurs when attempting to manually start the service.

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

--- TLP 1.5.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_WARN_LEVEL="3"
defaults.conf L0006: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0008: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0010: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0012: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0013: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0014: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
defaults.conf L0044: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
defaults.conf L0046: USB_EXCLUDE_BTUSB="0"
defaults.conf L0047: USB_EXCLUDE_PHONE="0"
defaults.conf L0048: USB_EXCLUDE_PRINTER="1"
defaults.conf L0049: USB_EXCLUDE_WWAN="0"
defaults.conf L0050: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0051: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0052: RESTORE_THRESHOLDS_ON_BAT="0"
[... 169 lines cut ...]
/sys/bus/pci/devices/0000:00:1f.2/power/control = auto (0x058000, Memory controller, no driver)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x040380, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1f.4/power/control = auto (0x0c0500, SMBus, i801_smbus)
/sys/bus/pci/devices/0000:01:00.0/power/control = auto (0x010802, Non-Volatile memory controller, nvme)
/sys/bus/pci/devices/0000:02:00.0/power/control = auto (0x028000, Network controller, iwlwifi)

+++ USB
Autosuspend       = enabled
Device allowlist  = (not configured)
Device denylist   = (not configured)
Exclude audio     = enabled
Exclude bluetooth = disabled
Exclude phones    = disabled
Exclude printers  = enabled
Exclude WWAN      = disabled

Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 003 ID 8087:0a2a control = auto, autosuspend_delay_ms = 2000 -- Intel Corp. Bluetooth wireless interface (btusb)
Bus 001 Device 002 ID 04f2:b593 control = auto, autosuspend_delay_ms = 2000 -- Chicony Electronics Co., Ltd HP Wide Vision FHD Camera (uvcvideo)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)

**Additional context**

The battery in this laptop has been replaced with an aftermarket unit.
Occasionally the laptop does not fully power down and I am forced to hold down the power key to achieve shut-off.

linrunner, 2024-07-10:
At this point, the service start is aborted:

> Jul 09 09:08:57 jLaptop tlp[1712]: Setting battery charge thresholds.../usr/sbin/tlp: **14: readonly: BATDRV_LG_BCL: is read only**

However, the error that the shell thinks it recognizes in the script does not exist. I cannot reproduce it here under Bookworm either.

I need to know which shell is running the scripts on your system. Please show the output of

    ls -l /bin/sh
    cat /etc/shells

Does the problem also occur with this command (show the output)?

    sudo tlp noop

Then please reinstall TLP and  check if the error still occurs (show the output):

    sudo apt install --reinstall tlp

JayH1119, 2024-07-10:
**After reboot:**

jay@jLaptop:~$ ls -l /bin/sh
lrwxrwxrwx 1 root root 4 Jan  5  2023 /bin/sh -> dash

jay@jLaptop:~$ cat /etc/shells
/bin/sh
/bin/bash
/usr/bin/bash
/bin/rbash
/usr/bin/rbash
/bin/dash
/usr/bin/dash
/bin/zsh
/usr/bin/zsh
/usr/bin/sh

jay@jLaptop:~$ echo $SHELL
/bin/zsh

jay@jLaptop:~$ sudo tlp noop  
/usr/sbin/tlp: 14: readonly: BATDRV_LG_BCL: is read only

linrunner, 2024-07-10:
Have you done the reinstallation?

JayH1119, 2024-07-10:
Apologies, yes. After reinstall and reboot...

linrunner, 2024-07-10:
Then let's take a look what the shell actually does:

    sudo sh -x /usr/sbin/tlp noop 2>&1 > tlp-noop.out

Please attach the output file **tlp-noop.out**.

linrunner, 2024-07-10:
> **/usr/share/tlp/bat.d/36-lg-legacy**

This file is the cause. It is not included in the Debian Bookworm package tlp 1.5.0-2. It is only available in later TLP versions.

How does it get on your disk?

JayH1119, 2024-07-11:
Well, that's embarrassing. I had previously installed and removed tlp, apparently from source rather than from the distro, and that was leftover. 

Thank you kindly for your time.

closed 2024-07-11 by JayH1119

linrunner, 2024-07-11:
That's what I suspected. You should uninstall the package once more, delete the directory **/usr/share/tlp** and install the package again.