linrunner/TLP #750: [Feature request] Use the tlp command without sudo

joaopauloalbq, 2024-07-13:
**Is your feature request related to a problem? Please describe.**

This is a usability problem, asking for authentication every time the user changes profiles manually.

**Describe the solution you'd like**

A clear and concise description of:

* Your use case(s)
I want to create a widget/dashboard to switch between ac and bat profiles.

* What you want to happen
Use the _tlp_ commands without sudo:
`$ tlp true`
`$ tlp false`

**Additional context**

powerprofilesctl doesn't need authentication to change the power profile.
Other operating systems change energy profiles without asking for privileges too.
> Mac
> [image]

> Windows
> [image]

linrunner, 2024-07-21:
Hi,

this would require a daemon running with root privilege, which TLP does not have. 

https://linrunner.de/tlp/developers/architecture.html

Adding a daemon would mean more or less rewriting TLP from scratch, what I do not intend to do anytime soon.

The basic idea of TLP is to automatically change the profile depending on the power source and providing just two profiles AC/BAT. The use case of a manual change has no priority.

You may of course write a root daemon (that calls tlp) yourself or you may look at the possibilities of /etc/sudoers.

Btw. I consider `tlp false|true` deprecated. Better use `tlp bat|ac`.

joaopauloalbq, 2024-07-24:
Hi Linrunner, 

I understand. 

Thank you for TLP

closed 2024-07-24 by joaopauloalbq