linrunner/TLP #751: Install tp-smapi kernel modules for ThinkPad battery thresholds and recalibration - Ubuntu 24.04

Max-1960, 2024-07-25:
I work with:
Ubuntu 24.04
Thinkpad X201
TLP 1.6.1

I have installed extra module ('sudo apt install tp-smapi-dkms') with this response:
"tp-smapi-dkms è già alla versione più recente (0.44-1ubuntu1) " .

When I ask with terminal command about battery care/status, I receive this message:
+++ Recommendations
* Install tp-smapi kernel modules for ThinkPad battery thresholds and recalibration

What do I need to do to complete a correct TLP installation ? 

In attacched there is output of the main commands TLP.

[Appunti-TLP.docx](https://github.com/user-attachments/files/16372109/Appunti-TLP.docx)

[Appunti-TLP.docx](https://github.com/user-attachments/files/16372081/Appunti-TLP.docx)

Thanks

linrunner, 2024-07-29:
Hi. Word files are not accepted here. Please show your outputs via gist https://gist.github.com/. Thanks!

Max-1960, 2024-07-29:
--- TLP 1.6.1 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad X201 3680AC2
BIOS           = 6QET70WW (1.40 )
EC Firmware    = 1.21
OS Release     = Ubuntu 24.04 LTS
Kernel         = 6.8.0-38-generic #38-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun  7 15:25:01 UTC 2024 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-6.8.0-38-generic root=UUID=852a41af-fad9-4b09-b97f-7966f23742f3 ro quiet splash
Init system    = systemd v255 (255.4-1ubuntu8.2)
Boot mode      = BIOS (CSM, Legacy)
Suspend mode   = s2idle [deep]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 16:46:19, 933 sec(s) ago
Mode           = battery
Power source   = battery

--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad-legacy
Supported features: none available
Driver usage:
* tp-smapi (tp_smapi) = inactive (kernel module 'tp_smapi' not installed)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 42T4649
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/status                         = Discharging

+++ Recommendations
* Install tp-smapi kernel modules for ThinkPad battery thresholds and recalibration

massimo@thinkpad-x201:~$ sudo apt install tp-smapi-dkms
[sudo] password di massimo: 
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze... Fatto
Lettura informazioni sullo stato... Fatto   
tp-smapi-dkms è già alla versione più recente (0.44-1ubuntu1).
0 aggiornati, 0 installati, 0 da rimuovere e 3 non aggiornati.

massimo@thinkpad-x201:~$ sudo tlp-stat
[sudo] password di massimo: 
--- TLP 1.6.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0006: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0009: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0010: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0011: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0013: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0014: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
[... 166 lines cut ...]
Exclude WWAN      = disabled

Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 002 ID 8087:0020 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 002 Device 002 ID 8087:0020 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)

+++ Battery Care
Plugin: thinkpad-legacy
Supported features: none available
Driver usage:
* tp-smapi (tp_smapi) = inactive (kernel module 'tp_smapi' not installed)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 42T4649
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/status                         = Discharging

+++ Recommendations
* Install tp-smapi kernel modules for ThinkPad battery thresholds and recalibrationInstall smartmontools for disk drive health info

massimo@thinkpad-x201:~$

Max-1960, 2024-07-29:
[b3b9e29e0aee8e664577ddeef14ac3d8-c578c909f02e865754f1852ef5708126420c52f2.zip](https://github.com/user-attachments/files/16409058/b3b9e29e0aee8e664577ddeef14ac3d8-c578c909f02e865754f1852ef5708126420c52f2.zip)

Max-1960, 2024-07-29:
[gistfile1.txt](https://github.com/user-attachments/files/16409082/gistfile1.txt)

Max-1960, 2024-07-29:
Thanks for your support. Can you help me ?

linrunner, 2024-07-29:
I will try, but only if you stop posting strange file attachments here immediately!

If you have problems using Gist, then paste the output here, as you have already done.

Now please post the output of

    sudo apt install --reinstall tp-smapi-dkms

Max-1960, 2024-07-29:
I execute cmd: sudo apt install --reinstall tp-smapi-dkms and this is output:

massimo@thinkpad-x201:~$ LC_ALL=C sudo apt install --reinstall tp-smapi-dkms
[sudo] password for massimo: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 3 not upgraded.
Need to get 32.3 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu noble/universe amd64 tp-smapi-dkms amd64 0.44-1ubuntu1 [32.3 kB]
Fetched 32.3 kB in 1s (44.4 kB/s)                     
(Reading database ... 191651 files and directories currently installed.)
Preparing to unpack .../tp-smapi-dkms_0.44-1ubuntu1_amd64.deb ...
Unpacking tp-smapi-dkms (0.44-1ubuntu1) over (0.44-1ubuntu1) ...
Setting up tp-smapi-dkms (0.44-1ubuntu1) ...
libdvd-pkg: Checking orig.tar integrity...
/usr/src/libdvd-pkg/libdvdcss_1.4.3.orig.tar.bz2: OK
libdvd-pkg: `apt-get check` failed, you may have broken packages.

After Boot, I execute command: sudo tlp-stat and this is output:

--- TLP 1.6.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0006: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0009: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0010: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0011: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0013: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0014: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
defaults.conf L0044: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
defaults.conf L0046: USB_EXCLUDE_BTUSB="0"
defaults.conf L0047: USB_EXCLUDE_PHONE="0"
defaults.conf L0048: USB_EXCLUDE_PRINTER="1"
defaults.conf L0049: USB_EXCLUDE_WWAN="0"
defaults.conf L0050: RESTORE_DEVICE_STATE_ON_STARTUP="0"
[... 142 lines cut ...]

Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 002 ID 8087:0020 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 005 ID 04e8:6863 control = auto, autosuspend_delay_ms = 2000 -- Samsung Electronics Co., Ltd Galaxy series, misc. (tethering mode) (rndis_host)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 002 Device 002 ID 8087:0020 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)

+++ Battery Care
Plugin: thinkpad-legacy
Supported features: none available
Driver usage:
* tp-smapi (tp_smapi) = inactive (kernel module 'tp_smapi' not installed)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 42T4649
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/status                         = Discharging

+++ Recommendations
* Install tp-smapi kernel modules for ThinkPad battery thresholds and recalibrationInstall smartmontools for disk drive health info

massimo@thinkpad-x201:~$

Max-1960, 2024-07-30:
massimo@thinkpad-x201:~$ wget -P /tmp http://ftp.de.debian.org/debian/pool/main/t/tp-smapi/tp-smapi-dkms_0.44-1_all.deb
--2024-07-30 11:02:20--  http://ftp.de.debian.org/debian/pool/main/t/tp-smapi/tp-smapi-dkms_0.44-1_all.deb
Risoluzione di ftp.de.debian.org (ftp.de.debian.org)... 141.76.2.4
Connessione a ftp.de.debian.org (ftp.de.debian.org)|141.76.2.4|:80... connesso.
Richiesta HTTP inviata, in attesa di risposta... 200 OK
Lunghezza: 41852 (41K) [application/vnd.debian.binary-package]
Salvataggio in: ‘/tmp/tp-smapi-dkms_0.44-1_all.deb.1’

tp-smapi-dkms_0.44- 100%[===================>]  40,87K   262KB/s    in 0,2s    

2024-07-30 11:02:21 (262 KB/s) - ‘/tmp/tp-smapi-dkms_0.44-1_all.deb.1’ salvato [41852/41852]

===================================

massimo@thinkpad-x201:~$ sudo apt install /tmp/tp-smapi-dkms_0.44-1_all.deb
[sudo] password di massimo: 
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze... Fatto
Lettura informazioni sullo stato... Fatto   
Nota, viene selezionato "tp-smapi-dkms" al posto di "/tmp/tp-smapi-dkms_0.44-1_all.deb"
I seguenti pacchetti saranno RETROCESSI:
  tp-smapi-dkms
0 aggiornati, 0 installati, 1 retrocessi, 0 da rimuovere e 3 non aggiornati.
È necessario scaricare 0 B/41,9 kB di archivi.
Dopo quest'operazione, verranno occupati 13,3 kB di spazio su disco.
Continuare? [S/n] S
Scaricamento di:1 /tmp/tp-smapi-dkms_0.44-1_all.deb tp-smapi-dkms all 0.44-1 [41,9 kB]
dpkg: attenzione: retrocessione di tp-smapi-dkms dalla versione 0.44-1ubuntu1 al
la 0.44-1
(Lettura del database... 191681 file e directory attualmente installati.)
Preparativi per estrarre .../tp-smapi-dkms_0.44-1_all.deb...
Estrazione di tp-smapi-dkms (0.44-1) su (0.44-1ubuntu1)...
Configurazione di tp-smapi-dkms (0.44-1)...
Loading new tp_smapi-0.44 DKMS files...
Building for 6.8.0-39-generic
Building initial module for 6.8.0-39-generic
Done.

thinkpad_ec.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-39-generic/updates/dkms/

tp_smapi.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-39-generic/updates/dkms/

hdaps.ko.zst:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.8.0-39-generic/updates/dkms/
depmod.....
libdvd-pkg: Checking orig.tar integrity...
/usr/src/libdvd-pkg/libdvdcss_1.4.3.orig.tar.bz2: OK
libdvd-pkg: `apt-get check` failed, you may have broken packages. Aborting…

===================================

massimo@thinkpad-x201:~$ sudo apt-mark hold tp-smapi-dkms
tp-smapi-dkms impostato come bloccato.

===================================

massimo@thinkpad-x201:~$ sudo tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad-legacy
Supported features: charge thresholds, recalibration
Driver usage:
* tp-smapi (tp_smapi) = active (status, charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  2..96(default)
* STOP_CHARGE_THRESH_BAT0/1:   6..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/devices/platform/smapi/BAT0/manufacturer               = SANYO
/sys/devices/platform/smapi/BAT0/model                      = 42T4649
/sys/devices/platform/smapi/BAT0/manufacture_date           = 2018-08-03
/sys/devices/platform/smapi/BAT0/first_use_date             = 2019-04-05
/sys/devices/platform/smapi/BAT0/cycle_count                =     87
/sys/devices/platform/smapi/BAT0/temperature                =     24 [°C]
/sys/devices/platform/smapi/BAT0/design_capacity            =  84240 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =  74410 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =  17260 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =     23 [%]
/sys/devices/platform/smapi/BAT0/remaining_running_time_now = not_discharging [min]
/sys/devices/platform/smapi/BAT0/remaining_charging_time    =     98 [min]
/sys/devices/platform/smapi/BAT0/power_now                  =  36805 [mW]
/sys/devices/platform/smapi/BAT0/power_avg                  =  36086 [mW]
/sys/devices/platform/smapi/BAT0/state                      = charging

/sys/devices/platform/smapi/BAT0/start_charge_thresh        =     96 [%]
/sys/devices/platform/smapi/BAT0/stop_charge_thresh         =    100 [%]
/sys/devices/platform/smapi/BAT0/force_discharge            =      0

Charge                                                      =   23.2 [%]
Capacity                                                    =   88.3 [%]

closed 2024-08-13 by linrunner