linrunner/TLP #752: STOP_CHARGE_THRESH_BAT0 etc not Honored

johnnya23, 2024-07-28:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Lap keeps charging beyond STOP_CHARGE_THRESH_BAT0 setting

**Expected behavior**

when the device reaches the threshold charging should stop and it should deplete the battery to the START_CHARGE_THRESH_BAT0 before "recharging"

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Only relevant to AC
2. set the STOP_CHARGE_THRESH_BAT0="80" in /etc/tlp.conf and plugin laptop and let charge to (and in my case beyond) 80%
3. n/a
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
  `sudo tlp-stat
--- TLP 1.6.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0006: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
/etc/tlp.conf L0056: TLP_PERSISTENT_DEFAULT="1"
defaults.conf L0009: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0010: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0011: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0013: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0014: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0032: WIFI_PWR_ON_AC="off"
defaults.conf L0033: WIFI_PWR_ON_BAT="on"
defaults.conf L0034: WOL_DISABLE="Y"
defaults.conf L0035: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0036: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0037: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0038: BAY_POWEROFF_ON_AC="0"
defaults.conf L0039: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0040: BAY_DEVICE="sr0"
defaults.conf L0041: RUNTIME_PM_ON_AC="on"
defaults.conf L0042: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0043: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon"
defaults.conf L0044: USB_AUTOSUSPEND="1"
defaults.conf L0045: USB_EXCLUDE_AUDIO="1"
defaults.conf L0046: USB_EXCLUDE_BTUSB="0"
defaults.conf L0047: USB_EXCLUDE_PHONE="0"
defaults.conf L0048: USB_EXCLUDE_PRINTER="1"
defaults.conf L0049: USB_EXCLUDE_WWAN="0"
defaults.conf L0050: RESTORE_DEVICE_STATE_ON_STARTUP="0"
/etc/tlp.conf L0549: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0052: NATACPI_ENABLE="1"
[... 165 lines cut ...]
* vendor (ideapad_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 0(off), 1(on) -- battery conservation mode

/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode = 0 (100%)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = LGES
/sys/class/power_supply/BAT1/model_name                     = L22L3PA4
/sys/class/power_supply/BAT1/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  56600 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  56600 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  47700 [mWh]
/sys/class/power_supply/BAT1/power_now                      =  33534 [mW]
/sys/class/power_supply/BAT1/status                         = Charging

Charge                                                      =   84.3 [%]
Capacity                                                    =  100.0 [%]

+++ Recommendations
* Install smartmontools for disk drive health info`

**Additional context**

This is a lenovo Slimpad, so I don't think the Thinkpad issues apply? THANKS

linrunner, 2024-07-29:
Hi, with non-thinkpad Lenovos, you cannot freely select the threshold value, as the hardware only supports a fixed threshold.

The documentation explains how to configure: 
https://linrunner.de/tlp/settings/bc-vendors.html#lenovo-non-thinkpad-series

johnnya23, 2024-07-29:
ok, it seems the start threshold vaule is irrelevant (I tried 0 and 70). Setting stop threshold to "1" causes my LENOVO IdeaPad Slim 5 16IRL8 82XF to hold at a charge of 79%. I guess that will do. thanks.

closed 2024-07-30 by linrunner