linrunner/TLP #753: Device type "unknown"

micmalti, 2024-07-30:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Hi, I've been happily using tlp for 5+ years and never had any issues until I recently switched over to a new laptop. Now, I keep getting this kernel message every 5 seconds at random intervals, both on battery and AC:
```sd 0:0:0:0: [sda] Starting disk```

From what I understand, a similar message is reported whenever tlp endlessly tries to stop/start the laptop disk, so I checked powertop and confirmed that the ```Runtime PM for disk sda``` option was marked as 'good'. The [recommendation](https://forum.manjaro.org/t/hard-drive-hdd-endlessly-cycling-stopping-starting-disk-when-laptop-is-on-battery/87089/6) for disabling this option is to uncomment ```SATA_LINKPWR_DENYLIST="hostX"```, yet when I ran ```tlp-stat -d``` to determine the host, it seems as though the disk isn't being recognized by tlp: [tlp-stat](https://gist.github.com/micmalti/54357801ae2cf356c79abb6162504c16)

Is this perhaps due to a general lack of support for UFS drives, which this particular model is making use of (SDINFDO4-128G)?

micmalti, 2024-07-30:
It isn't tlp that's spamming the log but a [bad kernel commit](https://lore.kernel.org/lkml/39143ca8-68e4-44eb-8619-0b935aa81603@kernel.org/T/). It's still unclear though why tlp isn't properly identifying the disk.

linrunner, 2024-08-08:
Hi, I have no idea what a "UFS drive" supposed to be and what type of controller it uses, obviously it's not SATA/AHCI.

Please show the output of

    ls -l /dev/disk/by-id

micmalti, 2024-08-08:
Here's the output (main and EFS):
```
total 0
lrwxrwxrwx 1 root root  9 Aug  8 14:24 scsi-0WDC_SDINFDO4-128G_WDC -> ../../sda
lrwxrwxrwx 1 root root 10 Aug  8 14:24 scsi-0WDC_SDINFDO4-128G_WDC-part1 -> ../../sda1
lrwxrwxrwx 1 root root 10 Aug  8 14:24 scsi-0WDC_SDINFDO4-128G_WDC-part2 -> ../../sda2
```

linrunner, 2024-09-12:
@micmalti Sorry for the late reply. I'm currently busy with the 1.7 beta test.

I checked the kernel docs for UFS
- https://docs.kernel.org/scsi/ufs.html
- https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-driver-ufs

but couldn't figure it out how UFS pm works. Without having the hardware, I can neither investigate further nor develop or even test anything.

Your output shows the controller as PCIe device, so at least runtime pm will be enabled ("auto") on battery power :

> /sys/bus/pci/devices/0000:00:12.7/power/control = on   (0x010901, Universal Flash Storage controller, ufshcd)

micmalti, 2024-09-12:
That's fine by me. It's just a reporting issue then.