linrunner/TLP #754: laptop fails to boot

igarca, 2024-08-06:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Linux Mint 22 does not boot if tlp is installed. After mint logo, before desktop (cinnamon) appears, I get a black screen.

**Expected behavior**

Laptop (thinkpad t480) booting and shutting down normally.

**To Reproduce**

- tlp-stat: https://gist.github.com/igarca/90d43f391312be31309158dde94baeea
- inxi -Fxxrzc0: https://gist.github.com/igarca/b85a80d8bb453cb47966dd1c13963815

**Additional context**
I thought it was due to a LM 22 upgrading issue, so I decided to do fresh installed. The problem seemed to be resolved, but after I installed some apps, the problem came back. After testing, I can say undoubdtely that tlp is causing the problem. 80% of the times it does not boot, and somestimes it does not even shutdown. 
When tlp was installed I think it recommended me installing "tp-smapi-dkms smartmontools linux-tools-generic | linux-tools", not sure how relevant it is though.

linrunner, 2024-08-07:
Hi,

in fact, I recently had boot hangs with my X1 Carbon 6th, the same hardware generation as your T480. 

I found out that it was due to the kernel 6.8.0-39-generic. The problems do not occur with 6.8.0-38-generic. Please try booting with the -38.

igarca, 2024-08-08:
Hi,

Thank you. I booted a couple of times with 6.8.0-38-generic with no issues.
Should I report it to Ubuntu?

linrunner, 2024-08-08:
Yes, I would suggest that.

igarca, 2024-08-12:
Apparently it's been solved with 6.8.0-40-generic. I no longer experience boot hangs.

linrunner, 2024-08-13:
Great. Thanks for your feedback

closed 2024-08-13 by linrunner