linrunner/TLP #755: Shutdown on critial battery level

Massimo-B, 2024-08-15:
**Is your feature request related to a problem? Please describe.**

Is there any way to do a clean shutdown when running on battery and reaching a critical battery level?

In old laptop-mode-tools that I already removed, there was at least DISABLE_LAPTOP_MODE_ON_CRITICAL_BATTERY_LEVEL=1, not a real shutdown but some sort of sync before the system is dying. On TLP I don't see anything like that. Recently I ran out of battery and there was nothing about it in the logs, which seems to be no shutdown but no sync to logfiles either.

Is this a job for TLP generally?

**Describe the solution you'd like**

A configuration to set

* critical battery level
* What to do on this threshold: nothing, poweroff, suspend-to-disk, suspend-to-ram

linrunner, 2024-08-21:
Hi, afaik this functionality is already built into most distributions. In order for it to work reliably as a feature in TLP, TLP would first have to deactivate the built-in one. Since the existing implementations differ greatly (depending on the desktopm for example), this would require an unacceptably high development and testing effort.

closed 2024-08-26 by linrunner

Massimo-B, 2024-08-26:
Which package is responsible for that in most distributions?

linrunner, 2024-08-26:
I'd guess upower.