linrunner/TLP #758: Broken charging state detection with qcom_battmgr driver

tamara-schmitz, 2024-09-03:
**Describe the bug**

TLP reports to be in battery mode regardless whether the device is charging via USB-C or not. It appears that only power supply `qcom-battmgr-ac` is checked which does not exist as a port on the laptop instead of the USB-C charging port `qcom-battmgr-usb`.

**Expected behavior**

TLP properly detects whether the battery is charging or not by checking both power supply classes.

**To Reproduce**

The device in question is the Lenovo Thinkpad X13s with a Qualcomm Snapdragon 8cx Gen 3 platform. It can only be charged via USB-C. The power supply kernel driver is the qcom_battmgr.

OS: openSUSE Tumbleweed 20240902
Kernel: 6.10.5-1-default
TLP: 1.6.1-2.1

Other applications can report the charging state just fine.

**Additional context**

Checking the battery and state with `tlp` and `tlp-stat -b`

<details>

```bash
$ tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: qcom-battmgr-bat
/sys/class/power_supply/qcom-battmgr-bat/manufacturer       =SMP
/sys/class/power_supply/qcom-battmgr-bat/model_name         = LNV-5B10W51878
/sys/class/power_supply/qcom-battmgr-bat/cycle_count        =     75
/sys/class/power_supply/qcom-battmgr-bat/energy_full_design =  49530 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/energy_full        =  48800 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/energy_now         =  40504 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/power_now          =  32509 [mW]
/sys/class/power_supply/qcom-battmgr-bat/status             = Charging

/sys/class/power_supply/qcom-battmgr-bat/charge_control_start_threshold = (not available)
/sys/class/power_supply/qcom-battmgr-bat/charge_control_end_threshold = (not available)

Charge                                                      =   83.0 [%]
Capacity                                                    =   98.5 [%]

$ tlp start
TLP started in battery mode (auto).
```

</details>

Some sysfs files:

<details>

```bash
$ ls /sys/class/power_supply/
qcom-battmgr-ac  qcom-battmgr-bat  qcom-battmgr-usb  qcom-battmgr-wls

# on USB-C charging
$ cat /sys/class/power_supply/qcom-battmgr-ac/online
0
$ cat /sys/class/power_supply/qcom-battmgr-bat/status
Charging
$ cat /sys/class/power_supply/qcom-battmgr-usb/online
1
$ cat /sys/class/power_supply/qcom-battmgr-wls/online
[... 78 lines cut ...]
Sep 03 18:54:36 fridayaxe tlp[120370]: set_usb_suspend.done
Sep 03 18:54:36 fridayaxe tlp[120370]: set_run_flag.touch: usb_done; rc=0
Sep 03 18:54:36 fridayaxe tlp[120370]: check_thinkpad.not_a_thinkpad: model=ThinkPad X13s Gen 1
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.thinkpad.not_a_thinkpad
Sep 03 18:54:36 fridayaxe tlp[120370]: check_thinkpad.not_a_thinkpad: model=ThinkPad X13s Gen 1
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.thinkpad-legacy.not_a_thinkpad
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.asus.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.huawei.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.lenovo.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.samsung.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.lg.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.lg-legacy.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.sony.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.system76.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.toshiba.no_match
Sep 03 18:54:36 fridayaxe tlp[120370]: batdrv_init.generic: batteries=qcom-battmgr-bat
Sep 03 18:54:36 fridayaxe tlp[120370]: set_charge_thresholds.no_method
Sep 03 18:54:36 fridayaxe tlp[120370]: set_radio_device_states(1): enable= disable=
Sep 03 18:54:36 fridayaxe tlp[120370]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpDnzKaP -> /run/tlp/run.conf
Sep 03 18:54:36 fridayaxe tlp[120370]: unlock_tlp()
Sep 03 18:54:36 fridayaxe tlp[120370]: TLP started in battery mode (auto).
Sep 03 18:54:37 fridayaxe tlp[120628]: parse_args4config: tlp-stat -T --
```

</details>

linrunner, 2024-09-04:
Interesting.  So far, no one has reported on TLP on this hardware.

Please show the output of:
 
    tlp-stat --psup

Please also show the complete output of:

    tlp-stat

tamara-schmitz, 2024-09-07:
> Interesting. So far, no one has reported on TLP on this hardware.
> 
> Please show the output of **EDIT:** with and without the charger connected:
> 
> ```
> tlp-stat --psup
> ```

<details>

```
root@fridayaxe:/home/binkular# tlp-stat --psup
--- TLP 1.6.1 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/qcom-battmgr-ac/type: Mains
/sys/class/power_supply/qcom-battmgr-ac/usb_type: (not available)
/sys/class/power_supply/qcom-battmgr-ac/online: 0
/sys/class/power_supply/qcom-battmgr-ac/voltage_max: (not available)
/sys/class/power_supply/qcom-battmgr-ac/voltage_min: (not available)
/sys/class/power_supply/qcom-battmgr-ac/voltage_now: (not available)
/sys/class/power_supply/qcom-battmgr-ac/present: (not available)
/sys/class/power_supply/qcom-battmgr-ac/charge_control_start_threshold: (not available)
/sys/class/power_supply/qcom-battmgr-ac/charge_control_end_threshold: (not available)
/sys/class/power_supply/qcom-battmgr-ac/status: (not available)
/sys/class/power_supply/qcom-battmgr-ac/device/path: (not available)
/sys/class/power_supply/qcom-battmgr-bat/type: Battery
/sys/class/power_supply/qcom-battmgr-bat/usb_type: (not available)
/sys/class/power_supply/qcom-battmgr-bat/online: (not available)
/sys/class/power_supply/qcom-battmgr-bat/voltage_max: (not available)
/sys/class/power_supply/qcom-battmgr-bat/voltage_min: (not available)
/sys/class/power_supply/qcom-battmgr-bat/voltage_now: 7677000
/sys/class/power_supply/qcom-battmgr-bat/present: 1
/sys/class/power_supply/qcom-battmgr-bat/charge_control_start_threshold: (not available)
/sys/class/power_supply/qcom-battmgr-bat/charge_control_end_threshold: (not available)
/sys/class/power_supply/qcom-battmgr-bat/status: Charging
/sys/class/power_supply/qcom-battmgr-bat/device/path: (not available)
/sys/class/power_supply/qcom-battmgr-usb/type: USB
/sys/class/power_supply/qcom-battmgr-usb/usb_type: (not available)
/sys/class/power_supply/qcom-battmgr-usb/online: 1
/sys/class/power_supply/qcom-battmgr-usb/voltage_max: (not available)
/sys/class/power_supply/qcom-battmgr-usb/voltage_min: (not available)
/sys/class/power_supply/qcom-battmgr-usb/voltage_now: (not available)
/sys/class/power_supply/qcom-battmgr-usb/present: (not available)
/sys/class/power_supply/qcom-battmgr-usb/charge_control_start_threshold: (not available)
/sys/class/power_supply/qcom-battmgr-usb/charge_control_end_threshold: (not available)
/sys/class/power_supply/qcom-battmgr-usb/status: (not available)
/sys/class/power_supply/qcom-battmgr-usb/device/path: (not available)
/sys/class/power_supply/qcom-battmgr-wls/type: Wireless
/sys/class/power_supply/qcom-battmgr-wls/usb_type: (not available)
/sys/class/power_supply/qcom-battmgr-wls/online: 0
/sys/class/power_supply/qcom-battmgr-wls/voltage_max: (not available)
/sys/class/power_supply/qcom-battmgr-wls/voltage_min: (not available)
/sys/class/power_supply/qcom-battmgr-wls/voltage_now: (not available)
/sys/class/power_supply/qcom-battmgr-wls/present: (not available)
/sys/class/power_supply/qcom-battmgr-wls/charge_control_start_threshold: (not available)
/sys/class/power_supply/qcom-battmgr-wls/charge_control_end_threshold: (not available)
/sys/class/power_supply/qcom-battmgr-wls/status: (not available)
/sys/class/power_supply/qcom-battmgr-wls/device/path: (not available)

+++ udev diagnostic
Fatal Error: TLP's udev rule for power source changes (85-tlp.rules) is not active -- possible package bug.
```

</details>

Without charger:

<details>

[... 247 lines cut ...]
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: qcom-battmgr-bat
/sys/class/power_supply/qcom-battmgr-bat/manufacturer       =SMP
/sys/class/power_supply/qcom-battmgr-bat/model_name         = LNV-5B10W51878
/sys/class/power_supply/qcom-battmgr-bat/cycle_count        =     78
/sys/class/power_supply/qcom-battmgr-bat/energy_full_design =  49530 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/energy_full        =  48140 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/energy_now         =  12035 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/power_now          =  21483 [mW]
/sys/class/power_supply/qcom-battmgr-bat/status             = Charging

/sys/class/power_supply/qcom-battmgr-bat/charge_control_start_threshold = (not available)
/sys/class/power_supply/qcom-battmgr-bat/charge_control_end_threshold = (not available)

Charge                                                      =   25.0 [%]
Capacity                                                    =   97.2 [%]
```

</details>

linrunner, 2024-09-11:
> Hmm. Perhaps a bug with the openSUSE package? Or do you think the missing udev rule is even relevant here?

The missing rule is relevant. Could you please show the contents of the package, i.e.

     rpm -ql tlp
     rpm -ql tlp-rdw

The actual difference between AC and BA is

```
< /sys/class/power_supply/qcom-battmgr-bat/status: Charging
---
> /sys/class/power_supply/qcom-battmgr-bat/status: Discharging
25c25
< /sys/class/power_supply/qcom-battmgr-usb/online: 1
---
> /sys/class/power_supply/qcom-battmgr-usb/online: 0
```
At first glance I do not see why the responsible routine does not recognise the change in **online**:
https://github.com/linrunner/TLP/blob/1.6.1/tlp-func-base.in#L702 

I need a trace:

1. Add `TLP_DEBUG="run ps"` to your configuration
2. Execute

       sudo tlp noop

3. Show the output of
 
       sudo tlp-stat -T

linrunner, 2024-09-12:
@tamara-schmitz I have an idea. Please change **/usr/share/tlp/tlp-func-base** -> https://github.com/linrunner/TLP/commit/4f50bb329dbcabfb264aad9ac316a47956bf6a34

tamara-schmitz, 2024-09-13:
> Could you please show the contents of the packages, i.e.
> 
> ```
>  rpm -ql tlp
>  rpm -ql tlp-rdw
> ```

<details>

tlp

```
/etc/tlp.conf
/etc/tlp.d
/etc/tlp.d/00-template.conf
/etc/tlp.d/README
/usr/bin/bluetooth
/usr/bin/nfc
/usr/bin/run-on-ac
/usr/bin/run-on-bat
/usr/bin/tlp-stat
/usr/bin/wifi
/usr/bin/wwan
/usr/lib/systemd/system-sleep
/usr/lib/systemd/system-sleep/tlp
/usr/lib/systemd/system/tlp.service
/usr/lib/udev/rules.d/85-tlp.rules
/usr/lib/udev/tlp-usb-udev
/usr/sbin/rctlp
/usr/sbin/tlp
/usr/share/bash-completion
/usr/share/bash-completion/completions
/usr/share/bash-completion/completions/bluetooth
/usr/share/bash-completion/completions/nfc
/usr/share/bash-completion/completions/tlp
/usr/share/bash-completion/completions/tlp-stat
/usr/share/bash-completion/completions/wifi
/usr/share/bash-completion/completions/wwan
/usr/share/doc/packages/tlp
/usr/share/doc/packages/tlp/AUTHORS
/usr/share/doc/packages/tlp/README.rst
/usr/share/doc/packages/tlp/changelog
/usr/share/licenses/tlp
/usr/share/licenses/tlp/COPYING
/usr/share/licenses/tlp/LICENSE
/usr/share/man/man1/bluetooth.1.gz
/usr/share/man/man1/nfc.1.gz
/usr/share/man/man1/run-on-ac.1.gz
/usr/share/man/man1/run-on-bat.1.gz
/usr/share/man/man1/wifi.1.gz
/usr/share/man/man1/wwan.1.gz
/usr/share/man/man8/tlp-stat.8.gz
/usr/share/man/man8/tlp.8.gz
/usr/share/man/man8/tlp.service.8.gz
/usr/share/metainfo
/usr/share/metainfo/de.linrunner.tlp.metainfo.xml
/usr/share/tlp
/usr/share/tlp/bat.d
/usr/share/tlp/bat.d/05-thinkpad
/usr/share/tlp/bat.d/10-thinkpad-legacy
/usr/share/tlp/bat.d/15-asus
/usr/share/tlp/bat.d/20-huawei
/usr/share/tlp/bat.d/25-lenovo
/usr/share/tlp/bat.d/30-samsung
/usr/share/tlp/bat.d/35-lg
/usr/share/tlp/bat.d/36-lg-legacy
/usr/share/tlp/bat.d/40-sony
/usr/share/tlp/bat.d/45-system76
/usr/share/tlp/bat.d/50-toshiba
/usr/share/tlp/bat.d/90-generic
[... 699 lines cut ...]
Sep 13 08:47:20 fridayaxe tlp[72904]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpGuqREw -> /run/tlp/run.conf
Sep 13 08:47:20 fridayaxe tlp[72936]: +++ init start (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 13 08:47:20 fridayaxe tlp[72936]: SHELL=/usr/bin/bash; umask=0022
Sep 13 08:47:20 fridayaxe tlp[72936]: get_sys_power_supply(qcom-battmgr-ac).ac_offline: syspwr=1
Sep 13 08:47:20 fridayaxe tlp[72936]: power_source=bat
Sep 13 08:47:20 fridayaxe tlp[72936]: manual_mode=none
Sep 13 08:47:20 fridayaxe tlp[72936]: power_mode=bat
Sep 13 08:47:20 fridayaxe tlp[72936]: compare_and_save_power_state(1).different: old=
Sep 13 08:47:20 fridayaxe tlp[72936]: Applying power save settings...done.
Sep 13 08:47:20 fridayaxe tlp[72936]: Setting battery charge thresholds...done.
Sep 13 08:47:20 fridayaxe tlp[72936]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpjxNiL6 -> /run/tlp/run.conf
Sep 13 08:48:17 fridayaxe tlp[73639]: +++ noop (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 13 08:48:17 fridayaxe tlp[73639]: SHELL=/usr/bin/bash; umask=0022
Sep 13 08:48:17 fridayaxe tlp[73639]: get_sys_power_supply(qcom-battmgr-ac).ac_offline: syspwr=1
Sep 13 08:48:17 fridayaxe tlp[73639]: power_source=bat
Sep 13 08:48:17 fridayaxe tlp[73639]: manual_mode=none
Sep 13 08:48:17 fridayaxe tlp[73639]: power_mode=bat
Sep 13 08:48:17 fridayaxe tlp[73639]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpL4Y0Xj -> /run/tlp/run.conf
```

</details>

> @tamara-schmitz I have an idea. Please change **/usr/share/tlp/tlp-func-base** -> [4f50bb3](https://github.com/linrunner/TLP/commit/4f50bb329dbcabfb264aad9ac316a47956bf6a34)

Hmm. I tried that and restarted TLP but I saw no change. It still thinks I am in battery mode.

tamara-schmitz, 2024-09-17:
Yes. Sorry I am a bit slow.

> > Sep 13 08:48:17 fridayaxe tlp[73639]: get_sys_power_supply(qcom-battmgr-ac).ac_offline: syspwr=1
> 
> This is not plausible after the change. The new regex should filter the **qcom-battmgr-ac** device. Are you sure a typo hasn't got into it? Or is the trace from _before_ the change? Then please show it again.
> 

I took that trace before the change. Here is a trace after the change:

<details>

```
--- TLP 1.6.1 --------------------------------------------

Sep 17 10:36:52 fridayaxe tlp[1515]: +++ auto (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 17 10:36:52 fridayaxe tlp[1515]: SHELL=/usr/bin/bash; umask=0022
Sep 17 10:36:52 fridayaxe tlp[1515]: get_sys_power_supply(qcom-battmgr-bat).bat_not_discharging_recheck: bs=Charging; syspwr=; wait=15
Sep 17 10:36:53 fridayaxe tlp[1515]: get_sys_power_supply(qcom-battmgr-bat).bat_discharging: syspwr=1; wait=11
Sep 17 10:36:53 fridayaxe tlp[1515]: power_source=bat
Sep 17 10:36:53 fridayaxe tlp[1515]: manual_mode=none
Sep 17 10:36:53 fridayaxe tlp[1515]: power_mode=bat
Sep 17 10:36:53 fridayaxe tlp[1515]: compare_and_save_power_state(1).different: old=
Sep 17 10:36:53 fridayaxe tlp[1515]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpkyKcZq -> /run/tlp/run.conf
Sep 17 10:36:53 fridayaxe tlp[1683]: +++ auto (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 17 10:36:53 fridayaxe tlp[1683]: SHELL=/usr/bin/bash; umask=0022
Sep 17 10:36:53 fridayaxe tlp[1683]: get_sys_power_supply(qcom-battmgr-bat).bat_discharging: syspwr=1; wait=
Sep 17 10:36:53 fridayaxe tlp[1683]: power_source=bat
Sep 17 10:36:53 fridayaxe tlp[1683]: manual_mode=none
Sep 17 10:36:53 fridayaxe tlp[1683]: power_mode=bat
Sep 17 10:36:53 fridayaxe tlp[1683]: compare_and_save_power_state(1).equal
Sep 17 10:36:53 fridayaxe tlp[1775]: +++ auto (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 17 10:36:53 fridayaxe tlp[1775]: SHELL=/usr/bin/bash; umask=0022
Sep 17 10:36:53 fridayaxe tlp[1775]: get_sys_power_supply(qcom-battmgr-bat).bat_discharging: syspwr=1; wait=
Sep 17 10:36:53 fridayaxe tlp[1775]: power_source=bat
Sep 17 10:36:53 fridayaxe tlp[1775]: manual_mode=none
Sep 17 10:36:53 fridayaxe tlp[1775]: power_mode=bat
Sep 17 10:36:53 fridayaxe tlp[1775]: compare_and_save_power_state(1).equal
Sep 17 10:36:53 fridayaxe tlp[1841]: +++ auto (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 17 10:36:53 fridayaxe tlp[1841]: SHELL=/usr/bin/bash; umask=0022
Sep 17 10:36:53 fridayaxe tlp[1841]: get_sys_power_supply(qcom-battmgr-bat).bat_discharging: syspwr=1; wait=
Sep 17 10:36:53 fridayaxe tlp[1841]: power_source=bat
Sep 17 10:36:53 fridayaxe tlp[1841]: manual_mode=none
Sep 17 10:36:53 fridayaxe tlp[1841]: power_mode=bat
Sep 17 10:36:53 fridayaxe tlp[1841]: compare_and_save_power_state(1).equal
Sep 17 10:36:54 fridayaxe tlp[1920]: +++ auto (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 17 10:36:54 fridayaxe tlp[1920]: SHELL=/usr/bin/bash; umask=0022
Sep 17 10:36:54 fridayaxe tlp[1920]: get_sys_power_supply(qcom-battmgr-bat).bat_not_discharging_recheck: bs=Charging; syspwr=; wait=15
Sep 17 10:36:55 fridayaxe tlp[1920]: get_sys_power_supply(qcom-battmgr-bat).bat_not_discharging: bs=Not charging; syspwr=0; wait=0
Sep 17 10:36:55 fridayaxe tlp[1920]: get_sys_power_supply(qcom-battmgr-usb).usb_online: syspwr=0
Sep 17 10:36:55 fridayaxe tlp[1920]: power_source=ac
Sep 17 10:36:55 fridayaxe tlp[1920]: manual_mode=none
Sep 17 10:36:55 fridayaxe tlp[1920]: power_mode=ac
Sep 17 10:36:55 fridayaxe tlp[1920]: compare_and_save_power_state(0).different: old=1
Sep 17 10:36:55 fridayaxe tlp[1920]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpv1dOq6 -> /run/tlp/run.conf
Sep 17 10:36:55 fridayaxe tlp[2299]: +++ auto (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 17 10:36:55 fridayaxe tlp[2299]: SHELL=/usr/bin/bash; umask=0022
Sep 17 10:36:55 fridayaxe tlp[2299]: get_sys_power_supply(qcom-battmgr-bat).bat_not_discharging_recheck: bs=Not charging; syspwr=; wait=15
Sep 17 10:36:57 fridayaxe tlp[2299]: get_sys_power_supply(qcom-battmgr-bat).bat_not_discharging: bs=Not charging; syspwr=0; wait=0
Sep 17 10:36:57 fridayaxe tlp[2299]: get_sys_power_supply(qcom-battmgr-usb).usb_online: syspwr=0
Sep 17 10:36:57 fridayaxe tlp[2299]: power_source=ac
Sep 17 10:36:57 fridayaxe tlp[2299]: manual_mode=none
Sep 17 10:36:57 fridayaxe tlp[2299]: power_mode=ac
Sep 17 10:36:57 fridayaxe tlp[2299]: compare_and_save_power_state(0).equal
Sep 17 10:36:57 fridayaxe tlp[2390]: +++ auto (1.6.1) ++++++++++++++++++++++++++++++++++++++++
Sep 17 10:36:57 fridayaxe tlp[2390]: SHELL=/usr/bin/bash; umask=0022
Sep 17 10:36:57 fridayaxe tlp[2390]: get_sys_power_supply(qcom-battmgr-bat).bat_not_discharging_recheck: bs=Not charging; syspwr=; wait=15
Sep 17 10:36:59 fridayaxe tlp[2390]: get_sys_power_supply(qcom-battmgr-bat).bat_not_discharging: bs=Not charging; syspwr=0; wait=0
Sep 17 10:36:59 fridayaxe tlp[2390]: get_sys_power_supply(qcom-battmgr-usb).usb_online: syspwr=0
Sep 17 10:36:59 fridayaxe tlp[2390]: power_source=ac
Sep 17 10:36:59 fridayaxe tlp[2390]: manual_mode=none
[... 222 lines cut ...]
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=89
POWER_SUPPLY_VOLTAGE_MAX_DESIGN=0
POWER_SUPPLY_VOLTAGE_NOW=8622000
POWER_SUPPLY_POWER_NOW=20201000
POWER_SUPPLY_ENERGY_FULL_DESIGN=49530000
POWER_SUPPLY_ENERGY_FULL=48510000
POWER_SUPPLY_ENERGY_EMPTY=4336000
POWER_SUPPLY_ENERGY_NOW=38808000
POWER_SUPPLY_TEMP=298
POWER_SUPPLY_MANUFACTURE_YEAR=2022
POWER_SUPPLY_MANUFACTURE_MONTH=5
POWER_SUPPLY_MANUFACTURE_DAY=18
POWER_SUPPLY_MODEL_NAME=LNV-5B10W51878
POWER_SUPPLY_MANUFACTURERSMP
POWER_SUPPLY_SERIAL_NUMBER=1388
SEQNUM=6590
USEC_INITIALIZED=10816387
NVME_HOST_IFACE=none
```
</details>

And looking at `tlp-stat --system` it changes the state correctly ;)

Thank you for fixing the issue ^^

linrunner, 2024-09-17:
Great. Thanks for reporting!

linrunner, 2024-09-18:
One more thing: `tlp-stat -p` does not display the CPU model. Can I get the full output of 

    cat /proc/cpuinfo

?

linrunner, 2024-09-19:
> I guess TLP assumes x86?

I have only ever developed on x86(_64). To be honest, I'm a bit surprised that only small changes are necessary for aarch64 so far. 
 
> The CPU name does not seem to appear anywhere in the kernel dmesg either. So I guess that the better way to obtain this info is using dmidecode, although one cannot rely on that perhaps fully either? But it does work well on both my x86_64 and aarch64 machines.

dmidecode isn't available on all systems, and I'm not keen on adding new dependencies on external tools. That's why I take a two-step approach in the above change to **/usr/bin/tlp-stat**. Please try it out.

Thanks a lot for your input on the ARM subject!

tamara-schmitz, 2024-09-19:
> Please try it out.
> 
> Thanks a lot for your input on the ARM subject!

Yey, works ;)

```
tlp-stat -p
--- TLP 1.6.1 --------------------------------------------

+++ Processor
CPU model = Snapdragon (TM) 8cx Gen 3 @ 3.0 GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = qcom-cpufreq-hw
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = schedutil
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = ondemand performance schedutil
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   300000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2438400 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies = 300000 403200 499200 595200 691200 806400 902400 1017600 1113600 1209600 1324800 1440000 1555200 1670400 1785600 1881600 1996800 2112000 2227200 2342400 2438400 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq  =   300000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq  =  2438400 [kHz]

/sys/devices/system/cpu/cpu1..cpu7: omitted for clarity, use -v to show all

/sys/devices/system/cpu/cpufreq/boost                  = (not available)
/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

+++ Platform Profile
/sys/firmware/acpi/platform_profile                    = (not available)
/sys/firmware/acpi/platform_profile_choices            = (not available)
```

linrunner, 2024-09-19:
Thanks. Coming with version 1.7 ... :-D

linrunner, 2024-09-20:
@tamara-schmitz TLP 1.7 Beta 2 is out, containing all changes from this issue https://github.com/linrunner/TLP/blob/main/changelog

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1:

closed 2024-09-27 by linrunner