linrunner/TLP #759: TLP refuses to recalibrate on a Lenovo W520

tadawson, 2024-09-03:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Attempts to recalibrate the battery on this system errantly fail, and claim a hardare error.  
(Basically a redeax of issue #438, with more insight)

**Expected behavior**

Successful recalibration, without error. 

**To Reproduce**

Run "tlp recalibrate"

--- TLP 1.7.0-alpha.1 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0029: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_MSG_COLORS="91 93 1 92"
defaults.conf L0009: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0010: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0011: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0013: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0014: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0015: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0016: NMI_WATCHDOG="0"
defaults.conf L0017: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0018: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0019: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0020: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0021: DISK_IOSCHED="keep keep"
defaults.conf L0022: SATA_LINKPWR_ON_AC="med_power_with_dipm"
defaults.conf L0023: SATA_LINKPWR_ON_BAT="med_power_with_dipm"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0025: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0026: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0027: PCIE_ASPM_ON_AC="default"
defaults.conf L0028: PCIE_ASPM_ON_BAT="default"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0030: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0032: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0033: AMDGPU_ABM_LEVEL_ON_AC="0"
defaults.conf L0034: AMDGPU_ABM_LEVEL_ON_BAT="1"
defaults.conf L0035: WIFI_PWR_ON_AC="off"
defaults.conf L0036: WIFI_PWR_ON_BAT="on"
defaults.conf L0037: WOL_DISABLE="Y"
/etc/tlp.conf L0330: SOUND_POWER_SAVE_ON_AC="0"
defaults.conf L0039: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0040: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0041: BAY_POWEROFF_ON_AC="0"
defaults.conf L0042: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0043: BAY_DEVICE="sr0"
defaults.conf L0044: RUNTIME_PM_ON_AC="on"
defaults.conf L0045: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0046: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon xhci_hcd"
defaults.conf L0047: USB_AUTOSUSPEND="1"
defaults.conf L0048: USB_EXCLUDE_AUDIO="1"
defaults.conf L0049: USB_EXCLUDE_BTUSB="0"
defaults.conf L0050: USB_EXCLUDE_PHONE="0"
defaults.conf L0051: USB_EXCLUDE_PRINTER="1"
defaults.conf L0052: USB_EXCLUDE_WWAN="0"
defaults.conf L0053: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0054: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0055: NATACPI_ENABLE="1"
defaults.conf L0056: TPSMAPI_ENABLE="1"
[... 214 lines cut ...]

Charge                                                      =   19.2 [%]
Capacity                                                    =   69.4 [%]

Add any other context about the problem here.

This problem did not occur when this system was on a 4.x based kernel and Slackware14.2 . . . I noted it once updated to a 5.15.165 kernel and Slackware15 (TLP installed from source).  I have subsequently tested the 5.17.18 kernel and natacpi functionality, with no difference. 

Note that this system *WILL* recallibrate cleanly from Windows, so the claim of a "hardware issue" is suspect at best . . . 

The other confirming point to this conclusion is:

I note that on this device Lenovo W520) with 1.7.0a and natacpi, that the recall is triggered by simply writing "force-discharge" to /sys/class/power_devices/BAT0/charge_behaviour with a fully charged battery. After seeing 100% fail of recal using tlp, I triggered the recall manually, and sat back and watched.

The recal ran perfectly with no glitches or issues of any kind!

This screams to me that something in the monitoring loop is getting bad data and focing the abort in tlp - because the EC certainly is not aborting!

(I did note that when I had tested in Windows, that when things got lower, that occasionally some parameters reporting would be absent for a polling cycle . . . I wonder if this is a factor? The monitor fails a read and forces an abort.)

Watching things run, I'm frankly unclear as to why the recal monitor watches anything but charge_behaviour . . . . If that doesn't toggle back to "auto", it's still running, and the percentage is irrelevant . . .

I had watched the status, charge_behaviour, and percentage remaining while I ran this test, and saw no issues.  I did note that when running the recal in Windows, as it progressed, there were intervals when some of the status data was blank - makes me wonder if the EC just gets busy.  Since batdrv_foce_discharge_active is watching the tp_smapi devices and not the natacpi (as I was), I have to wonder if that driver perhaps "stutters", since it appears that there is no underlying issue whatsoever, and that tlp is seeing something as an error, and termination a recalibration that would otherwise complete. 

To that end, I have done a bit of "test hackery" in batdvr_force_discharge_active to have it watch the natacpi device instead, since that appears to be the trigger for the loop to exit.  (Alternatively, the test might be altered to look *for* "Charging" as an exit condition, thus making it immune to null query returns).

tadawson, 2024-09-03:
Somewhat confirming one of my speculations, I not that wile tlp is polling, I am sometimes seeing this in my log, which seems consistent with getting a bad status read and aborting, when in actuality, nothing has failed:

Tue Sep  3 18:10:20 2024] thinkpad_ec: thinkpad_ec_read_row: failed waiting for data: (0x08:0x00)->0xfffffff0

tadawson, 2024-09-03:
confirmed! Usig the natacpi device resolves this issue!

Ideally, I think the actual problem lies in tp_smapci timjng out ec reads prematurely, since the napacpi driver sees the same data OK.Bassing that, for tlp, there would appear to be a few options:

1) Change the monitor device
2) Change the test to look  for [cH]arging and not [dD]ischarging to ignore numm results.
3) Insert a retry loop in this test in case an invalid result (or failed read) occirs.
4) ???

I hope this helps!

linrunner, 2024-09-04:
Hi, thanks for your report. 

To be honest: your explanations are quite long and I have difficulty understanding them fully.

Let's start at the very beginning, where the problem occurs. I need the complete output of (with the non-hacked code, of course):

    tlp recalibrate

> To that end, I have done a bit of "test hackery" in batdvr_force_discharge_active to have it watch the natacpi device instead

It would speed up the analysis if you show a `diff -u` of your changes to **/usr/share/tlp/bat.d/t05-thinkpad**

> (also in a gist, but not sure how to associate it here, so forgive my redundancy . . .)

Just post the browser URL to your gist ..

tadawson, 2024-09-04:
The error is pretty simple (and I really don't want to put another cycle on my batter to just grab the output, when I have seen this reported in numerous tickets). 

Basically, running "tlp recalibrate" exits incomplete with the error claiming bad hardware/firmware, typically at or below 20% remaining battery (but no consistent number).

A manual run of recal on the box without tlp works cleanly. (echo "force-discharge > /sys/class/power_devices/BAT0/charge_behaviour") so the hardware/firmware is validated to *NOT* have a fault. 

The issue appears to be that the thinkpad_ec (tp_smapi) driver sometimes does not return a value, and the monitoring loop in tlp is thinking that the recal aborted when it has not, and then tlp cancels the recal by forcing "auto" back into charge_behavious. 

Simply using the other status device (/sys/class/power_supply/BAT0/status vs /sys/devices/platform/smapi/BAT0/state) resolves this, since the natapci device does not appear to have this quirk, as crudely tested below.  It looks like when both tp_smapi and natacpi are available (as is my case), that tlp prefers the tp_smapi devices.  Were I to not have those installed, this likely would not fail either. 

Diff of bat.d/05-thinkpad:
+++ 05-thinkpad 2024-09-03 22:37:36.315725299 -0400
@@ -344,6 +344,8 @@
         natacpi) _bd_read="$ACPIBATDIR/$bs" ;;
         tpsmapi) _bd_read="$SMAPIBATDIR/$bs" ;;
     esac
+# Dawson - Test
+        _bd_readA="$ACPIBATDIR/$bs"
 
     echo_debug "bat" "batdrv.${_batdrv_plugin}.select_battery($1): bat_str=$_bat_str; cfg=$_bt_cfg_bat; bat_idx=$_bat_idx; bd_read=$_bd_read; bf_start=$_bf_start; bf_stop=$_bf_stop; bf_dischg=$_bf_dischg"
     return 0
@@ -817,6 +819,15 @@
         # force_discharge is still on, but quirky firmware (e.g. ThinkPad E-series)
         # may keep force_discharge on --> check battery status too
         case $_bm_read in
+            tpsmapi)
+# Dawson - test
+                _bat_st=$(read_sysf "$_bd_readA/status")
+                # determine soc for log entry
+                _bat_soc=$(read_sysf "$_bd_read/remaining_percent")
+                _bat_pwr=$(read_sysval "$_bd_read/power_avg")
+                _bat_volt=$(read_sysval "$_bd_read/voltage")
+                ;;
+
             natacpi)
                 _bat_st=$(read_sysf "$_bd_read/status")
                 # determine soc for log entry
@@ -833,14 +844,6 @@
                 fi
                 ;;
 
-            tpsmapi)
-                _bat_st=$(read_sysf "$_bd_read/state")
-                # determine soc for log entry
-                _bat_soc=$(read_sysf "$_bd_read/remaining_percent")
-                _bat_pwr=$(read_sysval "$_bd_read/power_avg")
-                _bat_volt=$(read_sysval "$_bd_read/voltage")
-                ;;
-
             *) # no read api
                 rc=255
                 ;;

linrunner, 2024-09-04:
I have now understood that discharging actually starts and then stops too early, i.e. at 20% (from what you wrote at https://github.com/linrunner/TLP/issues/438#issuecomment-2325189824). 

But the error message is really meant literally, it is the firmware of the embedded controller (EC) that aborts the discharging process! 

> Error: battery $_bat_str was not discharged completely i.e. **terminated by the firmware** -- check your hardware (battery, charger).

TLP polls the contents of **/sys/class/power_supply/BAT0/charge_behavior** in the loop. As soon as it is no longer **force-discharge**, the EC has finished discharging and `tlp recalibrate` terminates with the error message if the charge level is above 0%.

**Bottom line: the error is caused by your hardware aborting the discharge too early (or reporting too early). There is nothing to fix in the TLP code.** 

I could imagine that your 10 year(!) old battery suddenly drops from 20% to 0%, so that the EC reacts accordingly.

A W520 could also have this problem: https://linrunner.de/tlp/faq/battery.html#erratic-battery-behavior-on-thinkpad-t420-s-t520-w520-x220-and-possibly-later-models

By the way, your diff is completely illegible. I didn't bother to decipher it.

tadawson, 2024-09-05:
OK, I have completed the last test mentioned in my last post, and this confirms what I have been saying - the tp_smapi devices are failing, and the hardware is *NOT* aborting the recal!

I have restored 05-thinkpad to as distributed, and added 4 echo statements at line 849 (the string check to exit the monitoring loop): 

        # evaluate battery state
        case "$_bat_st" in
            [Dd]ischarging) rc=0 ;;

            *)
        echo Failed string test and exited at `date` > /tmp/TLPEXIT
        echo "BatSt from tlp" is: $_bat_st >> /tmp/TLPEXIT
        echo State from natapci dev: `cat /sys/class/power_supply/BAT0/status` >
> /tmp/TLPEXIT
        echo ChargeBehavious from natapci dev: `cat /sys/class/power_supply/BAT0
/charge_behaviour` >> /tmp/TLPEXIT
        rc=1
        ;;

This outputs the value of the string that failed the compare (in this case from the tp_smapi driver, as chosen by tlp) as well as the status reported by the quivalent natacpi status device, and the device that shows the mode (charge_behaviour). 

"tlp recalibrate" was started, and exited thus:

Currently discharging battery BAT0:
voltage            =  11403 [mV]
remaining capacity =  53400 [mWh]
remaining percent  =     90 [%]
remaining time     =    105 [min]
power              = -30252 [mW]
state              = discharging
force-discharge    = 1
Press Ctrl+C to cancel.
Error: battery BAT0 was not discharged completely i.e. terminated by the firmware -- check your hardware (battery, charger).
Battery recalibration aborted.

The values output via the echo commands are: 

Failed string test and exited at Thu Sep 5 17:37:05 EDT 2024
BatSt from tlp is: none
State from natapci dev: Discharging
ChargeBehavious from natapci dev: auto inhibit-charge [force-discharge]

Note that the tlp_smapi device gave an invalid result (should be "idle", "charging" or "discharging") but that the other monitoring devices continue to show the battery as discharging as well as the mode to be "force-discharge", as in in recalibrate. 

I also ran debug on bat in tlp, and that gave at this point in time (note that it does not log the variables above, hence my having done so):

Sep  5 17:36:55 bleeft tlp[25562]: batdrv.thinkpad.force_discharge_active(BAT0):
 bm_read=tpsmapi; st=discharging; soc=90; pwr=-29479; voltage=11426; rc=0
Sep  5 17:37:00 bleeft tlp[25562]: batdrv.thinkpad.force_discharge_active(BAT0):
 bm_read=tpsmapi; st=discharging; soc=90; pwr=-30252; voltage=11403; rc=0
Sep  5 17:37:05 bleeft tlp[25562]: batdrv.thinkpad.force_discharge_active(BAT0):
 bm_read=tpsmapi; st=none; soc=90; pwr=-29772; voltage=11416; rc=1
Sep  5 17:37:05 bleeft tlp[25562]: batdrv.thinkpad.discharge.not_emptied(BAT0):
bm_read=tpsmapi; fd=1; st=discharging; soc=90; pwr=-35652; voltage=11416

Sep  5 17:37:05 bleeft tlp[25562]: batdrv.thinkpad.write_force_discharge(BAT0, 0
): bm_dischg=natacpi; bf_dischg=/sys/class/power_supply/BAT0/charge_behaviour; b
at_idx=1; rc=0

Note alto that hte last line recorded above, calling write_force_discharge(BAT0, 0) is what is aborting the recal - not the hardware, since that function is: 

            case "$1" in
                0) write_sysf "auto" "$_bf_dischg" || rc=5 ;;
                1) write_sysf "force-discharge" "$_bf_dischg" || rc=5 ;;

where writing "auto" into the device aborts force_discharge.

This without a doubt confirms that the tp_smapi status device is unreliable, and that the hardware is fine!

Considering that the natacpi devices are required to be able to initiate recal, I strongly suggest that you also patch to use the natacpi device to monitor the discharge, since it appears to be reliable and will always be available of recal is possible. 

(I am also going to file a but with the tp_smapi group, since this is not actually a tlp bug, but one in the driver.  It just seems simplest to alter tlp at this point to work around it, since it can already use that device in some cases.

linrunner, 2024-09-06:
Well. Thank you for working it out and explaining it again. I have now understood your point:

Reading **/sys/devices/platform/smapi/BAT0/state** produces glitches and should better be replaced by **/sys/class/power_supply/BAT0/status**. That's what I'm going to do.

> confirms what I have been saying - the tp_smapi devices are failing, and the hardware is NOT aborting the recal!

You're talking about the case of _your_ machine, I'm talking about the general case, because that's where it is, as you can see in [line 807](https://github.com/linrunner/TLP/blob/main/bat.d/05-thinkpad#L807). The additional check of the battery status for ‘discharging’ was introduced because certain models - e.g. the E-series, initially notorious for its buggy firmware - did not reset charge_behaviour at the end of the recalibration.

I am always reluctant to introduce such workarounds because technical debts have to be repaid sooner or later. As happened in your case. A case like yours has never been reported to me in the 14 years that I have been developing TLP and using tp_smapi. 

@evgeni any thoughts on these glitches?

I'm still thinking about whether I want to secure **/sys/class/power_supply/BAT0/status** against glitches as a precaution and will get back to you with a new code version soon. We are still in time for 1.7.

.

evgeni, 2024-09-06:
I've not seen this behaviour so far, no.
But I've also must admit that I never used the smapi state file in a loop or anything, mostly just to set the upper and lower charging threshold and then rely on the "normal" ACPI output for the state (also, I don't use tlp).

That said, SandyBridge SMAPI is a tad weird (I hacked up support for it in a day at DebConf11 on a borrowed X220) and it requires [disabling some of the safety-features](https://github.com/linux-thinkpad/tp_smapi/commit/280da64b126ab8f59695920b0b4f5432d6b24fb7) of the code to even work, so I'd not be surprised if there is more instability on specifically this line of machines.

tadawson, 2024-09-06:
The interesting thing is that querying the same data via the natacpi driver seems to be 100% solid, which is why I speculate that this is not a problem unique to this system, but perhaps this generation.  I have also been unable (so far) to force this error out of tlp, but I have also not tried testing reading both devices at the same time as tlp is doing, so it may be a conflict between the two (I note that tp_smapi is trying to set a lock on access to the EC, and perhaps they are clashing).  It might also just be that the EC is slow on this gen of systems, but considering that 05-thinkpad in tlp is targeted towards this generation, I think that using the alternate device would be safe. 

I note also that when searching, I have read a number of reports/posts in various plalces that sound like it *could*  be this, I'm just not sure if anyone has taken the time to dig into it. 

In any case, thanks for your patience and assistance!

tadawson, 2024-09-06:
OK, one last data point!  I have tested doing reads against the tp_smapi and natacpi drives simultaneously, and when I do so, I get the errors from the tp_smapi pretty quickly.  Looking at the tp_smapi driver source, as noted above, it appears to set a lock when querying the EC, and it would seem that if the natacpi is trying to do so also, that the timeouts in tp_smapi (or other delay handling) are not adequate in that it times out instead of waiting for access.

To do this, I just set up two while loops in shells - one reading the tm_smapi devs and the other the natacpi devs, and watched syslog for the tp_smapi error - I would be interested if this behaviour can be found on any other Thinkpad system (all I have is 2x W520 and 1x T510, so no other generations to test here . . . .)

while true
do
cat /sys/class/power_supply/BAT0/* > /dev/null 2>/dev/null
done

and 

while true
do
cat /sys/devices/platform/smapi/BAT0/* > /dev/null 2>/dev/null
done

are what triggers this in tp_smapi here quite quickly.

evgeni, 2024-09-06:
I have x61t, x300 and x201s here to test, just need to remember that during the weekend

tadawson, 2024-09-06:
I think it would be a useful test.  Considering that this driver has been in use prior to the 1.7.x variants, it would seem that it *might* be a factor in any of the reports of discharge not reaching 0% on at least this gen or hardware (or any other that uses it).

linrunner, 2024-09-08:
@tadawson "thinkpad_ec_read_row: failed waiting for data" also happens - on my trusty X220 - if I do not read out _all_ files simultaneously, but only **/sys/class/power_supply/BAT0/status** and **/sys/devices/platform/smapi/BAT0/state**.

You can find the new discharge code using **/sys/class/power_supply/BAT0/status** only in branch https://github.com/linrunner/TLP/tree/issue/759. Please test.

linrunner, 2024-09-08:
> it would seem that it _might_ be a factor in any of the reports of discharge not reaching 0% on at least this gen or hardware (or any other that uses it).

As far as I remember, these were all models newer than *20 i.e. tp_smapi was not in the game.

tadawson, 2024-09-08:
Thanks! I've pulled the branch and installed . . . baseline battery functions and stats still seem good, and confirm that it is reading the correct dev based on the output case, and will enter/exit recal correctly. It's late, so will test a full recal tomorrow.

tadawson, 2024-09-08:
tlp recalibrate tested and confirmed to work as expected, and setcharge also confirmed good.

I also observed that while watching the recal, that a few of the other parameters shown in tlp-stat -b occasionally did not report due to the tp_smapi issue, but that is an inconvenience more than anything else.

All in all, this change looks good and OK to merge from here.

- Tim

tadawson, 2024-09-09:
Oh, and you might consider updating the tags on this issue, since there is a fix, and there is no battery defect (to help others search, if nothing else).

- Ti

linrunner, 2024-09-10:
@tadawson Of course. Merged to main. Thanks for reporting and testing this!

tadawson, 2024-09-10:
And thank you for your patience in letting me prove to you that this was actually a bug!

linrunner, 2024-09-10:
You're welcome.

closed 2024-09-10 by linrunner

tadawson, 2024-09-10:
Reported as bug #70 to the tp_smapi authors.

jedenastka, 2024-09-18:
I'm just here to report that I had this issue on my X230 (so, not 20 series), and the latest commit fixed it as well. Might be a good idea to push a new release, so distros can package it, and users can benefit from the fix.

evgeni, 2024-09-18:
Does smapi even work on X230? I thought 20 was the latest, but never had a 30 in my hands.

tadawson, 2024-09-18:
I don't have a 30 either, but always thought that the 30 and 20 were very similar, and the 40 was where things changed, but am not certain.

jedenastka, 2024-09-18:
Hmm...

```
~ $ LANG=en_US.UTF-8 less /sys/devices/platform/smapi/BAT0
/sys/devices/platform/smapi/BAT0: No such file or directory
~ $ lsmod | grep smapi                                    
~ $ 
```

jedenastka, 2024-09-19:
Nevermind, I think I'm just dumb. I was able to run `recalibrate` to completion with the latest released version. I think my charging cable is failing and this was simply a coincidence. Sorry.

linrunner, 2024-09-19:
@jedenastka The change under consideration here only affects the special case of *20 generation plus tp_smapi. So you had a different problem.

If you want to report a problem again, remember to give detailed information i.e. at least the output of the error message in context and a complete tlp-stat. You can't do anything with the “narrative” of a message. Thanks for your understanding.

linrunner, 2024-09-20:
@tadawson TLP 1.7 Beta 2 is out, containing all changes from this issue https://github.com/linrunner/TLP/blob/main/changelog

tadawson, 2024-09-20:
Installed, thanks.

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1: