linrunner/TLP #760: [TLP 1.7 Beta] Call for testers

linrunner, 2024-09-05:
Hi @ALL,

as with every upcoming release, I'm excited to get your help with test coverage!

**Test Objectives**

1. General regression testing (check if everything still works for you)
   - When changing the power source AC <-> BAT, do the tunables change accordingly (compare output of `tlp-stat`)?
   - Charge thresholds can be changed 
   - `tlp recalibrate` works - _ThinkPads only_

2. Battery Care
   - Check if newly supported hardware is able to set start and stop charge threshold: 
     - Apple Silicon Macbooks (M*) with MacOS 13.0 (or later) firmware and  Asahi Linux kernel 6.6 (or later) - **important: the firmware restricts the possible values to 75/80 and 95/100.**
     - MSI laptops with kernel 6.3 or later - **important: the kernel driver *msi_ec* only accepts individual models and BIOS versions, for the unsupported ones` tlp-stat -b` displays `Plugin: generic / Supported features: none available`**
   - ThinkPads with coreboot: does` tlp-stat -b` show complete battery readings? See https://linrunner.de/tlp/faq/battery.html#are-thinkpads-with-coreboot-supported

3.  Adaptive Backlight Modulation on AMD Vega or newer GPUs with kernel 6.9 or later
    - Configure `AMDGPU_ABM_LEVEL_ON_BAT=3` and check with `tlp-stat -g` if the setting applies on battery = `/sys/class/drm/card1/card1-eDP-1/amdgpu/panel_power_savings = 3` - on my AMD ThinkPad, it saves 0.5W at medium display brightness with a slight (subjective) loss of brightness, it has no effect at full or minimum display brightness.

4. General features
    - Are error, warning, notice and success messages shown in color?
    - Check the new Fish shell command completion
  
5. Bugfixes
    - `xhci_hcd` was added to `RUNTIME_PM_DRIVER_DENYLIST` defaults - do USB devices work properly on battery?

**Beta Packages**
See the [download page](https://download.linrunner.de/packages/).

**What do I need from the testers?**
After installing please start the beta with

    sudo tlp start

I'm expecting your unabbreviated outputs of

    sudo tlp-stat

on battery and AC (via [Gist](https://gist.github.com/) please).

**Bug Reports**
If something does not work as expected → please open an [issue](https://github.com/linrunner/TLP/issues).

**I  look forward to your feedback on the beta :-)**

**References**
- https://github.com/linrunner/TLP/blob/main/changelog
- https://github.com/linrunner/TLP/releases/tag/1.7.0-beta.1
- https://github.com/linrunner/TLP/issues/707
- https://github.com/linrunner/TLP/pull/685

Mylinde, 2024-09-05:
Let's test.

-  BAT https://gist.github.com/Mylinde/61d92a3a5b16c3833c27db51a0818c7a
-  AC https://gist.github.com/Mylinde/5dd937813cc1deb747560081806dd079

Battery Care:  START_CHARGE_THRESH_BAT0 not working

https://gist.github.com/Mylinde/1bcf27389a455df92989d05d35417c08

Adaptive Backlight Modulation working:

- BAT https://gist.github.com/Mylinde/5ebc4d6c058f76f8549f5a16071b07f8
- AC https://gist.github.com/Mylinde/3bb2a9e09a7afc7cbec97cb92132526f

linrunner, 2024-09-05:
@Mylinde thanks for testing and your outputs.

> Battery Care: START_CHARGE_THRESH_BAT0 not working, is set to 75 and has worked in latest release.

Negative. Your memory must be betraying you. The kernel driver _asus_wmi_ has always supported a stop threshold only.

Mylinde, 2024-09-05:
... yes will be, 89-asus tells you are right. Thanks so far, everythink works as aspected.

dkwo, 2024-09-09:
All looks good on Apple silicon https://paste.c-net.org/BrieflyCrushed

BarbUk, 2024-09-10:
I have some error during `sudo tlp-stat`:
```
❯ sudo tlp-stat  > /dev/null
Use of uninitialized value $bus in concatenation (.) or string at /usr/share/tlp/tlp-usblist line 111.
Use of uninitialized value $dev in concatenation (.) or string at /usr/share/tlp/tlp-usblist line 111.
Use of uninitialized value $desc in substitution (s///) at /usr/share/tlp/tlp-usblist line 112.
Use of uninitialized value $bus in concatenation (.) or string at /usr/share/tlp/tlp-usblist line 115.
Use of uninitialized value $dev in concatenation (.) or string at /usr/share/tlp/tlp-usblist line 115.
Use of uninitialized value $usbid in concatenation (.) or string at /usr/share/tlp/tlp-usblist line 115.
Use of uninitialized value in concatenation (.) or string at /usr/share/tlp/tlp-usblist line 115.
Use of uninitialized value $_[0] in string at /usr/share/tlp/tlp-usblist line 61.
```

Output of `tlp-stat`:
 - [AC](https://gist.github.com/BarbUk/e9e79cac09ea831cd93b7ad338933c8a)
 - [BAT](https://gist.github.com/BarbUk/5f7e235abfd7d1f741d3e07873f79bc0)

BarbUk, 2024-09-10:
[lsusb](https://gist.github.com/BarbUk/7f5febf53e571a5055892dcf9351c2e3)

linrunner, 2024-09-10:
@BarbUk unexpected linebreak in lsusb's output

> Bus 001 Device 006: ID 04d9:a293 Holtek Semiconductor, Inc. OBINS
> AnnePro2

Probably needs another plausibility check. Sigh. Stay tuned.

H-Sachse, 2024-09-11:
Here's my log files for the Lenovo Edge 11 with AMD Athlon(tm) II Neo K325 Dual-Core Processor
Debian Bookworm with kernel 6.1.106-3

A/C connected:
[tlp-stat-AC.log](https://github.com/user-attachments/files/16967858/tlp-stat-AC.log)
A/C disconnected:
[tlp-stat-BAT.log](https://github.com/user-attachments/files/16967859/tlp-stat-BAT.log)
calibration in discharge phase:
[tlp-stat-cal-dischg.log](https://github.com/user-attachments/files/16967864/tlp-stat-cal-dischg.log)
calibration, discharge done:
[tlp-stat-cal-chg.log](https://github.com/user-attachments/files/16967861/tlp-stat-cal-chg.log)
manually fixed charge thresholds for full charge:
[tlp-stat-cal-chg2.log](https://github.com/user-attachments/files/16967862/tlp-stat-cal-chg2.log)

Here's my issues:
- Charge thresholds cannot be set individually both values are the same (e.g. with tlp setcharge 80 85, both values will be 80, except if I use setcharge 96 100 it goes to 256/100 see tlp-stat-cal-chg2.log)
- charge thresholds are not set correctly by "recalibrate" for a full charge (see tlp-stat-cal-chg.log)
- battery was stuck in "idle" after discharge from calibration, had to re-plug the A/C to resolve, but could not reproduce.
- battery state toggles between idle and charging when almost full, but this might have to do with the battery itself and/or the driver

linrunner, 2024-09-12:
@H-Sachse thanks for testing and reporting.

Regarding you issues:
> * Charge thresholds cannot be set individually both values are the same (e.g. with tlp setcharge 80 85, both values will be 80, except if I use setcharge 96 100 it goes to 256/100 see [...]

False threshold _readings_ are typical for the Edge. Nonetheless the configured values are properly _written_ and work -> [FAQ](https://linrunner.de/tlp/faq/battery.html#charge-thresholds-shown-by-tlp-stat-b-do-not-correspond-to-the-configured-ones). 

Lenovo's firmware for the Edge and E series behaves differently to that of all other series. There is no solution other than ignoring the false readings. Just be happy that this beautiful old treasure is alive and kicking.

> * battery was stuck in "idle" after discharge from calibration, had to re-plug the A/C to resolve, but could not reproduce.
> * battery state toggles between idle and charging when almost full, but this might have to do with the battery itself and/or the driver

Possibly EC firmware issue too.

tisyang, 2024-09-12:
Test looks fine.

log ac&bat : https://gist.github.com/tisyang/477bd6e053f9a6f19a9870bc09dc1a77

BarbUk, 2024-09-12:
> @BarbUk fix commited.

> Please replace /usr/share/tlp/tlp-usblist with https://raw.githubusercontent.com/linrunner/TLP/main/tlp-usblist
and re-test - with keyboard connected of course.

Thanks for the fix.

No error with the new commit:

```
Bus 001 Device 001 ID 1d6b:0002 control = on,   autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 003 ID 0cf3:e300 control = auto, autosuspend_delay_ms = 2000 -- Qualcomm Atheros Communications QCA61x4 Bluetooth 4.0 (btusb)
Bus 001 Device 005 ID 04d9:a293 control = on,   autosuspend_delay_ms = 2000 -- Holtek Semiconductor, Inc. OBINS (usbhid)
Bus 002 Device 001 ID 1d6b:0002 control = on,   autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 002 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =    0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 002 Device 003 ID 04f2:b217 control = auto, autosuspend_delay_ms = 2000 -- Chicony Electronics Co., Ltd Lenovo Integrated Camera (0.3MP) (uvcvideo)
Bus 002 Device 004 ID 046d:c092 control = on,   autosuspend_delay_ms = 2000 -- Logitech, Inc. G102/G203 LIGHTSYNC Gaming Mouse (usbhid)
Bus 003 Device 001 ID 1d6b:0002 control = on,   autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = on,   autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)
```

linrunner, 2024-09-12:
@BarbUk great. Thanks for the confirmation.

linrunner, 2024-09-13:
Beta packages for Fedora 40 are now available. See the [download page](https://download.linrunner.de/packages/).

linrunner, 2024-09-19:
T14 Gen 4 AMD
<script src="https://gist.github.com/linrunner/a1f123e786aa1a2966ef7ca6d622b599.js"></script>

linrunner, 2024-09-20:
I have just released Beta 2:

* https://github.com/linrunner/TLP/blob/main/changelog
* https://github.com/linrunner/TLP/releases/tag/1.7.0-beta.2
* https://download.linrunner.de/packages/

I would be so grateful if you could retest and post the corresponding outputs. Thank you so much! I really appreciate your help with this.

The packages for the Ubuntu PPA and the Fedora Beta Repo will take some time.

BarbUk, 2024-09-20:
Thanks for the work and the new release :rocket: 

- [AC](https://gist.github.com/BarbUk/b9314650851618c8d2690f8ccd7a707a)
- [BAT](https://gist.github.com/BarbUk/42e4c7b3e37dedc81cb4a26624295941)

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1:

closed 2024-09-27 by linrunner

linrunner, 2025-01-19:
TLP 1.8.0 Beta 1 is now available! I'd love your help in testing it --> https://github.com/linrunner/TLP/issues/781

linrunner, 2025-12-01:
The beta test is complete, and [TLP 1.9.0](https://github.com/linrunner/TLP/releases/tag/1.9.0) is now released. Have fun!