linrunner/TLP #762: Charge level treated as about 10x more than it is on Lenovo X1 Extreme Gen 2?

qpkorr, 2024-09-13:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

On my Lenovo X1 Extreme Gen 2, it looks as though tlp doing something weird when reading the charge percentage - missing the decimal point, and rounding things - I've not worked it out yet.

For example, with charge thresholds set to 60% and 64%, and charge at 7.1%, it's not charging:
<details>
<summary>tlp-stat -b</summary>
#: john@j4lt:~ ; sudo tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B-+���5'+8"$��
/sys/class/power_supply/BAT0/cycle_count                    =    159
/sys/class/power_supply/BAT0/energy_full_design             =  13840 [mWh]
/sys/class/power_supply/BAT0/energy_full                    = 653840 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  46290 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

/sys/class/power_supply/BAT0/charge_control_start_threshold =     60 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     64 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =    7.1 [%]
Capacity                                                    = 4724.3 [%]
</details>

It's actually claiming the battery is Full - I've seen some other status when I first saw this problem a month or two ago - "not-charging" or something - can't recall sorry.

Increasing the charge thresholds to 67% and 68% (after a bit of fluffing around to narrow down the problem, which resulted in a little charging, hence 7.1% is up to 7.3%), the problem remains
<details>
<summary>tlp-stat -b</summary>
#: john@j4lt:~ ; sudo tlp setcharge 67 68
Setting temporary charge thresholds for BAT0:
  stop  =  68
  start =  67
#: john@j4lt:~ ; sudo tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B-+���5'+8"$��
/sys/class/power_supply/BAT0/cycle_count                    =    159
/sys/class/power_supply/BAT0/energy_full_design             =  13840 [mWh]
/sys/class/power_supply/BAT0/energy_full                    = 653840 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  47410 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

/sys/class/power_supply/BAT0/charge_control_start_threshold =     67 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     68 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =    7.3 [%]
Capacity                                                    = 4724.3 [%]

</details>

But increasing the start percentage to 68% and 69%, and suddenly it starts charging:
<details>
<summary>tlp-stat -b</summary>
#: john@j4lt:~ ; sudo tlp setcharge 68 69
Setting temporary charge thresholds for BAT0:
  stop  =  69
  start =  68
#: john@j4lt:~ ; sudo tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B-+���5'+8"$��
/sys/class/power_supply/BAT0/cycle_count                    =    159
/sys/class/power_supply/BAT0/energy_full_design             =  13840 [mWh]
/sys/class/power_supply/BAT0/energy_full                    = 653840 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  47410 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     68 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     69 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =    7.3 [%]
Capacity                                                    = 4724.3 [%]
</details>

Until it charges enough, and stops charging again:
<details>
<summary>tlp-stat -b</summary>
#: john@j4lt:~ ; sudo tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B-+���5'+8"$��
/sys/class/power_supply/BAT0/cycle_count                    =    159
/sys/class/power_supply/BAT0/energy_full_design             =  13840 [mWh]
/sys/class/power_supply/BAT0/energy_full                    = 653840 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  48110 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

/sys/class/power_supply/BAT0/charge_control_start_threshold =     68 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     69 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =    7.4 [%]
Capacity                                                    = 4724.3 [%]

</details>

I first saw this issue a few weeks ago (after upgrading to Ubuntu 24? I'm not sure), but somehow - perhaps after a reboot? it charged properly.  Now that the battery level has dropped below 10%, I'm getting warnings of low battery - and hence I noticed it again. 

**Expected behavior**

When the charge level is below the charge start threshold, I expect it to charge.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? AC since can't charge on battery.
2. Actions to reproduce the behaviour? I don't know sorry.  Happy to run any diagnostics you like while I've got it in this state.
3. Shell commands entered and their output
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
https://gist.github.com/qpkorr/4dfaad7928101d7534d54e303260bf87

linrunner, 2024-09-13:
Your battery reports impossible values 

> /sys/class/power_supply/BAT0/energy_full_design = **13840 [mWh]**
> /sys/class/power_supply/BAT0/energy_full = **653840 [mWh]**
> /sys/class/power_supply/BAT0/energy_now = 48110 [mWh]

They state that the battery was last charged to 654 Wh (there are no laptop batteries that big), although the design capacity is only 13.8 Wh.

Conclusion: your battery is defective, replace it.

Btw: the Bug Reporting Howto that you confirmed above states:
> **What not to report**
> * Hardware issues e.g. worn out or malfunctioning batteries

I don't want the TLP issue tracker to become a hardware support forum. Thank you for your understanding.

closed 2024-09-13 by linrunner

qpkorr, 2024-09-13:
Thank you - weird.  I ran `tlp-recalibrate` and once it was done, things look a lot more normal:
<details>
<summary>tlp-stat -b</summary>
#: john@j4lt:~/Downloads ; sudo tlp-stat -b
[sudo] password for john: 
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 5B10W13900
/sys/class/power_supply/BAT0/cycle_count                    =    160
/sys/class/power_supply/BAT0/energy_full_design             =  80400 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  69300 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  69300 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =  100.0 [%]
Capacity                                                    =   86.2 [%]
</details>

To my untrained eye at least, it looks as though there's nothing wrong with the battery (it's 4 years old but has been sitting at 60% on AC power the vast majority of that time).  Sorry for bothering you - thanks so much for such a great tool!

linrunner, 2024-09-14:
Maybe you're lucky and it was just a one-off glitch in the battery electronics.