linrunner/TLP #763: Selinux block tlp for 2 processes

keysteal72, 2024-09-26:
Hi, every time I boot my Fedora, it appears a Selinux warning that has blocked 2 processes started bye tlp with destination dac_override and search (on /var/lib/snapd folder). Anyone has same issue?

linrunner, 2024-09-29:
Hi, how do you expect to start a discussion if you are silent about all the facts? The very least would be the output of:

     sudo ausearch -su 'tlp' -ts boot

Please don't just copy the long output here. Instead, use https://gist.github.com/. Thanks!

keysteal72, 2024-09-29:
it's not long:
---
time->Sun Sep 29 15:40:01 2024
type=AVC msg=audit(1727617201.069:188): avc:  denied  { search } for  pid=2712 comm="tlp" name="snapd" dev="sda3" ino=718591 scontext=system_u:system_r:tlp_t:s0 tcontext=system_u:object_r:snappy_var_lib_t:s0 tclass=dir permissive=0
----
time->Sun Sep 29 15:40:01 2024
type=AVC msg=audit(1727617201.311:189): avc:  denied  { dac_override } for  pid=2712 comm="tlp" capability=1  scontext=system_u:system_r:tlp_t:s0 tcontext=system_u:system_r:tlp_t:s0 tclass=capability permissive=0
----
time->Sun Sep 29 15:40:01 2024
type=AVC msg=audit(1727617201.311:190): avc:  denied  { dac_override } for  pid=2712 comm="tlp" capability=1  scontext=system_u:system_r:tlp_t:s0 tcontext=system_u:system_r:tlp_t:s0 tclass=capability permissive=0

linrunner, 2024-09-29:
You are welcome to wait and see if anyone gets in touch. 

However, please note that I do not take care of SELinux issues upstream. If you want this fixed, you need to open a Fedora bug report against the selinux-policy package. If you do, please post the link here. Thank you.

bonswouar, 2025-02-10:
Not sure if it's related (but the title seems general enough)

I personally have an AVC denial when plugging/unplugging the AC. As far as I know that didn't happen before, not sure if it's config related or due to a tlp update or something else
Here are the details:
```
type=AVC msg=audit(1739221020.273:671): avc:  denied  { getattr } for  pid=19928 comm="tlp" path="/proc/sys/kernel/nmi_watchdog" dev="proc" ino=12660 scontext=system_u:system_r:tlp_t:s0-s0:c0.c1023 tcontext=system_u:object_r:sysctl_nmi_watchdog_t:s0 tclass=file permissive=0
```

It seems related to nmi_watchdog, not sure what it is, but I didn't uncomment / change `NMI_WATCHDOG` in the conf so I guess it should be disabled?

linrunner, 2025-02-13:
@bonswouar TLP's default is to disable nmi_watchdog. Please file a Fedora bug as explained above.

sigmavirus24, 2025-03-16:
For what it's worth, the existing bug is https://bugzilla.redhat.com/show_bug.cgi?id=2343780. I don't know that it makes sense to keep this open given this is a policy problem managed outside of this project.