linrunner/TLP #764: CPU max scaling bug that defaults CPU to do its maximum speed to its min frequency speed.

Ja4e, 2024-09-29:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
CPU max scaling bug that defaults CPU to do its maximum speed to its min frequency speed.
latest 1.7

A clear and concise description of what the bug is.

**Expected behavior**
If I touched it to min 0  and max 1800 it should do 1.8ghz max not more or less.

A clear and concise description of what you expected to happen.
*"Works fine" is not enough to analyze the problem!*

**To Reproduce**
Touching CPU Max Scalling defaults the current cpu policy to be within 400 and 400.

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? I tried battery well for AC not yet.
3. Actions to reproduce the behaviour. 
4. Shell commands entered and their output *including error message(s)* systemctl status tlp shows no error whatsoever 
5. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

`╰─$ sudo tlp-stat           
--- TLP 1.7.0 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0035: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_MSG_COLORS="91 93 1 92"
defaults.conf L0009: TLP_PERSISTENT_DEFAULT="0"
/etc/tlp.conf L0082: DISK_IDLE_SECS_ON_AC="0"
/etc/tlp.conf L0083: DISK_IDLE_SECS_ON_BAT="2"
/etc/tlp.conf L0088: MAX_LOST_WORK_SECS_ON_AC="15"
/etc/tlp.conf L0089: MAX_LOST_WORK_SECS_ON_BAT="60"
/etc/tlp.conf L0146: CPU_ENERGY_PERF_POLICY_ON_AC="performance"
/etc/tlp.conf L0147: CPU_ENERGY_PERF_POLICY_ON_BAT="power"
/etc/tlp.conf L0185: NMI_WATCHDOG="0"
defaults.conf L0017: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0018: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0019: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0020: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0021: DISK_IOSCHED="keep keep"
defaults.conf L0022: SATA_LINKPWR_ON_AC="med_power_with_dipm"
/etc/tlp.conf L0273: SATA_LINKPWR_ON_BAT="min_power"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0292: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0026: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0027: PCIE_ASPM_ON_AC="default"
/etc/tlp.conf L0392: PCIE_ASPM_ON_BAT="powersupersave"
/etc/tlp.conf L0331: RADEON_DPM_PERF_LEVEL_ON_AC="high"
/etc/tlp.conf L0332: RADEON_DPM_PERF_LEVEL_ON_BAT="low"
/etc/tlp.conf L0346: RADEON_POWER_PROFILE_ON_AC="high"
/etc/tlp.conf L0347: RADEON_POWER_PROFILE_ON_BAT="low"
defaults.conf L0033: AMDGPU_ABM_LEVEL_ON_AC="0"
defaults.conf L0034: AMDGPU_ABM_LEVEL_ON_BAT="1"
/etc/tlp.conf L0363: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0364: WIFI_PWR_ON_BAT="on"
defaults.conf L0037: WOL_DISABLE="Y"
defaults.conf L0038: SOUND_POWER_SAVE_ON_AC="1"
/etc/tlp.conf L0378: SOUND_POWER_SAVE_ON_BAT="10"
defaults.conf L0040: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0041: BAY_POWEROFF_ON_AC="0"
/etc/tlp.conf L0307: BAY_POWEROFF_ON_BAT="1"
defaults.conf L0043: BAY_DEVICE="sr0"
defaults.conf L0044: RUNTIME_PM_ON_AC="on"
defaults.conf L0045: RUNTIME_PM_ON_BAT="auto"
[... 195 lines cut ...]
Plugin: lenovo
Supported features: charge threshold
Driver usage:
* vendor (ideapad_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 0(off), 1(on) -- battery conservation mode

/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode = 0 (off)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = Sunwoda
/sys/class/power_supply/BAT0/model_name                     = L20D4PC1
/sys/class/power_supply/BAT0/cycle_count                    =    207
/sys/class/power_supply/BAT0/energy_full_design             =  80000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  70610 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =   4950 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   8231 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

Charge                                                      =    7.0 [%]
Capacity                                                    =   88.3 [%]`

**Additional context**

Add any other context about the problem here.

Ja4e, 2024-09-29:
╰─$ cpupower frequency-info
analyzing CPU 12:
  driver: amd-pstate-epp
  CPUs which run at the same hardware frequency: 12
  CPUs which need to have their frequency coordinated by software: 12
  maximum transition latency:  Cannot determine or is not supported.
  hardware limits: 400 MHz - 4.46 GHz
  available cpufreq governors: performance powersave
  current policy: frequency should be within 400 MHz and 400 MHz.
                  The governor "powersave" may decide which speed to use
                  within this range.
  current CPU frequency: Unable to call hardware
  current CPU frequency: 399 MHz (asserted by call to kernel)
  boost state support:
    Supported: yes
    Active: yes
    AMD PSTATE Highest Performance: 166. Maximum Frequency: 4.46 GHz.
    AMD PSTATE Nominal Performance: 119. Nominal Frequency: 3.20 GHz.
    AMD PSTATE Lowest Non-linear Performance: 41. Lowest Non-linear Frequency: 1.10 GHz.
    AMD PSTATE Lowest Performance: 15. Lowest Frequency: 400 MHz.

Ja4e, 2024-09-29:
As you can see the current policy's frequency are set to 400 MHz and 400 MHz.

linrunner, 2024-09-29:
Hi, 

what you observe is the designed behaviour of the kernel driver _amd-pstate-epp_  with the settings _you_ specified. TLP is not responsible for this.

I suggest you use a less agressive EPP setting than this
> /etc/tlp.conf L0147: CPU_ENERGY_PERF_POLICY_ON_BAT="**power**"

By the way, _amd-pstate-epp_ ignores all frequency settings. It doesn't matter what you set there. This is explained in the docs https://linrunner.de/tlp/settings/processor.html#cpu-driver-opmode-on-ac-bat
> In active mode, the processor selects the operating frequencies autonomously within the limits imposed by the hardware, including turbo boost. The powersave governor can also lead to max frequency. **Configuring the min and max frequency (as shown below) is not supported.**

Please ask further questions about which settings to choose and the _amd_pstate_ driver in the forum of your distribution (or in another Linux forum of your choice). The issue tracker is reserved for *TLP* bug reports. Thanks for your understanding.

closed 2024-09-29 by linrunner

Ja4e, 2024-09-30:
if i set iCPU_DRIVER_OPMODE_ON_BAT="guided" or "passive", it does not work regardless having PU_ENERGY_PERF_POLICY_ON_BAT on any settings. I will try to seek out arch linux forums thankyou!