linrunner/TLP #765: [Framework] Battery care support based on the cros_charge-control driver (as of Linux 6.10)

linrunner, 2024-10-04:
Based on the [preliminary work here](https://github.com/linrunner/TLP/pull/728), here is a rewrite of the plugin requiring the new driver _cros_charge-control_ released with Linux 6.10.

For the test you either need TLP 1.7.0, supplemented by the download of **[/usr/share/tlp/bat.d/55-framework](https://raw.githubusercontent.com/linrunner/TLP/refs/heads/feature/bat.d/framework/bat.d/55-framework)**, or you can build a package for your distribution based on Branch [feature/bat.d/framework](https://github.com/linrunner/TLP/tree/feature/bat.d/framework).

I need at least the output of 

    sudo tlp-stat -s -b 

Please also run the automatic test script (with sudo or as root) and show the output:
**[unit-tests/charge-thresholds_framework](https://github.com/linrunner/TLP/blob/feature/bat.d/framework/unit-tests/charge-thresholds_framework)**

It requires the tool **[clitest](https://github.com/aureliojargas/clitest)** to be installed. Should be available as a package in most distributions. 

Reference: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/power/supply/cros_charge-control.c

@t-8ch @Simerax @chutz

t-8ch, 2024-10-05:
* The module is from 6.11
* The vendor check actually reads the model_name, so always fails.

With this fixed:

```
# tlp-stat -s -b
--- TLP 1.7.0 --------------------------------------------

+++ System Info
System         = Framework A7 Laptop 13 (AMD Ryzen 7040Series)
BIOS           = 03.05
OS Release     = Arch Linux
Kernel         = 6.12.0-rc1-00312-g798df27d5d3f #12 SMP PREEMPT_DYNAMIC Sat Oct  5 09:18:53 CEST 2024 x86_64
/proc/cmdline  = initrd=\initramfs-linux-upstream.img rw cryptdevice=/dev/nvme0n1p3:system:discard root=/dev/mapper/system
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle]

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 09:36:13, 420 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: framework
Supported features: charge thresholds
Driver usage:
* natacpi (cros_charge_control) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = NVT
/sys/class/power_supply/BAT1/model_name                     = FRANGWA
/sys/class/power_supply/BAT1/cycle_count                    =     38
/sys/class/power_supply/BAT1/charge_full_design             =   3915 [mWh] *
/sys/class/power_supply/BAT1/charge_full                    =   3522 [mWh] *
/sys/class/power_supply/BAT1/charge_now                     =   3522 [mWh] *
/sys/class/power_supply/BAT1/current_now                    =      0 [mW] *
/sys/class/power_supply/BAT1/status                         = Not charging

/sys/class/power_supply/BAT1/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT1/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =  100.0 [%]
Capacity                                                    =   90.0 [%]
```

```
[~/src/tlp]$ sudo clitest unit-tests/charge-thresholds_framework
#1	# +++ Framework laptops +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2	#
#3	# --- tlp start
#4	sudo tlp start -- START_CHARGE_THRESH_BAT1= STOP_CHARGE_THRESH_BAT1= START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#5	sudo tlp start -- START_CHARGE_THRESH_BAT1="60" STOP_CHARGE_THRESH_BAT1="100" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#6	sudo tlp start -- START_CHARGE_THRESH_BAT1="100" STOP_CHARGE_THRESH_BAT1="100" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#7	sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="0" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#8	sudo tlp start -- START_CHARGE_THRESH_BAT1="0" STOP_CHARGE_THRESH_BAT1="101" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#9	sudo tlp start -- START_CHARGE_THRESH_BAT1="97" STOP_CHARGE_THRESH_BAT1="97" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#10	sudo tlp start -- START_CHARGE_THRESH_BAT1="95" STOP_CHARGE_THRESH_BAT1="96" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#11	sudo tlp start -- START_CHARGE_THRESH_BAT1="DEF" STOP_CHARGE_THRESH_BAT1="DEF" START_CHARGE_THRESH_BAT0= STOP_CHARGE_THRESH_BAT0=
#12	sudo tlp start -- NATACPI_ENABLE=0 START_CHARGE_THRESH_BAT1="DEF" STOP_CHARGE_THRESH_BAT1="DEF"
#13	#
#14	# --- tlp setcharge w/o arguments
#15	sudo tlp setcharge -- START_CHARGE_THRESH_BAT1="60" STOP_CHARGE_THRESH_BAT1="100" X_SOC_CHECK=0
[... 291 lines cut ...]
-/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge
+* natacpi (cros_charge_control) = active (charge thresholds)
+/sys/class/power_supply/BAT1/charge_control_start_threshold =      0 [%]
+/sys/class/power_supply/BAT1/charge_control_end_threshold   =      3 [%]
+/sys/class/power_supply/BAT1/charge_behaviour               = [auto] inhibit-charge force-discharge
----------------------------------------------------------------------------------------------------------------------------------------------------------------
#45	sudo tlp-stat -b -- X_BAT_PLUGIN_SIMULATE=framework X_THRESH_SIMULATE_READERR=1 | grep -E 'charge_(control|behaviour)'
----------------------------------------------------------------------------------------------------------------------------------------------------------------
[FAILED #45, line 110] sudo tlp-stat -b -- X_BAT_PLUGIN_SIMULATE=framework X_THRESH_SIMULATE_READERR=1 | grep -E 'charge_(control|behaviour)'
@@ -1,3 +1,4 @@
-/sys/class/power_supply/BAT0/charge_control_start_threshold = (not available) [%]
-/sys/class/power_supply/BAT0/charge_control_end_threshold   = (not available) [%]
-/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge
+* natacpi (cros_charge_control) = active (charge thresholds)
+/sys/class/power_supply/BAT1/charge_control_start_threshold = (not available) [%]
+/sys/class/power_supply/BAT1/charge_control_end_threshold   = (not available) [%]
+/sys/class/power_supply/BAT1/charge_behaviour               = [auto] inhibit-charge force-discharge
----------------------------------------------------------------------------------------------------------------------------------------------------------------
#46	#
#47	# --- Reset test machine to configured thresholds
#48	sudo tlp setcharge BAT0  > /dev/null 2>&1
#49	#

FAIL: 29 of 49 tests failed
```

linrunner, 2024-10-05:
>  * The vendor check actually reads the model_name, so always fails.

Thanks, I now check if **/sys/class/dmi/id/sys_vendor** is "Framework".

> [~/src/tlp]$ sudo clitest **unit-tests/charge-thresholds_framework**

Improved. Please test again.

> [~/src/tlp]$ sudo clitest **unit-tests/charge-thresholds_framework-simulate** 

Oh, no! Do not  run this: It only works as a simulation on a ThinkPad (for my dev qa) not on real Framework hardware ...

The fact that  the mAh/mA sysfiles read plausible values is new to me:
> /sys/class/power_supply/BAT1/charge_full_design             =   3915 [mWh] *
> /sys/class/power_supply/BAT1/charge_full                    =   3522 [mWh] * [...]

I only know of implausible values that are presumably to be interpreted as mWh/mW, see [here](https://github.com/linrunner/TLP/pull/728#issuecomment-2185173431). 

> +++ System Info
> System         = Framework AA Laptop
> BIOS           = 03.17
> 
> ++ Battery Care
> 
> +++ Battery Status: BAT1
> /sys/class/power_supply/BAT1/manufacturer                   = NVT
> /sys/class/power_supply/BAT1/model_name                     = Framewo
> /sys/class/power_supply/BAT1/charge_full_design             =  35720 [mAh]
> /sys/class/power_supply/BAT1/charge_full                    =  31970 [mAh]
> /sys/class/power_supply/BAT1/charge_now                     =  14530 [mAh]
> /sys/class/power_supply/BAT1/current_now                    =      1 [mA]
> 

A well-known coreboot disease. Doesn't seem to be the case with your model then, I have removed the coreboot workaround.

Please show 

    sudo tlp-stat -s -b -v

again.

t-8ch, 2024-10-06:
```
$ sudo clitest unit-tests/charge-thresholds_framework
#1	# +++ Framework laptops ++++++++++++++++++++++++++++++++++++++++++++++++++++++++
...
OK: 48 of 48 tests passed
```

```
$ sudo tlp-stat -s -b -v                           
--- TLP 1.7.0 --------------------------------------------

+++ System Info
System         = Framework A7 Laptop 13 (AMD Ryzen 7040Series)
BIOS           = 03.05
OS Release     = Arch Linux
Kernel         = 6.12.0-rc1-00308-g695f7def8da4 #14 SMP PREEMPT_DYNAMIC Sat Oct  5 10:06:10 CEST 2024 x86_64
/proc/cmdline  = initrd=\initramfs-linux-upstream.img rw cryptdevice=/dev/nvme0n1p3:system:discard root=/dev/mapper/system
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle]

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 15:51:02, 52 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: framework
Supported features: charge thresholds
Driver usage:
* natacpi (cros_charge_control) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = NVT
/sys/class/power_supply/BAT1/model_name                     = FRANGWA
/sys/class/power_supply/BAT1/cycle_count                    =     38
/sys/class/power_supply/BAT1/charge_full_design             =   3915 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   3483 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =    844 [mAh]
/sys/class/power_supply/BAT1/current_now                    =   2438 [mA]
/sys/class/power_supply/BAT1/status                         = Charging

/sys/class/power_supply/BAT1/voltage_min_design             =  15480 [mV]
/sys/class/power_supply/BAT1/voltage_now                    =  16204 [mV]

/sys/class/power_supply/BAT1/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT1/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   24.2 [%]
Capacity                                                    =   89.0 [%]
```

> A well-known coreboot disease. Doesn't seem to be the case with your model then, I have removed the coreboot workaround.

Only the Framework Chromebook uses coreboot.
That's not the case for my machine.

linrunner, 2024-10-06:
@t-8ch LGTM :-). Thank you for testing.

linrunner, 2024-10-11:
@t-8ch I have finished the second stage of the plugin with discharge/recalibrate. Please test it with

     sudo tlp discharge

You can force an earlier end to the discharging process by specifying a certain charge level (in mAh) with the debug parameter `X_BAT_DISCHG_END`. Example:

     sudo tlp discharge -- X_BAT_DISCHG_END=3000

I would  also like to ask for your feedback on the safety message:

> **Caution: discharge/recalibrate operation is blocked for safety reasons.**
>
> **The Framework EC firmware will not stop until the battery is completely discharged.**
> **This will cause the laptop to suddenly power off, which can lead to data loss and data corruption.**
> **TLP will stop discharging *before* this happens, but it can't respond to unforeseen battery behavior.**
>
> Add FRAMEWORK_UNBLOCK_DISCHARGE=1 to your configuration to accept the risk and unblock
> discharge/recalibrate. Make sure all data is saved and backups are in place.

linrunner, 2024-10-16:
@t-8ch I converted the plugin from a framework-specific to a generic one for all laptops with ChromeOS EC. See 1st post.

The automatic test script moves to **[unit-tests/charge-thresholds_chromeos-ec](https://github.com/linrunner/TLP/blob/feature/bat.d/framework/unit-tests/charge-thresholds_chromeos-ec)**

MithicSpirit, 2024-10-21:
Doesn't seem to be working on my Framework 16, even though the `cros_charge_control` kmod is loaded.
```
$ sudo tlp-stat -s -b -v
--- TLP 1.8.0-alpha.0 --------------------------------------------

+++ System Info
System         = Framework AG Laptop 16 (AMD Ryzen 7040 Series)
BIOS           = 03.03
OS Release     = NixOS 24.11 (Vicuna)
Kernel         = 6.11.3-xanmod1 #1-NixOS SMP PREEMPT_DYNAMIC Tue Jan  1 00:00:00 UTC 1980 x86_64
/proc/cmdline  = init=/nix/store/xic0jg2pw0d635v9x4f1xhpvqmvbm84h-nixos-system-hipparchus-24.11.20241018.4c2fcb0/init amd_pstate=active root=fstab loglevel=4 vt.default_red=0x29,0xbf,0xa3,0xeb,0x81,0xb4,0x88,0xec,0x43,0xcb,0xb2,0xee,0x9a,0xc0,0x99,0xe5 vt.default_grn=0x2e,0x61,0xbe,0xcb,0xa1,0x8e,0xc0,0xef,0x4c,0x8d,0xcc,0xd5,0xb3,0xa5,0xce,0xe9 vt.default_blu=0x38,0x6a,0x8c,0x8b,0xc1,0xad,0xd0,0xf4,0x5e,0x92,0x9c,0xaa,0xce,0xbb,0xdd,0xf0
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 03:53:31 PM, 314 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: chromeos-ec
Supported features: none available
Driver usage:
* natacpi (cros_charge_control) = inactive (laptop not supported, incompatible firmware)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = NVT
/sys/class/power_supply/BAT1/model_name                     = FRANDBA
/sys/class/power_supply/BAT1/cycle_count                    =     28
/sys/class/power_supply/BAT1/charge_full_design             =   5491 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   5583 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   4748 [mAh]
/sys/class/power_supply/BAT1/current_now                    =      0 [mA]
/sys/class/power_supply/BAT1/status                         = Charging

/sys/class/power_supply/BAT1/voltage_min_design             =  15480 [mV]
/sys/class/power_supply/BAT1/voltage_now                    =  16990 [mV]

Charge                                                      =   85.0 [%]
Capacity                                                    =  101.7 [%]
$ lsmod | grep cros_charge_control
cros_charge_control    16384  0
battery                28672  2 cros_charge_control,framework_laptop
```

linrunner, 2024-10-22:
@MithicSpirit please show the output of

    grep . /sys/class/power_supply/BAT1/*

MithicSpirit, 2024-10-22:
```
$ grep . /sys/class/power_supply/BAT1/*
/sys/class/power_supply/BAT1/alarm:550000
/sys/class/power_supply/BAT1/capacity:85
/sys/class/power_supply/BAT1/capacity_level:Normal
/sys/class/power_supply/BAT1/charge_control_end_threshold:85
/sys/class/power_supply/BAT1/charge_full:5595000
/sys/class/power_supply/BAT1/charge_full_design:5491000
/sys/class/power_supply/BAT1/charge_now:4759000
/sys/class/power_supply/BAT1/current_now:0
/sys/class/power_supply/BAT1/cycle_count:28
grep: /sys/class/power_supply/BAT1/device: Is a directory
grep: /sys/class/power_supply/BAT1/hwmon3: Is a directory
/sys/class/power_supply/BAT1/manufacturer:NVT
/sys/class/power_supply/BAT1/model_name:FRANDBA
grep: /sys/class/power_supply/BAT1/power: Is a directory
/sys/class/power_supply/BAT1/present:1
/sys/class/power_supply/BAT1/serial_number:0342
/sys/class/power_supply/BAT1/status:Charging
grep: /sys/class/power_supply/BAT1/subsystem: Is a directory
/sys/class/power_supply/BAT1/technology:Li-ion
/sys/class/power_supply/BAT1/type:Battery
/sys/class/power_supply/BAT1/uevent:DEVTYPE=power_supply
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_NAME=BAT1
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_TYPE=Battery
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_STATUS=Charging
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_PRESENT=1
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_TECHNOLOGY=Li-ion
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CYCLE_COUNT=28
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_VOLTAGE_MIN_DESIGN=15480000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_VOLTAGE_NOW=16996000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CURRENT_NOW=0
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CHARGE_FULL_DESIGN=5491000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CHARGE_FULL=5595000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CHARGE_NOW=4759000
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CAPACITY=85
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_CAPACITY_LEVEL=Normal
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_MODEL_NAME=FRANDBA
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_MANUFACTURER=NVT
/sys/class/power_supply/BAT1/uevent:POWER_SUPPLY_SERIAL_NUMBER=0342
/sys/class/power_supply/BAT1/voltage_min_design:15480000
/sys/class/power_supply/BAT1/voltage_now:16996000
```

linrunner, 2024-10-22:
> /sys/class/power_supply/BAT1/charge_control_end_threshold:85

The out-of-tree module _framework_laptop_ is loaded and has moved in front of _cros_charge-control_. Unfortunately, atm I don't know how to prevent this. Try to unload _framework_laptop_ and _cros_charge-control_, then load _cros_charge-control_ only.

linrunner, 2024-10-25:
@MithicSpirit I have now restored the original plugin for the _framework_laptop_ module which support only a stop threshold. You need to install **[/usr/share/tlp/bat.d/55-chromeos-ec](https://github.com/linrunner/TLP/blob/feature/bat.d/framework/bat.d/55-chromeos-ec)** and  **[/usr/share/tlp/bat.d/56-framework](https://github.com/linrunner/TLP/blob/feature/bat.d/framework/bat.d/56-framework)**. Then show

     sudo tlp-stat -s -b -v

MithicSpirit, 2024-10-25:
Seems to be working now, although if this is just to support a charge limit, I'm not sure what's the point. It was already able to do that beforehand.
```
$ sudo tlp-stat -s -b -v
--- TLP 1.8.0-alpha.0 --------------------------------------------

+++ System Info
System         = Framework AG Laptop 16 (AMD Ryzen 7040 Series)
BIOS           = 03.03
OS Release     = NixOS 24.11 (Vicuna)
Kernel         = 6.11.3-xanmod1 #1-NixOS SMP PREEMPT_DYNAMIC Tue Jan  1 00:00:00 UTC 1980 x86_64
/proc/cmdline  = init=/nix/store/2cvwm5nzb7gs2sinsjy7a44ypkl6rc0c-nixos-system-hipparchus-24.11.20241020.1997e4a/init amd_pstate=active root=fstab loglevel=4 vt.default_red=0x29,0xbf,0xa3,0xeb,0x81,0xb4,0x88,0xec,0x43,0xcb,0xb2,0xee,0x9a,0xc0,0x99,0xe5 vt.default_grn=0x2e,0x61,0xbe,0xcb,0xa1,0x8e,0xc0,0xef,0x4c,0x8d,0xcc,0xd5,0xb3,0xa5,0xce,0xe9 vt.default_blu=0x38,0x6a,0x8c,0x8b,0xc1,0xad,0xd0,0xf4,0x5e,0x92,0x9c,0xaa,0xce,0xbb,0xdd,0xf0
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 10:03:07 AM, 18 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: framework
Supported features: charge threshold
Driver usage:
* natacpi (framework_laptop) = active (charge thresholds)
Parameter value ranges:
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = NVT
/sys/class/power_supply/BAT1/model_name                     = FRANDBA
/sys/class/power_supply/BAT1/cycle_count                    =     28
/sys/class/power_supply/BAT1/charge_full_design             =   5491 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   5505 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   4674 [mAh]
/sys/class/power_supply/BAT1/current_now                    =   1055 [mA]
/sys/class/power_supply/BAT1/status                         = Discharging

/sys/class/power_supply/BAT1/voltage_min_design             =  15480 [mV]
/sys/class/power_supply/BAT1/voltage_now                    =  16579 [mV]

/sys/class/power_supply/BAT1/charge_control_end_threshold   =     80 [%]

Charge                                                      =   84.9 [%]
Capacity                                                    =  100.3 [%]
```

linrunner, 2024-10-25:
@MithicSpirit lhe **framework** plugin is intended as a fallback in case you need _framework_laptop_ for reasons other than battery care. I would be happy if you could still blacklist _framework_laptop_ and then test the extended functionality of the **chromeos-ec** plugin.

linrunner, 2024-12-31:
The plugin has been fundamentally revised. **A kernel update to 6.12.8 or 6.13-rc5 is required.** 

Please follow the updated instructions in [post 1](https://github.com/linrunner/TLP/issues/765#issue-2566394649).

linrunner, 2025-01-11:
Still looking for testers. For new packages and updated instructions see [Post 1( https://github.com/linrunner/TLP/issues/765#issue-2566394649).

linrunner, 2025-01-13:
For reference: successful test on [ThinkPad C13 Yoga Gen 1 Chromebook](https://psref.lenovo.com/Product/ThinkPad/ThinkPad_C13_Yoga_Gen_1_Chromebook) documented here
https://thinkpad-forum.de/threads/c13-yoga-gen-1-chromebook-%E2%80%93-tipps-erfahrungen.240712/page-2#post-2415521

linrunner, 2025-01-19:
TLP 1.8.0 Beta 1 has been released. Please test again if it works for you -> https://github.com/linrunner/TLP/issues/781

linrunner, 2025-02-13:
I have just released [TLP 1.8.0](https://github.com/linrunner/TLP/releases/tag/1.8.0). Thanks for your contribution. Have fun!

closed 2025-02-13 by linrunner