linrunner/TLP #766: Bluetooth always turned on at startuo no matter the settings

krpknstn, 2024-10-05:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Bluetooth is always enabled on startup. It shouldn't be as per tlp.config as it is set to disabled on startup. It can be manually disabled without problem.

**Expected behavior**

Bluetooth should be disabled at startup if configured to be diasabled.

**To Reproduce**

Have Ubuntu 24.04.1 LTS
Remove power-profiles-demon
Install TLP from the repository
Start TLP
Make sure that:
#DEVICES_TO_DISABLE_ON_STARTUP="bluetooth nfc wifi wwan"
#DEVICES_TO_ENABLE_ON_STARTUP="wifi"
It doesn't matter if:
#RESTORE_DEVICE_STATE_ON_STARTUP is set to 1 or 0, it turns on either way.
Turn of Bluetooth
Reboot
Bluetooth is turned on, this shouldn't be.

1. Does the problem occur on battery or AC or both? BOTH
2. Actions to reproduce the behaviour: See above
3. Shell commands entered and their output *including error message(s)*: no error messages
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots):
  
<script src="https://gist.github.com/krpknstn/a58b8ed5e8c041397a071a76b4c28c49.js"></script>
<script src="https://gist.github.com/krpknstn/f2d2897a653a6ed9811869c6abcacf6f.js"></script>

krpknstn, 2024-10-05:
I'm really sorry. Please delete this issue I found the solution one minute after posting it after trying for two hours:

As stated in the FAQ (yes I know) an applet can enable Bluetooth. In case of Ubuntu you need to change "AutoEnable=false"in /etc/bluetooth/main.con.

It might be helpful to state this explicitly since Ubuntu is so common, but I understand of you like it as short as it is right now! Thank you for making this great tool!

closed 2024-10-05 by krpknstn

linrunner, 2024-10-05:
> #DEVICES_TO_DISABLE_ON_STARTUP="bluetooth nfc wifi wwan"
> #DEVICES_TO_ENABLE_ON_STARTUP="wifi"
> #RESTORE_DEVICE_STATE_ON_STARTUP is set to 1 or 0, it turns on either way.

Did you remember to remove the leading #'s? Otherwise your settings won't be activated.

krpknstn, 2024-10-05:
I'm even more of an idiot, this also now works.
Thank you very much!