linrunner/TLP #768: TLP wrongly running in battery mode on desktop PC

alienth, 2024-10-29:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

TLP believes I'm running on battery and is enacting battery-mode settings as a result. My desktop PC has no battery. This _might_ be due to my mouse having a battery, which it is detecting.

Initially discovered because my hard disk was spinning down every 15 seconds.

tlp-stat output: https://gist.github.com/alienth/60f3b53004d7849f1906241ec1868550
tlp-stat --psup output: https://gist.github.com/alienth/55ff4f906b5fd1d3e94f1b7ae8e7e819

**Expected behavior**

TLP should not be detecting me in battery mode when my PC has no battery.

*"Works fine" is not enough to analyze the problem!*

**To Reproduce**

Unclear how to reproduce. Probably a combination of specific hardware and my kernel version.

**Additional context**

Running Ubuntu Oracular, kernel 6.11.0-9-generic, TLP

alienth, 2024-10-29:
Looks similar to #313, which was fixed in 1.1. Maybe a regression but I haven't checked to be sure.

alienth, 2024-10-29:
This is not a regression of #313 - I do happen to have a mouse with a battery, but TLP is rightly excluding that.

What it is reading is the UCSI "power supply", which I'm guessing happens to exist due to the USB-C ports on my motherboard. The `type` of this power supply is `USB`, and its online state is `0`.

```
/sys/class/power_supply/ucsi-source-psy-0-00081/type:USB
/sys/class/power_supply/ucsi-source-psy-0-00081/online:0
```

Because of this, `ac0seen` gets set to `1`, which results in `get_sys_power_supply` returning `1`, which erroneously indicates the system is running on a "battery".

https://github.com/linrunner/TLP/blob/7ee67a0fa97fcb3b2e5b67469aff2f3830e39c84/tlp-func-base.in#L874
https://github.com/linrunner/TLP/blob/7ee67a0fa97fcb3b2e5b67469aff2f3830e39c84/tlp-func-base.in#L933-L935

alienth, 2024-10-29:
This is fixable by setting `TLP_PS_IGNORE` to `USB`. Wondering if there is some way the logic could be adjusted to handle this correctly by default, though. It would be easily for someone to accidentally end up in this state, leading to performance problems plaguing them with no clear cause.

alienth, 2024-10-29:
Also filed a bug with Ubuntu: https://bugs.launchpad.net/ubuntu/+source/tlp/+bug/2085812

Oracular installs TLP by default, so anyone on a desktop having a UCSI device is likely to be bitten by this.

alienth, 2024-10-29:
Somewhat related symptom here with some discussion of the ucsi-psy driver: https://lore.kernel.org/all/20201110101519.GC1224435@kuha.fi.intel.com/T/

linrunner, 2024-10-29:
> Wondering if there is some way the logic could be adjusted to handle this correctly by default, though.

Unfortunately I am not aware of any magical heuristics that gets along with all the faulty or incomplete ACPI BIOS implementations which cause power source detection problems. That is why `TLP_PS_IGNORE` remains the method of choice. However, I will gladly accept pull requests ...

A correct BIOS implementation would provide this (via the kernel):
```
/sys/class/power_supply/AC/type: Mains
/sys/class/power_supply/AC/online: 1

```

alienth, 2024-10-30:
@linrunner I take it that it would not safe to default to "assume powered by mains" if no mains power supplies are defined?

Looking through the code I'm guessing there are a lot of Chesterton's Fences that have been built up over the years accommodating for various configurations.

alienth, 2024-10-30:
Going to pose a question on the Ubuntu side about maybe not installing TLP unless it can be assured that the system relies upon a battery.

MoonSweep, 2024-10-30:
Hi, I'm the maintainer for the TLP package on Debian.

IMHO the fault is on `ubuntu-mate-desktop` for installing the package unconditionally. As the main developer said, it should not be installed on desktops.

The distribution installer is usually able to detect if the installation is performed on a laptop or a desktop (see for example the [`laptop-detect`](https://tracker.debian.org/pkg/laptop-detect) Debian package (main script's source code [here](https://sources.debian.org/src/laptop-detect/latest/laptop-detect.in)), which is run at install time, and only then, installs laptop-specific packages, if and only if the installer thinks they're needed.

That being said, TLP could also do a simple check at start, and gracefully exit if the system is not a laptop (with some kind of `force` option and/or configuration directive for users who know what they're doing).

For example, here's a simplified form of one of the tests performed by `laptop-detect` (8 = Portable, 9 = Laptop, 10 = Notebook, 11 = handheld). It doesn't need any dependency, just pure shell and a not too old kernel :

```shell
laptop_check () {
  local SYSFILE
  if [ $FORCE -ne 1 ] ; then
    SYSFILE="/sys/devices/virtual/dmi/id/chassis_type"
    if [ -e "$SYSFILE" ] ; then
      case "$(cat "$SYSFILE")" in
        8|9|10|11) true ;;
        *) printf "We're not on a laptop, exiting gracefully\n" ; exit 0 ;;
      esac
    fi
  fi
}
```

But again, if not on a laptop, a package such as TLP shouldn't be installed at all.

Now I know that Ubuntu likes to patch stuff up to suit their needs (that's why their TLP package diverts from mine and from upstream), so they may take this code snippet and do whatever they want with it.

Or, upstream could add such a fail-safe to avoid time-consuming bug reports like this one, caused by careless package maintainers :expressionless:)

alienth, 2024-10-31:
> IMHO the fault is on ubuntu-mate-desktop for installing the package unconditionally. As the main developer said, it should not be installed on desktops.

Yeah, after the discussion here I'm in agreement. I'll mark this closed and see how I might interface with the `ubuntu-mate-desktop` maintainers. Thanks for the discussion, all!

closed 2024-10-31 by alienth

linrunner, 2024-10-31:
Hi Raphaël,

> IMHO the fault is on `ubuntu-mate-desktop` for installing the package unconditionally. As the main developer said, it should not be installed on desktops.

Exactly.

> That being said, TLP could also do a simple check at start, and gracefully exit if the system is not a laptop (with some kind of `force` option and/or configuration directive for users who know what they're doing).

Please forget this suggestion and never use it again. TLP does not have a single entry point where it is started.

MoonSweep, 2024-10-31:
> Please forget this suggestion and never use it again. TLP does not have a single entry point where it is started.

Noted :smiley:

linrunner, 2025-02-13:
I have just released [TLP 1.8.0](https://github.com/linrunner/TLP/releases/tag/1.8.0). Thanks again for your contribution. Have fun!