linrunner/TLP #770: Unable to set charge limits on Thinkpad T410 on MX Linux

beesyrupfactory, 2024-11-17:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I am unable to set my desired charge limit on my Thinkpad T410 on MX Linux. When I try to set `sudo tlp setcharge 0 80` I get the following error:
`Error: charge start threshold (0) for BAT0 is not specified, invalid or out of range (2..96). Aborted.`

**Expected behavior**
As I have installed of the prerequisite packages, I should be able to set a charge limit: `tlp tlp-rdw tp-smapi-dkms`.

In actuality, my battery is charging to 100%.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? This happens when 
2. Actions to reproduce the behaviour. 
a. Set the following in `/etc/tlp.conf`:
```
START_CHARGE_THRESH_BAT0=75
STOP_CHARGE_THRESH_BAT0=80
```
b. Check `sudo tlp-stat -b`.
c. Assert that the charge limit isn't being honored
```
Resulting output of running `tlp-stat -b` located in [this gist](https://gist.github.com/beesyrupfactory/753c2cc5d198b99ddca1105e7d8962e1)

3. Shell commands entered and their output *including error message(s)*
 `sudo tlp setcharge 0 80` I get the following error:
`Error: charge start threshold (0) for BAT0 is not specified, invalid or out of range (2..96). Aborted.`

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
Complete output of tlp-stat located in [this gist](https://gist.github.com/beesyrupfactory/99e754395f691bb34a48db91e74e0de0)

**Additional context**
This is for a third party battery, but the same behavior happened against an official battery as well.

If there is anything else that is needed from my side, I will be happy to assist.

linrunner, 2024-11-17:
Your config with start = 75 and stop = 80 looks okay. These are applied on boot - **if the tlp service is enabled**.

It makes no sense to try to set a start threshold of zero with your command
> sudo tlp setcharge 0 80

A value of zero is not supported by your hardware and the tp-smapi driver and TLP rejects it. Works as designed. The tlp-stat outputs tells you the actual limits:
> Parameter value ranges:
> * START_CHARGE_THRESH_BAT0/1:  **2..96**(default)
> * STOP_CHARGE_THRESH_BAT0/1:   **6..100**(default) 

So what happens when you just apply the configured thresholds with

    sudo tlp setcharge
    sudo tlp-stat -b

?

linrunner, 2024-11-17:
The thresholds are now set by TLP. If your hardware does not respect them, this would not be a TLP but a hardware issue.

Please remember make sure that the TLP service is enabled. If I remember correctly, before systemd you have to check the status with 
 
    sudo service tlp status

beesyrupfactory, 2024-11-17:
Thanks so much for your support. There seems to be a problem with the BMS on this battery, so I'll give it a try with my known working one.

```
$ sudo service tlp status
--- TLP 1.5.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad T410 252237J
BIOS           = 6IET85WW (1.45 )
OS Release     = Debian GNU/Linux 12 (bookworm)
Kernel         = 6.1.0-27-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.115-1 (2024-11-01) x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-6.1.0-27-amd64 root=UUID=70ee097d-5c1f-4f30-9709-4b9f29813de7 ro quiet splash resume=UUID=70ee097d-5c1f-4f30-9709-4b9f29813de7 resume_offset=29550592
Init system    = sysvinit
Boot mode      = BIOS (CSM, Legacy)

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 01:56:14 AM,    495 sec(s) ago
Mode           = AC
Power source   = AC
```

I'll close this for now as this was on me. The problem was between the laptop and the chair :smile:

closed 2024-11-17 by beesyrupfactory

beesyrupfactory, 2024-11-19:
Just wanted to send an update and let you know that it's working flawlessly. Thank you for your patience with me @linrunner. I appreciate you.