linrunner/TLP #771: Error: writing charge start threshold for BAT0 failed.

keysteal72, 2024-11-18:
I've recenly upgraded my Fedora to 41 and kernel at 6.11.7-300.fc41. Since that time my TLP service has an error "Writing charge start threshold for BAT0 failed".In fact "charge_control_start_threshold file" is read only, but even if i change it to be writable, it doesn't work. I've an old Macbook PRO 16".

Any idea?

--- TLP 1.7.0 --------------------------------------------

+++ Battery Care
Plugin: APPLESMC
Supported features: charge thresholds
Driver usage:
* natacpi (APPLESMC_MOD) = active (charge thresholds)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = ASMARK
/sys/class/power_supply/BAT0/model_name                     = A1493
/sys/class/power_supply/BAT0/cycle_count                    =    276
/sys/class/power_supply/BAT0/charge_full_design             =   7000 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   6233 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   6220 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Full

/sys/class/power_supply/BAT0/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
/sys/class/power_supply/BAT0/charge_control_full_threshold  =    100 [%]

Charge                                                      =   99.8 [%]
Capacity                                                    =   89.0 [%]

linrunner, 2024-11-18:
Hi, the plugin whose output you are showing is not part of an official TLP release and is not included in any current development branch. You have to deal with this yourself -> **no support for it**. 

By the way, there is also no 1.7.0 package in Fedora.

You are welcome to use the official 1.7.0, which contains a tested Macbook plugin - if you can build a correct package yourself. 

The corresponding documentation will also tell you that the start threshold is not freely selectable, but start = stop - 5.
https://linrunner.de/tlp/settings/bc-vendors.html#apple-macbooks

closed 2024-11-18 by linrunner

linrunner, 2024-11-24:
ps. for Fedora you may as well use the beta 2 packages -> https://download.linrunner.de/packages/