linrunner/TLP #773: Chargeonce text output issue

thomas2net, 2024-12-05:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

This is a user interface issue.
I have thresholds:
charge_control_start_threshold =     79 [%]
charge_control_end_threshold   =     80 [%]
My battery level is at 60%, connected to power supply, and I would like to charge back to 80% (which I had configured with setcharge 60 80).
So I execute: sudo tlp chargeonce
Output (confusing!):
"Setting temporary charge threshold for BAT0:
  start =  79
Charging starts now, keep AC connected."

That's not what I would have expected in plain English.
Charging does not actually start!

**Expected behavior**

I should have read:
"Your current battery level is 60%. 
We are about to charge to a level of 80%.
You are using a Thinkpad X1. Please briefly unplug and replug your laptop from the power adapter for charging to start."

*"Works fine" is not enough to analyze the problem!*

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
2. Actions to reproduce the behaviour
3. Shell commands entered and their output *including error message(s)*
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

**Additional context**

Add any other context about the problem here.
--- TLP 1.6.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0006: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0009: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0010: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0011: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0013: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0014: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0015: NMI_WATCHDOG="0"
defaults.conf L0016: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0017: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0018: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0019: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0020: DISK_IOSCHED="keep keep"
defaults.conf L0021: SATA_LINKPWR_ON_AC="med_power_with_dipm"
defaults.conf L0022: SATA_LINKPWR_ON_BAT="med_power_with_dipm"
defaults.conf L0023: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0025: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0026: PCIE_ASPM_ON_AC="default"
defaults.conf L0027: PCIE_ASPM_ON_BAT="default"
defaults.conf L0028: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0030: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_BAT="default"
[... 188 lines cut ...]
+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = Sunwoda
/sys/class/power_supply/BAT0/model_name                     = 5B10W13975
/sys/class/power_supply/BAT0/cycle_count                    =     97
/sys/class/power_supply/BAT0/energy_full_design             =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  56990 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  40010 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  42097 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     79 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   70.2 [%]
Capacity                                                    =  100.0 [%]

linrunner, 2024-12-08:
I tried to reproduce this: my X1 Carbon Gen9 with identical BIOS and EC version starts charging immediately after tlp chargeonce.

Over the years, TLP users have reported a lot of quirks in Lenovo's EC firmware. Trying to differentiate all this in TLP is pointless, because it would result in high technical debt. No thanks.

What I can offer you is a general note in the output, something like 

> Charging starts now, keep AC connected. 
> Note: If charging does not start, disconnect the charger for a short time and then reconnect it.

thomas2net, 2024-12-09:
Your offer sounds good to me, Linrunner. Many thanks.
I suggest, however, to explicitly recite in the output message the current detected charge percentage and the specified end of charge percentage.

linrunner, 2025-01-12:
Done. Thanks for your suggestion.

Please test and report. Get packages from here: https://download.linrunner.de/packages/

linrunner, 2025-01-19:
@thomas2net TLP 1.8.0 Beta 1 has been released. Please test again if it works for you -> https://github.com/linrunner/TLP/issues/781

closed 2025-01-26 by linrunner

linrunner, 2025-02-13:
I have just released [TLP 1.8.0](https://github.com/linrunner/TLP/releases/tag/1.8.0). Thanks again for your contribution. Have fun!