linrunner/TLP #775: When waking from suspend does not follow max frequency.

user12812, 2024-12-09:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)
https://gist.github.com/user12812/a54cb30619182e82ffc199bd27f7882c

**Describe the bug**

When waking from suspend on BAT and AC mode it disables turbo, and does not follow the set CPU frequency for BAT mode.
for example;
laptop is away from wall power and is on BAT profile, which limits CPU clocks down to 1200 MHz and does not allow to boost higher, when I wake from suspend it removes the clock speed limiter and is able to boost up to ~2800 MHz.

when laptop is on wall power and is on AC profile, which has no such limits and has turbo enabled, when woken it clocks down and limits it to ~2800 MHz when it would usually be able to reach 3800 MHz.

**Expected behavior**

When woken from sleep on BAT mode I expect it to maintain predefined settings and limit clock speeds to 1200 MHz,
and when woken on AC it should not disable Turbo.

**To Reproduce**

Steps to reproduce the unexpected behavior:

set CPU_SCALING_MAX_FREQ_ON_BAT=0 or CPU_MAX_PERF_ON_BAT=0
and suspend system, when it resumes it should uncap the frequency, while disabling boost.

1. Does the problem occur on battery or AC or both?
>> yes.
2. Actions to reproduce the behavior
>> previously listed
3. Shell commands entered and their output *including error message(s)*
>> ? N/A
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
https://gist.github.com/user12812/a54cb30619182e82ffc199bd27f7882c
sorry I don't know what "for all cases of 1" means

**Additional context**

sorry if this isn't the right place to report this, but I'm 90% sure that tlp is the reason that this is happening, I also looked at previous bug reports and could not find any similar issues.

potentially relevant hardware;
Thinkpad T430
i7-3840qm
bios: libreboot

linrunner, 2024-12-11:
>  sorry I don't know what "for all cases of 1" means

Your problem occurs on AC and battery power i.e. two cases. In your scenario, I need the output of 

    sudo tlp-stat -s -p

before and after the suspend in order to be able to compare:

1. on AC power before suspend
2. on AC power after suspend
3. on battery power before suspend
4. on battery power after suspend

user12812, 2024-12-11:
ac before suspend:
https://gist.github.com/user12812/afdd22cde509629860f4471c9d264c98
ac after suspend:
https://gist.github.com/user12812/c2b85904ce604dafbc4d1b71265134cc
bat before suspend:
https://gist.github.com/user12812/eed8fedc29e0dbc40b65fc70702b3ff5
bat after suspend:
https://gist.github.com/user12812/dbb236e7275ebfd379f8692f5dcc449b

I ran them through diff and there doesn't seem to be any difference but the problem still persists.

linrunner, 2024-12-13:
If the settings written by TLP - which `tlp-stat -p` reads back - are identical before and after the suspend, then it is not due to TLP, but the kernel is behaving differently than expected.

You will have to experiment with the settings.

> /etc/tlp.conf L0097: CPU_DRIVER_OPMODE_ON_AC="active"
> /etc/tlp.conf L0098: CPU_DRIVER_OPMODE_ON_BAT="active"

Your 10-year-old CPU does not support hardware P-states (HWP), so it is better to select _passive_ mode rather than _active_ mode. Incidentally, this is the kernel default. 

If this does not help, then the BIOS is probably slowing down the CPU after the resume via ACPI or it is a kernel bug.

Good luck.

---
Your other settings don't fit either:

> /etc/tlp.conf` L0158: CPU_MAX_PERF_ON_BAT="0"

Completely pointless, because the smallest maximum that the _intel_pstate_ driver supports is 31%:

> /sys/devices/system/cpu/intel_pstate/min_perf_pct      =  **31** [%]

Zero ist just ignored by the driver.

---
> /etc/tlp.conf L0195: PLATFORM_PROFILE_ON_AC="performance"
> /etc/tlp.conf L0196: PLATFORM_PROFILE_ON_BAT="low-power"

You can remove that, your ThinkPad is about 10 years too old for the feature:

> +++ Platform Profile
> /sys/firmware/acpi/platform_profile                    = **(not available)**

user12812, 2024-12-14:
yes, thank you.
I've done most of the things you recommended and none of them worked, though I seemed to solve the problem of the turbo turning off by enabling it on both battery mode and ac. I have tested with a generic kernel and had a quick search through my custom kernel and did not see anything that might fix it, so I will just leave it at it being unsupported buggy old hardware, a likely permanent temporary solution to the only remaining problem of it boosting up after i set the clocks to minimum is setting it to ac mode, then back to bat and restarting the daemon, since just restarting it does not seem to fix it.
Also would you recommend using CPU_SCALING_MAX_FREQ_ON_BAT or CPU_MAX_PERF_ON_BAT for limiting clock speeds, since it is recommended to use CPU_MAX_PERF_ON_BAT while using the intel_pstate driver, but since I am using it in passive mode what would you recommend?

linrunner, 2024-12-15:
This all sounds very much like a kernel problem. You'll just have to keep experimenting with the settings.

Note: CPU_SCALING_MAX_FREQ_ON_BAT=0 is not meant to deactivate the setting, just leave it commented out. 

For limiting the performance I prefer CPU_MAX_PERF_ON_BAT, but I'm not sure if this also works in passive mode and I don't have a chance to test it right now.

linrunner, 2024-12-15:
ps. you may also experiment with CPU_ENERGY_PERF_POLICY_ON_AC/BAT. But remember to change CPU_SCALING_GOVERNOR_ON_AC to _powersave_ as _performance may restrict ENERGY_PERF_POLICY_ON_AC to _performance_ too.

user12812, 2024-12-16:
I tried it with passive mode and it for some reason locks it to minimum clock speeds on both AC and BAT, so I've gone back to active. I've got it to work the way I needed it to with a script that restarts the daemon on wake-up, and I do have both CPU_SCALING_GOVERNER_ON_AC/BAT and CPU_ENERGY_PERF__POLICY_ON_BAT/AC on both powersave and performance respectively, I don't know if it does lock it, as this is the behavior I want. Ill just chuck it up to a kernel issue or maybe I set something up wrong on my part, but I think its fine to consider this closed.

closed 2024-12-16 by user12812