linrunner/TLP #777: Cant Use Battery Care In thinkbook 14+

gczgcz2015, 2025-01-02:
[✔] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[✔] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
I cant use Battery Care in my ThinkBook 14+ （amd 8845h）

**Expected behavior**

fix Battery can not use

```
system: archlinux
```

command tlp-stat -b shows:
```
--- TLP 1.7.0 --------------------------------------------

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = COSMX
/sys/class/power_supply/BAT1/model_name                     = L23X4PG1
/sys/class/power_supply/BAT1/cycle_count                    =     23
/sys/class/power_supply/BAT1/energy_full_design             =  85000 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  84320 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  84320 [mWh]
/sys/class/power_supply/BAT1/power_now                      =     17 [mW]
/sys/class/power_supply/BAT1/status                         = Full

/sys/class/power_supply/BAT1/charge_control_start_threshold = (not available) 
/sys/class/power_supply/BAT1/charge_control_end_threshold   = (not available) 

Charge                                                      =  100.0 [%]
Capacity                                                    =   99.2 [%]
```

thanks a lot

linrunner, 2025-01-02:
> [✔] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

Really? Please show the **complete** output of

    sudo tlp-stat

I also need the output of

    sudo modprobe -v --first-time ideapad_laptop
    grep . /sys/bus/platform/drivers/ideapad_acpi/* /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/*

gczgcz2015, 2025-01-02:
> Really? Please show the **complete** output of
>
> ```
> sudo tlp-stat
> ```
>
> I also need the output of
>
> ```
> sudo modprobe -v --first-time ideapad_laptop
> grep . /sys/bus/platform/drivers/ideapad_acpi/* [...]

sorry, here : https://gist.github.com/gczgcz2015/a6a773f01e0f9498287a89b24655606e
and 

sudo modprobe -v --first-time ideapad_laptop
```
insmod /lib/modules/6.12.7-arch1-1/kernel/drivers/input/sparse-keymap.ko.zst 
insmod /lib/modules/6.12.7-arch1-1/kernel/drivers/acpi/platform_profile.ko.zst 
insmod /lib/modules/6.12.7-arch1-1/kernel/drivers/platform/x86/ideapad-laptop.ko.zst 
```

grep . /sys/bus/platform/drivers/ideapad_acpi/* /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/*

```
grep: /sys/bus/platform/drivers/ideapad_acpi/bind: Permission denied
grep: /sys/bus/platform/drivers/ideapad_acpi/module: Is a directory
grep: /sys/bus/platform/drivers/ideapad_acpi/uevent: Permission denied
grep: /sys/bus/platform/drivers/ideapad_acpi/unbind: Permission denied
grep: /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00: Is a directory
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode:0
grep: /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/driver: Is a directory
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/driver_override:(null)
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/fan_mode:133
grep: /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/firmware_node: Is a directory
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/fn_lock:0
grep: /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/input: Is a directory
grep: /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/leds: Is a directory
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/modalias:acpi:VPC2004:
grep: /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/power: Is a directory
grep: /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/subsystem: Is a directory
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/uevent:DRIVER=ideapad_acpi
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/uevent:MODALIAS=acpi:VPC2004:
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/usb_charging:1
```

linrunner, 2025-01-03:
This looks promising:

> /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/**conservation_mode:0**

Now `tlp-stat -b` should announce the availability of a charge threshold. 

However, since Thinkbooks have an _Ideapad_ interior. There are no arbitrary threshold values like in ThinkPads, but only a fixed stop threshold (60 or 80% depending on the model). You must change your configuration accordingly:

    START_CHARGE_THRESH_BAT0=0 # dummy value
    STOP_CHARGE_THRESH_BAT0=1 

and run

    sudo tlp setcharge
then show 

    sudo tlp-stat -b
again.

See https://linrunner.de/tlp/settings/bc-vendors.html#lenovo-non-thinkpad-series

To ensure that `ideapad_laptop` is always loaded at boot time, create a file **/etc/modules-load.d/local.conf** containing just the line:

    ideapad_laptop

gczgcz2015, 2025-01-04:
thanks for reply.
I changed config to 
[image]
and restart tlp.service
but when I run ```sudo tlp setcharge``` return an error:
```
Error: battery charge thresholds not available.
```

> /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode:0
And shoulud I config conservation_mode to 1?  (I dont know how to do it, I enter BOIS and turn Battery mode to “Energy saving mode”， but still not working)
tanks again

gczgcz2015, 2025-01-04:
thank you very much. I get it.
[image]

And ThinkBook 14 G6+ AHP ideapad_laptop module has a faulty, according to https://wiki.archlinux.org/title/Laptop/Lenovo . so it is blocked by ```/etc/modprobe.d/blacklist-ideapad-laptop.conf``` need remove it.
Fanlly thank you very much!😄

linrunner, 2025-01-04:
You're welcome.

closed 2025-01-04 by linrunner