linrunner/TLP #779: The package is outdate on Fedora

sanjarbek-ismatov, 2025-01-11:
**Is your feature request related to a problem? Please describe.**

Fedora provides 1.6.1 version of TLP while 1.7.0 is already out

**Describe the solution you'd like**

I wanted to get the latest version of it from the official repo to not conflict with selinux

* New kernel interfaces: please provide links to the description and usage examples
* Sample shell code
```
uname -r
6.12.8-200.fc41.x86_64
```
```
rpm -q tlp
tlp-1.6.1-1.fc41.noarch
```

akmsk1999, 2025-01-11:
Please update it on Fedora

linrunner, 2025-01-11:
You are in the wrong place. As an upstream project, I do not provide distribution packages. Please address your request to the Fedora developers.

However, you could use the beta repo temporarily: https://download.linrunner.de/packages/

closed 2025-01-11 by linrunner