linrunner/TLP #780: Can't disable battery threshold

sudo-userv2, 2025-01-16:
So like around 1.5 years ago , i set up tlp with a max battery threshold, and now that i want to remove it i noticed that my device somehow lost support ? 

[image]

[image]

# System Details
---

## Report details
- **Date generated:**                              2025-01-16 22:27:13

## Hardware Information:
- **Hardware Model:**                              SAMSUNG ELECTRONICS CO., LTD. 950QDB
- **Memory:**                                      8.0 GiB
- **Processor:**                                   11th Gen Intel® Core™ i5-1135G7 × 8
- **Graphics:**                                    Intel® Iris® Xe Graphics (TGL GT2)
- **Disk Capacity:**                               256.1 GB

## Software Information:
- **Firmware Version:**                            P12AKG.028.240429.MK
- **OS Name:**                                     Fedora Linux 41 (Workstation Edition)
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               47
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.12.8-200.fc41.x86_64

linrunner, 2025-01-17:
Ramdom parts of the tlp-stat output and information from external tools are unfortunately of little help in the diagnosis.

The required kernel module may not be loaded. Please show the output of

    sudo modprobe -v --first-time thinkpad_acpi

and

    sudo tlp-stat -s -b -v

Please copy the terminal input and output here, not as a screenshot. Thank you.

sudo-userv2, 2025-01-21:
running the first show : 
`insmod /lib/modules/6.12.9-200.fc41.x86_64/kernel/drivers/platform/x86/samsung-laptop.ko.xz 
modprobe: ERROR: could not insert 'samsung_laptop': No such device`

and for the second: 

`--- TLP 1.6.1 --------------------------------------------

+++ System Info
System         = SAMSUNG ELECTRONICS CO., LTD. P12AKG 950QDB
BIOS           = P12AKG.028.240429.MK
OS Release     = Fedora Linux 41 (Workstation Edition)
Kernel         = 6.12.9-200.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Jan  9 16:05:40 UTC 2025 x86_64
/proc/cmdline  = BOOT_IMAGE=(hd0,gpt2)/vmlinuz-6.12.9-200.fc41.x86_64 root=UUID=36ad4ab7-8a6a-49af-ae50-8a7ac3a567b0 ro rootflags=subvol=root rhgb quiet
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle] deep
SELinux status = enabled (enforcing)

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 09:06:46 PM, 393 sec(s) ago
Mode           = battery
Power source   = battery

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = SAMSUNG Electronics
/sys/class/power_supply/BAT1/model_name                     = SR Real Battery
/sys/class/power_supply/BAT1/cycle_count                    =    259
/sys/class/power_supply/BAT1/charge_full_design             =   4340 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4200 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   3360 [mAh]
/sys/class/power_supply/BAT1/current_now                    =    402 [mA]
/sys/class/power_supply/BAT1/status                         = Discharging

/sys/class/power_supply/BAT1/voltage_min_design             =  15440 [mV]
/sys/class/power_supply/BAT1/voltage_now                    =  16103 [mV]

/sys/class/power_supply/BAT1/charge_control_start_threshold = (not available) 
/sys/class/power_supply/BAT1/charge_control_end_threshold   = (not available) 

Charge                                                      =   80.0 [%]
Capacity                                                    =   96.8 [%]`

linrunner, 2025-01-24:
If the kernel module does not load, you have no way to disable the threshold under Linux. 

You can try an older kernel or submit a kernel bug report. Did you check if it is possible to remove the threshold in the BIOS setup?I 

I can do no more for you than give you these hints.

linrunner, 2025-03-02:
I am closing here due to inactivity. If you have already found a solution, I would be happy if you could share it here to help others.

closed 2025-03-02 by linrunner