linrunner/TLP #782: Even after purge battery still dont charge  over 80%

VladaZemunac, 2025-01-22:
[shinoby tpe14 sudo tlp.txt](https://github.com/user-attachments/files/18510067/shinoby.tpe14.sudo.tlp.txt)

[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

AFER disabling TLP, battery still won't charge over 79%, even when the laptop is powered off.

Then I have purged TLP, but still the battery won't charge over 79%, even when the laptop is powered off.

**Expected behavior**

To be able to charge the battery outside the scope of initial set parameters by TLP

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
2. Actions to reproduce the behaviour
3. Shell commands entered and their output *including error message(s)*
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)

**Additional context**

Add any other context about the problem here.

linrunner, 2025-01-23:
I guess, you missed the FAQ entry for your concern: https://linrunner.de/tlp/faq/battery.html#how-to-remove-the-charge-thresholds

VladaZemunac, 2025-01-24:
Thank you for your response