linrunner/TLP #784: [TLP 1.8 beta] discharge malfunction on ThinkPad E590

linrunner, 2025-01-25:
$ sudo tlp-stat -b
--- TLP 1.8.0-beta.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = Celxpert
/sys/class/power_supply/BAT0/model_name                     = 01AV448
/sys/class/power_supply/BAT0/cycle_count                    =    877
/sys/class/power_supply/BAT0/energy_full_design             =  45730 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  36610 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  30290 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   5759 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Other command, `tlp recalibrate BAT0` the output is: 
 
$ sudo tlp recalibrate BAT0
Setting temporary charge thresholds for battery BAT0:
   start =  96 (no change)
   stop  = 100 (no change)
Initiating discharge of battery BAT0 ...............
Error: discharge BAT0 malfunction -- check your hardware (battery, charger).

$ sudo tlp discharge 80
Initiating discharge of battery BAT0 ...............
Error: discharge BAT0 malfunction -- check your hardware (battery, charger).

[Battery](https://gist.github.com/Senior-Moe/d087dad8121cdb9003a2192d5bec6e40)
[AC](https://gist.github.com/Senior-Moe/06959832d067385e20d3171225b8740c)
>  

 _Originally posted by @Senior-Moe in [#781](https://github.com/linrunner/TLP/issues/781#issuecomment-2613363622)_

linrunner, 2025-01-25:
@Senior-Moe let's get some traces:

1. Install  **1.8.0-beta.1**

2. Add the following line to your configuration:

     TLP_DEBUG="arg bat run"

3. Show the output of

     sudo tlp discharge 
     sudo tlp-stat -T

Senior-Moe, 2025-01-25:
@linrunner 
Apologize for late. output for `sudo tlp discharge` 

```
Currently discharging battery BAT0 to 0% (0 mWh):
voltage            =  11043 [mV]
remaining energy   =  22600 [mWh]
remaining percent  =   61.6 [%]
remaining time     =    112 [min]
power              =  12036 [mW]
status             = Discharging
force-discharge    = 1
Press Ctrl+C to cancel.
```
here output of `sudo tlp-stat -T`

```
 $ sudo tlp-stat -T
--- TLP 1.8.0-beta.1 --------------------------------------------

Error: /var/log/debug does not exist.

Solution: create an rsyslog conffile /etc/rsyslog.d/90-debug.conf with the following contents
 *.=debug;\
 mail,authpriv,cron.none;\
 local0,local1,local3,local4,\
 local5,local6,local7.none    -/var/log/debug
and restart the rsyslog daemon.
```
Note: I don't have rsyslog on runit directory.

linrunner, 2025-01-26:
> Apologize for late. 

No problem. Troubleshooting takes time.
 
> output for `sudo tlp discharge`

> Currently discharging battery BAT0 to 0% (0 mWh):
> voltage            =  11043 [mV]`
> remaining energy   =  22600 [mWh]
> remaining percent  =   61.6 [%]
> remaining time     =    112 [min]
> power              =  12036 [mW]
> status             = Discharging
> force-discharge    = 1
> Press Ctrl+C to cancel.

> In addition, `discharge` command worked faster than yesterday.

Obviously the error no longer occurs and `tlp discharge` starts correctly. As long as the error does not happen again, further analysis is not possible.

> [AC](https://gist.github.com/Senior-Moe/5790696cf238b1046ce6bffc61e1f204) 
> [battery](https://gist.github.com/Senior-Moe/3ad7bb713bb0ea6d12befcfae91fe386) 
> [Log file](https://gist.github.com/Senior-Moe/abc4013ce0110f6a1c35dbc48fcd5f4f)

I can't find a discharge command in your trace outputs. The log timestamps should match the actual execution time of the `tlp setcharge` command.

One last note: if you test `tlp discharge`, just keep the charger connected all the time. Outputs for battery are unnecessary here.

Senior-Moe, 2025-01-28:
Dear @linrunner 

Here what did you ask it for:
[Log file](https://gist.github.com/Senior-Moe/ba8484a1fabf836474d916d3fef8fde3)
[tlp-stat -T](https://gist.github.com/Senior-Moe/62e0dd2df4d9568be2159a7b6065246d)

linrunner, 2025-01-29:
@Senior-Moe the tlp discharge error no longer occurs, right? May I close the issue?

Senior-Moe, 2025-01-29:
@linrunner yes, the errors were gone. Before close it can you please explain what is going on? Why did  issue comes and left? Is it related  emergency reset?

Thanks for everything and I appreciate it.

linrunner, 2025-01-29:
@Senior-Moe If I knew what was happening, I wouldn't ask you for traces.

You're welcome.

closed 2025-01-29 by linrunner

linrunner, 2025-02-13:
I have just released [TLP 1.8.0](https://github.com/linrunner/TLP/releases/tag/1.8.0). Have fun!