linrunner/TLP #785: [TLP 1.8.0 Beta]

linrunner, 2025-01-25:
_Originally posted by @linrunner in [#781](https://github.com/linrunner/TLP/issues/781#issuecomment-2607804203)_

We still have a problem with the new Dell plugin.

After you, @Sempro97, were unable to write the thresholds with TLP in [your first test](https://github.com/linrunner/TLP/issues/379#issuecomment-2537294720), you helped yourself with `cctk` and set **charge_types** to _Custom_. After that, `tlp setcharge` worked.
 
I then extended the code by setting **charge_types** to _Custom_ before writing the thresholds.
 
In your test, @samuelemusiani, however, the writing of **charge_types**  fails.
 
 At the moment I am at a loss as to how TLP could get Dell laptops to accept threshold values. Which is also due to the fact that I don't know this hardware better.
 
 I only had a brief opportunity to test on a Dell. It initially had **charge_types** set to _Adaptive_ and changing it to _Custom_ and afterwards writing the thresholds worked immediately.
 
 I don't find it acceptable that it would take an intervention with `cctk` to get TLP to work. 
 
 Any ideas guys?

linrunner, 2025-01-25:
In the meantime, I have been in contact with the kernel dev, who is now trying to reproduce the problem. We would be grateful for any further information on the circumstances of the error.

So far we have had the following results:
- XPS 15 9560 - failed to write **charge_control_*_threshold** - works after cctk intervention https://github.com/linrunner/TLP/issues/379#issuecomment-2537294720
- Latitude 7400 2-in-1 - failed to write **charge_types** - no workaround tried yet
- Precision 7530 - works ootb

Sempro97, 2025-02-03:
In my Dell 9560 the error "Error: failed to set 'Custom' charge type for battery BAT0. Aborted." appears if i set an "admin password" in bios settings. Maybe @samuelemusiani has set a password

samuelemusiani, 2025-02-03:
> In my Dell 9560 the error "Error: failed to set 'Custom' charge type for battery BAT0. Aborted." appears if i set an "admin password" in bios settings. Maybe @samuelemusiani has set a password

Yes, I have an admin password in the bios. I'll try disabling it and I'll let you know.

But in general I prefer having a password on the bios, is it generally possible to use TLP with a bios password or is something that is not compatible? Sorry but I don't know how the code works

linrunner, 2025-02-04:
@Sempro97 thanks for checking. Now we get clarity.

@samuelemusiani thanks for the info. Can you please share the result of your test without admin password? 

This is obviously a security feature that Dell has deliberately built in and that cannot simply be bypassed. TLP does not write the thresholds to the EC registers itself, the kernel driver does that. I'll ask the kernel dev, but I don't expect a workaround to be possible.

linrunner, 2025-02-06:
The kernel dev has confirmed my assumption. There will be no way to write the thresholds through the kernel despite the BIOS admin password being set.

I have add a respective note in the error messages -> https://github.com/linrunner/TLP/commit/f1f1312d4db4efae86d0d331a0448210a7a6d36c

As the Dell hardware seems to save the thresholds permanently, it is sufficient to temporarily remove the Admin password, write the thresholds once with `tlp setcharge` and then restore the Admin password.

@samuelemusiani was your test without a password successful?

closed 2025-02-13 by linrunner