linrunner/TLP #786: Consider doing away with defaults.conf in /usr/share

gdevenyi, 2025-01-29:
**Is your feature request related to a problem? Please describe.**

The current default installation of tlp incldues a defaults.conf file, which sets a number of settings in a place where a user does not expect. If I want to install TLP and have it "not do anything" I expect to be able to provide a blank tlp.conf (and no /etc/tlp.d/* files) and have tlp do nothing. Instead I inherit a "hidden" config file.

**Describe the solution you'd like**

I would like to propose that defaults.conf be merged with tlp.conf and placed in /etc/tlp.d/00-defaults.conf so that users can see the defaults, and *remove* them if they wish to. Prefixing the filename will ensure additional settings made in other files are properly used.

linrunner, 2025-01-30:
Your actual requirement is a convenient way to deactivate (almost) all defaults, isn't it? That's open to discussion.

I cannot and will not remove or move the **defaults.conf**. That would suddenly deconfigure all existing installations with the next version upgrade. I'm not going to discuss that.

> so that users can see the defaults, 

All defaults are documented inside **tlp.conf** as well as on the website.

gdevenyi, 2025-01-30:
> Your actual requirement is a convenient way to deactivate (almost) all defaults, isn't it? That's open to discussion.

Yes. The existence of defaults.conf outside of standard configuration locations makes this "hard" and opaque.

Very happy to consider alternative solutions.

spikespaz, 2025-04-25:
> I cannot and will not remove or move the defaults.conf. That would suddenly deconfigure all existing installations with the next version upgrade. I'm not going to discuss that.

You can stop creating `defaults.conf` in that silly location for future versions, and still read the file on machines where it exists already.

This (reasonable) change doesn't have to be breaking.

linrunner, 2025-07-17:
> If I want to install TLP and have it "not do anything" I expect to be able to provide a blank tlp.conf (and no /etc/tlp.d/* files) and have tlp do nothing.

@gdevenyi Done. Please test. For some packages see the [download page](https://download.linrunner.de/packages/).