linrunner/TLP #789: [Thinkpad W530] TLP 1.7.0 does not recognize Thinkpad for Battery Care, on Gentoo Linux, Kernel 6.13.0

sct0, 2025-02-18:
The bug:

TLP 1.70 does not load the thinkpad plugin on my Thinkpad W530, running Gentoo Linux with kernel version 6.13.0.
Therefore the battery charging thresholds don't work.

Expectation:

I expected TLP to recognize the battery since I am on a supported Thinkpad model, with the required kernel and TLP versions and have thinkpad_acpi built into the kernel.

Steps to reproduce the unexpected behavior:

1. Issue occurs on AC and battery

2. To reproduce one runs `sudo tlp-stat -b` on AC or battery:

```
--- TLP 1.7.0 --------------------------------------------

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 45N1011
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  93600 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  81470 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  81430 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  16008 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]

Charge                                                      =  100.0 [%]
Capacity                                                    =   87.0 [%]
```

4. I also ran `sudo modprobe -v --first-time thinkpad_acpi` to verify that the module is installed:

```
sudo modprobe -v --first-time thinkpad_acpi
modprobe: ERROR: could not insert 'thinkpad_acpi': Module already in kernel
```
This is expected since I have almost all modules including this one built into the kernel.

Additionally I checked my kernel parameters in my GRUB config:
```
# Append parameters to the linux kernel command line
GRUB_CMDLINE_LINUX="root=UUID=b516315a-f599-4c3b-a746-df75e4f4dbbf intel_pstate=force intel_pstate=active nowatchdog"
```

5. Here are the full `sudo tlp-stat` outputs for both AC and battery:

<script src="https://gist.github.com/sct0/8bc9fff25b56bfe902874fd66c049ae2.js"></script>

<script src="https://gist.github.com/sct0/4df5ccb860f5df69909fd961b2938de6.js"></script>

6. I am quite certain that the battery I got with the laptop is an original Lenovo battery, but I don't want to turn the laptop off to make sure right now since I am in the middle of working.
I will append that information asap though should it be relevant.
I use a custom kernel. Except for one consistent login bug and about 4 random reboots, the system has been running stable for several months now. Can't think of anything kernel-side that could be wrong, but here's my .config just in case:

<script src="https://gist.github.com/sct0/77edec80f9a7a6e71fb3994f2b968949.js"></script>

linrunner, 2025-02-20:
Please provide the output of:

```
ls -l /sys/devices/platform/thinkpad_acpi
grep . /sys/class/dmi/id/*
```

sct0, 2025-02-20:
Here you go:

```
ls -l /sys/devices/platform/thinkpad_acpi
total 0
-rw-r--r--  1 root root 4096 Feb 20 19:30 bluetooth_enable
--w-------  1 root root 4096 Feb 20 19:30 cmos_command
lrwxrwxrwx  1 root root    0 Feb 20 19:30 driver -> ../../../bus/platform/drivers/thinkpad_acpi
-rw-r--r--  1 root root 4096 Feb 20 19:30 driver_override
-r--r--r--  1 root root 4096 Feb 20 19:30 hotkey_adaptive_all_mask
-r--r--r--  1 root root 4096 Feb 20 19:30 hotkey_all_mask
-r--r--r--  1 root root 4096 Feb 20 19:30 hotkey_bios_enabled
-r--r--r--  1 root root 4096 Feb 20 19:30 hotkey_bios_mask
-rw-r--r--  1 root root 4096 Feb 20 19:30 hotkey_enable
-rw-r--r--  1 root root 4096 Feb 20 19:30 hotkey_mask
-rw-r--r--  1 root root 4096 Feb 20 19:30 hotkey_poll_freq
-r--r--r--  1 root root 4096 Feb 20 19:30 hotkey_radio_sw
-r--r--r--  1 root root 4096 Feb 20 19:30 hotkey_recommended_mask
-rw-r--r--  1 root root 4096 Feb 20 19:30 hotkey_source_mask
drwxr-xr-x  3 root root    0 Feb 20 19:30 input
drwxr-xr-x 10 root root    0 Feb 20 19:30 leds
-r--r--r--  1 root root 4096 Feb 20 19:30 modalias
drwxr-xr-x  2 root root    0 Feb 20 19:30 power
drwxr-xr-x  3 root root    0 Feb 20 13:32 rfkill
lrwxrwxrwx  1 root root    0 Feb 20 19:30 subsystem -> ../../../bus/platform
-rw-r--r--  1 root root 4096 Feb 20 19:30 uevent
-r--r--r--  1 root root 4096 Feb 20 19:30 wakeup_hotunplug_complete
-r--r--r--  1 root root 4096 Feb 20 19:30 wakeup_reason
```

```
grep . /sys/class/dmi/id/*
grep: /sys/class/dmi/id/*: No such file or directory
```

linrunner, 2025-03-02:
You seem to have found a solution. I'm closing here, but would be happy if you could share it here to help others.

closed 2025-03-02 by linrunner

sct0, 2025-04-10:
Hi,
sorry, I was busy with exams.
I recompiled my kernel with the DMI settings enabled and it did indeed fix the issue.
Thank you very much!

linrunner, 2025-04-10:
I guessed so. Thanks for your feedback.