linrunner/TLP #790: Implement a cpu hotplug feature where user can pass what cores can be disabled

nwildner, 2025-02-19:
In this feature request idea, the user would be allowed to power down some of the CPU cores when on battery mode for some extra juice. 

On my current model(ASUS TUF Dash F15) keeping only `core0` and `core1` and the "efficiency cores"(E-Cores)  `cpu12`, `cpu13`, `cpu14` and `cpu15` last four cores provide me roughly and extra of 20 to 35% of battery time. I think that the fact you have less cores turned on and the 4 E-Cores have lower max mhz helps a bit with some indirect power saving: 

```
lscpu -e 
CPU NODE SOCKET CORE L1d:L1i:L2:L3 ONLINE    MAXMHZ   MINMHZ       MHZ
  0    0      0    0 0:0:0:0          yes 4600.0000 400.0000  400.0000
  1    0      0    0 0:0:0:0          yes 4600.0000 400.0000  400.0000
  2    0      0    1 4:4:1:0          yes 4600.0000 400.0000  728.3510
  3    0      0    1 4:4:1:0          yes 4600.0000 400.0000  400.0000
  4    0      0    2 8:8:2:0          yes 4700.0000 400.0000  598.7270
  5    0      0    2 8:8:2:0          yes 4700.0000 400.0000  400.0000
  6    0      0    3 12:12:3:0        yes 4700.0000 400.0000  843.1310
  7    0      0    3 12:12:3:0        yes 4700.0000 400.0000  400.0000
  8    0      0    4 16:16:4:0        yes 4600.0000 400.0000  400.0000
  9    0      0    4 16:16:4:0        yes 4600.0000 400.0000  491.3900
 10    0      0    5 20:20:5:0        yes 4600.0000 400.0000  400.0000
 11    0      0    5 20:20:5:0        yes 4600.0000 400.0000  400.0000
 12    0      0    6 28:28:7:0        yes 3500.0000 400.0000  400.0000
 13    0      0    7 29:29:7:0        yes 3500.0000 400.0000  615.0990
 14    0      0    8 30:30:7:0        yes 3500.0000 400.0000 1102.9550
 15    0      0    9 31:31:7:0        yes 3500.0000 400.0000  400.0000
```

The initial solution would be: A parameter where the user can pass down which CPUs can be shutdown through hotplug event or if the laptop is starting on battery only.

Example:

```
CPU_DISABLE_CORES_BAT="2 3 4 5 6 7 8 9 10 11"
```
Some insights on how this would work: 

- A script would send `echo 0 > /sys/devices/system/cpu/cpuX/online` where X is the CPU index provided on `CPU_DISABLE_CORES_BAT`
- `cpu0` could be ignored since it can't be hotplugged to off. On my current setup, I ignore the first 2 cores just in case (`cpu0` an `cpu1`) 
- Implicitly turn on all cores back again when power is plugged if there is something declared on `CPU_DISABLE_CORES_BAT`

That is a feature I miss most from `laptop-mode-tools` and there were some predefined values like "turn off half of the cores" were available but, TLP could start  by simply allowing to hotplug CPUs by index on this first implementation.

I'm currently controlling this through `udev` since laptop-mode is dead in the water and I've added a patch specifically to manage e-cores which I think the owner lost interest 

- https://github.com/rickysarraf/laptop-mode-tools/pull/206
- https://github.com/rickysarraf/laptop-mode-tools/blob/lmt-upstream/usr/share/laptop-mode-tools/modules/cpuhotplug

Since I've migrated recently to TLP and didn't had the time to learn the code, I'm opening a Feature Request here :) 

Related:

- https://docs.kernel.org/core-api/cpu_hotplug.html

linrunner, 2025-03-02:
I think this is a feature for a _very_ small niche of users. Nevertheless, I would accept a PR from you.

First, to make this usable and user-friendly, `tlp-stat -p` must be able to output the type (P, E, LP-E) for each core. So that users get to know what they should enter into the configuration. There was already an attempt to write something like this, but the creator lost interest: https://github.com/linrunner/TLP/issues/736#issuecomment-2036295981

The configuration must be symmetrical, i.e. CPU_DISABLE_CORES_ON_AC and CPU_DISABLE_CORES_ON_BAT, so that it is defined what happens when switching the power source. Both defaults would be empty of course.

nwildner, 2025-03-23:
Thanks for the heads up @linrunner 

Wouldn't it be better to just categorize the CPUs into 2 groups, the E-Core and the Performance ones? As quoted on the other case, even Intel itself through their software only really categorize them into these two groups simplifying the nomenclature a little bit:

- https://github.com/linrunner/TLP/issues/736#issuecomment-2065437644
- https://industrialpc.com/blog/intel-cpu-ecore-pcore/

Also, it is fairly easy to probe that using only 2 files and no complex calculations. Here:

```
[nwildner@arrakis ~]$ cat /sys/devices/cpu_atom/cpus 
12-15
[nwildner@arrakis ~]$ cat /sys/devices/cpu_core/cpus 
0-11
```

Cores listed inside `/sys/devices/cpu_atom/cpus`  are E-Cores and cores listed at `/sys/devices/cpu_core/cpus`  are the performance ones.

Another way to compare would be to match what `CPU` lines from the `lscpu -e` output have more than one `CORE` column associated with it, indicating a SMT cpu, thus, performance. E-Core are by design non-SMT. I still prefer relying on the information that is already provided on the `/sys` interface on the files referenced above.

What do you think?

linrunner, 2025-03-23:
I think a distinction in P- and E/LP-E cores based on **/sys/devices/cpu_{atom,core}/cpus** is the most pragmatic. I always try to avoid dependencies on external tools if there is a way in sysfs. However, I do not have a computer with these Intel CPUs.