linrunner/TLP #791: Loop 10 minutes: Stopping disk -> Starting disk -> Synchronizing SCSI cache -> Stopping disk

KAGEYAM4, 2025-02-20:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

https://gist.github.com/KAGEYAM4/9abae205e6a5b22d312974cdbc60bb59

**Describe the bug**
-
  ```
  Feb 19 23:30:49 ArchLinux kernel: sd 5:0:0:0: [sda] Synchronizing SCSI cache
  Feb 19 23:30:49 ArchLinux kernel: sd 5:0:0:0: [sda] Stopping disk
  Feb 19 23:30:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:30:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:30:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:31:17 ArchLinux kernel: sd 5:0:0:0: [sda] Synchronizing SCSI cache
  Feb 19 23:31:17 ArchLinux kernel: sd 5:0:0:0: [sda] Stopping disk
  Feb 19 23:40:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:40:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:40:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:41:17 ArchLinux kernel: sd 5:0:0:0: [sda] Synchronizing SCSI cache
  Feb 19 23:41:17 ArchLinux kernel: sd 5:0:0:0: [sda] Stopping disk
  Feb 19 23:50:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:50:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:50:58 ArchLinux kernel: sd 5:0:0:0: [sda] Starting disk
  Feb 19 23:51:17 ArchLinux kernel: sd 5:0:0:0: [sda] Synchronizing SCSI cache
  Feb 19 23:51:17 ArchLinux kernel: sd 5:0:0:0: [sda] Stopping disk
  ```

- The above output is looping every 10 minutes. The disk in question is my secondary disk which is only ever used by wallpaper app or by smartd. My disk is in bad state, going to die soon but i still wanted to know why this happens as i am again going to replace it with segate drive.

**Expected behavior**

To completely stop the disk from waking and only be woken by apps (i.e wallpaper, smartd) OR to reduce the number of wakeups.

**To reproduce**
<details><summary>Details</summary>
<p>

- From now on, `standby` refers to `sudo smartctl -i -n standby /dev/sda` which is used to check the status of device only if it is not currently in standby mode. Below everything is tested with above command. 

- with **TLP disabled** (even with powertop --auto-tune, below results are the same)

	- AHCI_RUNTIME_PM_ON_BAT=on (through powertop)

		- APM_level = 1, goes in standby

		- APM_level = 127 goes in standby

		- APM_level = 128, doesnt go in standby

	- AHCI_RUNTIME_PM_ON_BAT=auto (through powertop)

		- APM_level = 1, goes in standby

		- APM_level = 127 goes in standby

		- APM_level = 128, dosent go in standby

	-
	  ```
	     Bad           VM writeback timeout
	     Bad           Enable SATA link power management for host3
	     Bad           Enable SATA link power management for host1
	     Bad           Enable Audio codec power management
	     Bad           Enable SATA link power management for host2
	     Bad           NMI watchdog should be turned off
	     Bad           Autosuspend for USB device HD Webcam [SunplusIT Inc]
	     Bad           Runtime PM for port ata5 of PCI device: Intel Corporation Cannon Lake Mobile PCH SATA AHCI Controller
	     Bad           Runtime PM for PCI Device Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller
	     Bad           Runtime PM for port ata4 of PCI device: Intel Corporation Cannon Lake Mobile PCH SATA AHCI Controller
	     Bad           Runtime PM for port ata1 of PCI device: Intel Corporation Cannon Lake Mobile PCH SATA AHCI Controller
[... 97 lines cut ...]

		- APM_level = 128, device does not go into standby

		- `standby` command does not take time and it hdd does not make sound like it is was with `auto`. 
		  ```
		   sudo smartctl -i -n standby /dev/sda
		  smartctl 7.4 2023-08-01 r5530 [x86_64-linux-6.13.3-arch1-1] (local build)
		  Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org
		  
		  Device is in STANDBY mode, exit(2)
		  ```

</p>
</details> 

**Additional context**

It seems like TLP even with level 128 is able to make disk stop `ArchLinux kernel: sd 5:0:0:0: [sda] Stopping disk`

 In tlp docs it says 128 is good compromise, but with tlp-disabled and with value of 128, device never goes into standby. It seems like tlp is the one who makes the disk go in sleep mode (if what i understand is correct)

Is `sleep` different from `standby`? https://www.smartmontools.org/wiki/Powermode says that `standby` won't check the device unless it is in SLEEP or STANDBY mode.
	- But when i invoke `sudo hdparm -Y /dev/sda` (tlp is disabled) and then after sometime as seen in recording  , the command returns result instead of saying `SLEEP`

https://github.com/user-attachments/assets/f2d9cce9-8f9e-4426-b823-ce805d599ceb

linrunner, 2025-02-22:
It's dead simple: TLP itself does not stop disk drives. Instead, disks react autonomously to the APM level set, very individually depending on the brand, even by stopping the spindle motor. Try higher values or disable APM with 255. Good luck with that.

closed 2025-02-22 by linrunner

KAGEYAM4, 2025-12-24:
Anyidea about why when using AHCI_RUNTIME_PM, running command `smartctl -n standby /dev/sda` spins the disk up? it is not suppose to be.