linrunner/TLP #793: Battery discharge/recalibrate broken since upgrading to TLP 1.8

4goettma, 2025-03-08:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

Since upgrading to TLP 1.8.0, calibrating or discharging the battery (Type 29+) of my ThinkPad x220 fails.
Using tlp recalibrate returns "Error: cannot read charge level of battery BAT0."
The problem disappears when downgrading to [TLP 1.7.0](https://archive.archlinux.org/packages/t/tlp/tlp-1.7.0-1-any.pkg.tar.zst).

**Expected behavior**

TLP switching to discharging / calibration mode until the battery is depleted.

**To Reproduce**

1. Does the problem occur on battery or AC or both?
Problem can't be reproduced without AC being attached due to the design / technical limitations of TLP.

2. Actions to reproduce the behaviour
Install TLP version 1.8.0-1 from the official Arch Linux Repositories on a ThinkPad X220 using a Type 29+ battery. 
run `sudo tlp discharge` or `sudo run recalibrate`

3. Shell commands entered and their output *including error message(s)*
```
$> sudo tlp discharge
Error: cannot read charge level of battery BAT0.

$> sudo tlp recalibrate 
Setting temporary charge thresholds for battery BAT0:
  start =  96 (no change)
  stop  = 100 (no change)
Error: cannot read charge level of battery BAT0.
```

5. **Full output of `tlp-stat` via https://gist.github.com/ for *all*   matching cases of 1** (not as file attachment, no screenshots)
https://gist.githubusercontent.com/4goettma/cba65db99e2ff575278d779e1467f8a1/raw/754f25474305a864670f8b4ac540eaf2d21eaa4c/gistfile1.txt

**Additional context**

n/a

linrunner, 2025-03-10:
Hi,

please replace the file **/usr/share/tlp/bat.d/05-thinkpad** with [this file](https://raw.githubusercontent.com/linrunner/TLP/refs/heads/issue/793/bat.d/05-thinkpad) and re-test. Also show the output of

    sudo tlp-stat -b -v

4goettma, 2025-03-11:
The modified shell script solves the problem. Thank you for the quick reply.

I have some newer and older ThinkPad models available if more test data is needed.

The requested output:
```
$ sudo tlp-stat -b -v
--- TLP 1.8.0 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
* tp-smapi (tp_smapi)     = readonly (status)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/devices/platform/smapi/BAT0/manufacturer               = SANYO
/sys/devices/platform/smapi/BAT0/model                      = 42T4861
/sys/devices/platform/smapi/BAT0/manufacture_date           = 2012-05-11
/sys/devices/platform/smapi/BAT0/first_use_date             = 2012-07-02
/sys/devices/platform/smapi/BAT0/cycle_count                =    306
/sys/devices/platform/smapi/BAT0/temperature                =     25 [°C]
/sys/devices/platform/smapi/BAT0/design_capacity            =  57720 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =  40820 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =  39370 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =     96 [%]
/sys/devices/platform/smapi/BAT0/remaining_running_time_now = not_discharging [min]
/sys/devices/platform/smapi/BAT0/remaining_charging_time    = not_charging [min]
/sys/devices/platform/smapi/BAT0/power_now                  =      0 [mW]
/sys/devices/platform/smapi/BAT0/power_avg                  =      0 [mW]
/sys/devices/platform/smapi/BAT0/state                      = idle

/sys/devices/platform/smapi/BAT0/design_voltage             =  11100 [mV]
/sys/devices/platform/smapi/BAT0/voltage                    =  12446 [mV]
/sys/devices/platform/smapi/BAT0/group0_voltage             =   4147 [mV]
/sys/devices/platform/smapi/BAT0/group1_voltage             =   4152 [mV]
/sys/devices/platform/smapi/BAT0/group2_voltage             =   4147 [mV]
/sys/devices/platform/smapi/BAT0/group3_voltage             =      0 [mV]

/sys/class/power_supply/BAT0/charge_control_start_threshold =     70 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     90 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   96.4 [%]
Capacity                                                    =   70.7 [%]

+++ Recommendations
* BAT0 charge level is above the stop threshold. Use your laptop on battery power until the battery is discharged to the stop threshold.

```

linrunner, 2025-03-13:
> I have some newer and even older ThinkPad models available if more test data is needed.

Testing the change on some newer ThinkPads would be helpful. 

The AUR packages **tlp-git** and **tlp-rdw-git** track the main branch, containing the fix now.

There are automated tests available (just not for real discharge ...). Install the **clitest** tool (available as Arch package). Then clone the TLP GitHub repository, change to the **unit-tests** directory and run

    ./test-bc_thinkpad.sh

Post the full outputs (via Gist please).

I would be even happier if you could help me with tests for the releases on a regular basis :-).

4goettma, 2025-03-16:
Okay, so I've run a few tests today. I have selected some models from my collection and tried to cover a wide range of devices - new and old as well as tablet versions with lithium-polymer battery. The oldest and newest models available to me are included.
I'm always using the c8ed791230254ae2e74c82315b0eda32ba44ced7 commit. 
I'm using a custom build Arch Linux image (created today, changes regarding pre-installed packages and config files), booted in non-persistent mode from a USB stick.

Preparations:
```
sudo systemctl stop tlp.service
sudo rm /etc/tlp.conf
sudo pacman -R tlp tlp-rdw --noconfirm
yay -S tlp-git tlp-rdw-git
sudo systemctl restart tlp.service
sudo systemctl enable tlp.service
git clone https://github.com/linrunner/TLP
cd ~/TLP/unit-tests/
sudo pacman -S clitest --noconfirm
```

TL;DR:
Errors are showing up on the X60s, X200s, X201, T400, X270.

Results from a ThinkPad X60s, featuring a "Main / Internal" (removable) battery (replacement part, "TradeShop"), operating system based on Arch Linux 32, build in October 2024:

Results from a ThinkPad T400, featuring a "Main / Internal" (removable) battery:

Results from a ThinkPad X200s, featuring a "Main / Internal" (removable) battery (replacement part, "Green Cell"):

Results from a ThinkPad X201, featuring a "Main / Internal" (removable) battery (replacement part, "Green Cell"):

Results from a ThinkPad X230, featuring a "Main / Internal" (removable) battery:

Results from a ThinkPad X230t (tablet version), featuring a "Main / Internal" (removable) battery:

Results from a ThinkPad T440p, featuring a "Main / Internal" (removable) battery:

Results from a ThinkPad X250, featuring both a "Main / Internal" as well as a "Removable / Ultrabay / Slice" battery:

Results from the same ThinkPad X250, featuring only a "Main / Internal" ("Removable / Ultrabay / Slice" battery removed moments ago):

Results from a ThinkPad X270, featuring only a "Removable / Ultrabay / Slice" battery (and no "Main / Internal" battery):

Results from a ThinkPad T14 Gen 1, featuring a "Main / Internal" battery:

linrunner, 2025-03-16:
Thanks for running the tests. You have a nice collection there :-)

The supposed errors on the tp-smapi based ThinkPads are because `kmod-helper remove` doesn't seem to work on Arch. I only have Debian on my X200. I'll have to see if I can fix that or remove the test altogether.

The discrepancy on the X200s about “BAT0 charge level is above the stop threshold” is irrelevant, I've adjusted the test so that calling `tlp setcharge` suppresses that.

On the X201, it looks like 1.8 is still at work with the original bug. Can you check the software version there again?

The X270 with BAT1 but no BAT0 is a special case where I have to consider whether I should take it into account at all.

linrunner, 2025-03-16:
ps. for the X210 please show 

    grep . /sys/class/power_supply/BAT0/* 2> /dev/null | grep -v uevent

4goettma, 2025-03-16:
Thank you! :D

I've run some tests again using b1b894bed770428639e589611e22ca56c1a74a96.

X200s (`OK: 67 of 67 tests passed`):
`test-bc_thinkpad.sh` https://gist.github.com/4goettma/21f5303b0c8751182416d79b1e6063ca
`sudo tlp-stat -v` https://gist.github.com/4goettma/97e63f2fcdc8cb2582960b15e3d49953

X201 (`FAIL: 2 of 67 tests failed`):
`test-bc_thinkpad.sh` https://gist.github.com/4goettma/62ff297a2e92c4338ed502c190129c3c
`sudo tlp-stat -v` https://gist.github.com/4goettma/840a7e6b69e79d9087e1252fd3559438

Why do you think version 1.8 was still being used? I can't find the version number 1.8 in the posted outputs.

Requested output:
```
$> grep . /sys/class/power_supply/BAT0/* 2> /dev/null | grep -v uevent
/sys/class/power_supply/BAT0/alarm:223000
/sys/class/power_supply/BAT0/capacity:99
/sys/class/power_supply/BAT0/capacity_level:Normal
/sys/class/power_supply/BAT0/charge_behaviour:[auto]
/sys/class/power_supply/BAT0/charge_full:4143000
/sys/class/power_supply/BAT0/charge_full_design:4400000
/sys/class/power_supply/BAT0/charge_now:4136000
/sys/class/power_supply/BAT0/current_now:0
/sys/class/power_supply/BAT0/cycle_count:0
/sys/class/power_supply/BAT0/manufacturer:SANYO
/sys/class/power_supply/BAT0/model_name:92P1137
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/serial_number: 1523
/sys/class/power_supply/BAT0/status:Not charging
/sys/class/power_supply/BAT0/technology:Li-ion
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/voltage_min_design:10800000
/sys/class/power_supply/BAT0/voltage_now:10991000
```

X270:
Using only the secondary battery isn't uncommon for me - the remaining capacity of an aging internal lithium polymer battery (specified at mere 23 Whrs) isn't very useful compared to the up to 72 Whrs of the secondary battery (using cylindrical lithium-ion cells). I tend to avoid internal batteries whenever possible (exception: hot swapping secondary batteries on travels) so I mostly removed these from my devices. Lenovo Platform Specs including possible battery configurations [here](https://psref.lenovo.com/syspool/Sys/PDF/ThinkPad/ThinkPad_X270/ThinkPad_X270_Spec.PDF).

4goettma, 2025-03-16:
What's about including the git commit id in the output of tlp-stat, e.g. `--- TLP 1.9.0-alpha.0 b1b894b ---` instead of `--- TLP 1.9.0-alpha.0 ---` to avoid confusion?

> I would be even happier if you could help me with tests for the releases on a regular basis :-).

I've been thinking about automating the testing procedure so I can run these tests much more efficiently. I could imagine running these at least on some of my devices every few weeks or months again.

Currently I'm building tlp-git / tlp-rdw-git on each device from scratch (AUR / Git repo). Do you think there's any benefit in doing so? Otherwise (at least for x86_64), I would bundle a precompiled package with the automation script.

linrunner, 2025-03-16:
> Why do you think version 1.8 was still being used? I can't find the version number 1.8 in the posted outputs.

Because the error has reappeared. I already had strange test results because the user (however) managed to mix two versions during installation. 

Of course, this does not apply in your case, because the cause of the error is now different:

> /sys/class/power_supply/BAT0/**charge_full**:4143000
> /sys/class/power_supply/BAT0/**charge_full_design**:4400000
> /sys/class/power_supply/BAT0/**charge_now**:4136000

These **charge_** sysfiles for mA(h) units are untypical for a ThinkPad. Normally there are only **energy_** for mW(h) units. Does the X201 perhaps have coreboot?

X201 is now fixed.

linrunner, 2025-03-16:
X270: My point was not that your case is not relevant. It's just inconvenient to implement with the available primitive means (clitest) and I was wondering if I really need the test coverage for it. I usually test **charge-thresholds_thinkpad-BAT1** with an X260 that has both batteries.

I have now removed all references to BAT0 from the test. Try again.

linrunner, 2025-03-16:
> What's about including the git commit id in the output of tlp-stat, e.g. `--- TLP 1.9.0-alpha.0 b1b894b ---` instead of `--- TLP 1.9.0-alpha.0 ---` to avoid confusion?
The AUR packages would have to be adapted for this. I will try this out for my local package building workflow first and if it works, I will talk to the AUR maintainer.

> I've been thinking about automating the testing procedure so I can run these tests much more efficiently. **I could imagine running these at least on some of my devices every few weeks or months again.**

> 
> Currently I'm building tlp-git / tlp-rdw-git on each device from scratch (AUR / Git repo). Do you think there's any benefit in doing so? Otherwise (at least for x86_64), I would bundle a precompiled package with the automation script.

4goettma, 2025-03-18:
> Does the X201 perhaps have coreboot?

I don't think so, it looks and behaves like the stock bios from Lenovo. But I didn't perform any updates on these devices so far, so the BIOS release date from the X201 is October 2010 while the X200s is using a version from December 2011. Embedded controller software is bundled with the BIOS so the X200s might already got fixes it's successor didn't. Also keep in mind that I'm using not a genuine battery for these devices, thought it's exactly the same battery I've used with both (I only own this one).
X200s with BIOS 3.21 / EC 1.06, X201 with BIOS 1.31 / EC 1.12

> X201 is now fixed.

`OK: 67 of 67 tests passed` in add68d7c76182233bf32be5102bc7c1b5eabb5d4

> X270 [...] I have now removed all references to BAT0 from the test. Try again.

`OK: 50 of 50 tests passed` in add68d7c76182233bf32be5102bc7c1b5eabb5d4

linrunner, 2025-03-23:
@4goettma Thanks. 

If you like, you can also run **test-cpufreq.sh** and **test-gpufreq.sh** on a sample of your machines. I would be particularly interested in older models.

4goettma, 2025-03-28:
ThinkPad X60s (without battery, Core2Duo using 32 bit architecture):
`test-cpufreq.sh` ("FAIL: 3 of 7 tests failed"): https://gist.github.com/4goettma/df1bb3532e78e917e1e568096b360578
`test-gpufreq.sh` ("OK: 1 of 1 tests passed"): https://gist.github.com/4goettma/dcdef5e745cbf4bd92fbcca7f01ad8e2

linrunner, 2025-03-28:
There are some special cases that have not yet been caught, so I will improve that. Determining suitable CPU frequency limit values  for the test remains a challenge. 

Thanks for the input!

Philip-Wittamore, 2025-04-08:
On my Thinkpad X220
tlp recalibrate BAT0 
Error: cannot read charge level of battery BAT0

Replacing the default packages with the git versions solved it.
yay -S tlp-git tlp-rdw-git
Thanks!

4goettma, 2025-04-08:
@Philip-Wittamore Thank you for the confirmation!

> Replacing the default packages with the git versions solved it. yay -S tlp-git tlp-rdw-git

This is to be expected, as the version with the suffix “-git” already contains the fix 400e4c63b7f4519c3337f73cb2548ab88df235aa, which is to be released with [Release 1.9](https://github.com/linrunner/TLP/milestone/18). So with 1.9 we will be able to use the package from the official extra repository on Arch Linux again.

n0zmu, 2025-05-21:
I had the same problem on my t420s. I had to downgrade to version 1.7 to recalibrate my new battery...

n0zmu, 2025-05-24:
Thanks for all this work.