linrunner/TLP #794: usb autosupend corrupts serial IO, denylist ineffective

UltraBlackLinux, 2025-03-13:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

I own a flea-scope (https://github.com/rtestardi/usbte), and I've been having an issue with a weird bug: If USB autosuspend is enabled, the exposed serial port has weirdly broken data transfer. If I connect to the serial port via e.g. `minicom`  and press enter or ctrl+c I can see the activity LED of my scope flashing, meaning, that it's receiving data. The computer however doesn't read any data returning from the device unless I throw minicom at the serial interface the moment it comes up. 

I have now figured out that this is due to USB autosuspend, but even with the scope in the denylist I'm still facing the same problem. Turning usb autosuspend off entirely solves this however. Since I don't have any battery power to waste I'd like to keep that enabled however.

I don't know if this is just a weird edge case but it'd probably make sense to add an option to disable suspending of all serial devices altogether since not every serial device is connected to immediately using a udev rule or similar.

**Expected behavior**

The denylist should have an effect.

**To Reproduce**

Enable usb autosuspend. Plug in flea-scope, try to connect to it.

**Additional context**

tlp-stat: https://gist.github.com/UltraBlackLinux/3ea9ce37b97b4c24543a89b82ac0ee5b

linrunner, 2025-03-13:
There's a typo in your config:

> /etc/tlp.conf L0419: SB_DENYLIST="04d8:e66e"

Correct would be:

`USB_DENYLIST="04d8:e66e"`

UltraBlackLinux, 2025-03-13:
oh that is a good call, thank you! No idea when that went missing.

So do I close this or keep this open because of that serial issue?

linrunner, 2025-03-14:
An "exlude serial" feature would require significantly more affected users to report here. I would also need information on how to properly recognize “serial" devices.

Let's start with the output from your device:

    lsusb -vv -d 04d8:e66e

UltraBlackLinux, 2025-03-14:
I did just for testing also whip up an arduino serial console that just echos what you enter into it and got no problems of this sort, which I find quite weird. Maybe this is a one-off problem that just affects this device but I have no idea what would set this device apart from others... The most that I can think of is that the developer behind the device has created his own homebrew serial library from the looks of it...

I've attached the arduino's usb info as well.

<details>
<summary>Flea-Scope</summary>

```
Bus 003 Device 009: ID 04d8:e66e Microchip Technology, Inc. Flea-Scope
Couldn't open device, some information will be missing
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.01
  bDeviceClass            2 Communications
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x04d8 Microchip Technology, Inc.
  idProduct          0xe66e Flea-Scope
  bcdDevice            2.27
  iManufacturer           1 github.com/rtestardi/usbte
  iProduct                2 Flea-Scope
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x00
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
[... 97 lines cut ...]
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
````
</details>

linrunner, 2026-10-08:
@UltraBlackLinux @quite USB serial adapters, e.g. those that provide a **ttyUSB*** interface, are now excluded by default. The implementation is available in the main branch. Get packages from [here](https://download.linrunner.de/packages/).

Enable the trace in the configuration: `TLP_DEBUG="usb"`

Please test by plugging in the device and showing the output of

     sudo tlp-stat -T

@quite with your Tillitis device, it's possible that it is recognized as a mobile data modem (wwan); unfortunately, these cannot be excluded from autosuspend without explicit configuration.

UltraBlackLinux, 2026-10-08:
@linrunner 
```
∑  sudo tlp-stat -T
--- TLP 1.11.0-alpha.1_a4ea9ce --------------------------------------------

Okt 08 18:22:47.736866 forest tlp[1150]: Applying power save settings...done.
Okt 08 18:22:47.814911 forest tlp[1150]: Setting battery charge thresholds...done.
Okt 08 18:22:47.861827 forest tlp[1150]: Enabling radios: wifi wwan.
```
(Am I supposed to be seeing more? I have enabled the trace. Even restarted the systemd service to be sure.)

Unfortunately I can't quite test the `ttyUSB` exclude functionality here since the device that I created this issue for uses sits at `ttyACM`. It seems unaffected. I can still reproduce the initial problem as of now.

linrunner, 2026-10-08:
@UltraBlackLinux ttyACM will work too, the match is **tty*** really.

Which distribution? How did you install the main branch?

The trace does not appear to be configured. Please show `tlp-stat --cdiff`

UltraBlackLinux, 2026-10-08:
@linrunner 
> Which distribution? How did you install the main branch?

Arch linux. Therefore went ahead and obtained the -git package. According to the commit hash I am on the correct version.

cdiff shows
```
+++ Configured Settings (only differences to defaults):
/etc/tlp.conf L0643: TLP_DEBUG="usb"
```
I did also update the config just in case.

Edit: Seems like I missed *something*? Restarting the tlp service wasn't enough, apparently. Just re-ran the command after a reboot and I'm seeing debug info:
```
Okt 08 19:58:23.260121 forest tlp[651]: usb_suspend_device.usb.auto().no_change: Bus 4 Dev 1 ID 1d6b:0002 /sys/devices/pci0000:00/0000:00:1d.0/usb4
Okt 08 19:58:23.260346 forest tlp[652]: usb_suspend_device.usb.auto().no_change: Bus 3 Dev 1 ID 1d6b:0003 /sys/devices/pci0000:00/0000:00:14.0/usb3
Okt 08 19:58:23.278527 forest tlp[650]: usb_suspend_device.usb.auto().no_change: Bus 1 Dev 1 ID 1d6b:0002 /sys/devices/pci0000:00/0000:00:14.0/usb1
Okt 08 19:58:23.286968 forest tlp[665]: usb_suspend_device.usb.auto().no_change: Bus 2 Dev 1 ID 1d6b:0002 /sys/devices/pci0000:00/0000:00:1a.0/usb2
Okt 08 19:58:23.417976 forest tlp[767]: usb_suspend_device.usb.auto().no_change: Bus 2 Dev 2 ID 8087:0024 /sys/devices/pci0000:00/0000:00:1a.0/usb2/2-1
Okt 08 19:58:23.442261 forest tlp[766]: usb_suspend_device.usb.auto().no_change: Bus 4 Dev 2 ID 8087:0024 /sys/devices/pci0000:00/0000:00:1d.0/usb4/4-1
Okt 08 19:58:23.568881 forest tlp[856]: usb_suspend_device.usb.auto().no_change: Bus 2 Dev 3 ID 06cb:009a /sys/devices/pci0000:00/0000:00:1a.0/usb2/2-1/2-1.3
Okt 08 19:58:24.018372 forest tlp[1120]: usb_suspend_device.usb.auto(): Bus 2 Dev 4 ID 0a5c:21e6 /sys/devices/pci0000:00/0000:00:1a.0/usb2/2-1/2-1.4; control: rc=0
Okt 08 19:58:26.992308 forest tlp[1303]: Applying power save settings...done.
Okt 08 19:58:27.018305 forest tlp[1303]: Setting battery charge thresholds...done.
Okt 08 20:02:56.852220 forest tlp[3060]: usb_suspend_device.usb.auto(wwan): Bus 1 Dev 6 ID 04d8:e66e /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1; control: rc=0
```
But I just confirmed -- the problem still persists

linrunner, 2026-10-09:
@UltraBlackLinux Setting `TLP_DEBUG="usb"` in **tlp.conf** takes effect _immediately_ without restarting the tlp.service; in fact, it isn't a daemon at all. 

When you plug in an USB device, a udev event is triggered that calls tlp-usb-udev, which reads the current **tlp.conf** each time and writes the log entry:
> Okt 08 20:02:56.852220 forest tlp[3060]: usb_suspend_device.usb.**auto(**wwan**)**: Bus 1 Dev 6 ID **04d8:e66e** /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1; control: rc=0

I already had a suspicion because of the “ttyACM,” and it has now been confirmed. The Flea-Scope is recognized as a wwan device (mobile data modem), and unfortunately, I can't change that logic. So in your case, the only options are `USB_EXCLUDE_WWAN=1` or `USB_DENYLIST="04d8:e66e"`.

Regarding the trace I can't tell you why the event doesn't show up right away in the journal (`tlp-stat -T` is just a wrapper for `journalctl`). Possibly some journald config. During development on my systems - also Arch Linux -  I get plugin events immediately in the journal with

     journalctl -p debug --output=short-precise --no-pager SYSLOG_IDENTIFIER=tlp SYSLOG_IDENTIFIER=tlp-pd -b 0 -f

I've tested this feature with typical serial adapters based on PL2303 and CH340 chips. It works fine. Unfortunately, your device is a special case that requires explicit configuration.

Still, thank you very much for testing (EDIT: and for the initial suggestion of course)!

UltraBlackLinux, 2026-10-09:
@linrunner 

Ah that's unfortunate. Thanks for trying your best to get to the bottom of this regardless :)

Close the issue then?

linrunner, 2026-10-09:
Not yet please.

linrunner, 2026-10-23:
The 1.11 beta test has been **[launched](https://github.com/linrunner/TLP/issues/909)**. This feature is included in the beta. Have fun and find bugs!

closed 2026-10-23 by linrunner