linrunner/TLP #796: Charging status randomly flips between Not Charging/Discharging

Rudraksh88, 2025-04-10:
Until recently, never really had an issue; perhaps this might not be due to TLP. But here's the thing. I've got this Lenovo Yoga Pro 7 (Ryzen AI 9 365) running Arch. This laptop supports battery threshold and tt now randomly (and quite frequently now) flips between Not charging/Discharging (when connected to power). Trying to diagnose, booted into Windows (dual boot), and there is no such issue. There are no power fluctuations either.

The charging status drops to battery and then randomly again shows that its connected to power. I've tried downgrading kernels and TLP too. Infact, even tried disabling TLP altogether, but its still the same. This has been happening since a week or two and all this while I never noticed any battery drain, when it goes to discharging. 

TLP worked absolutely fine out of the box with no setup. Tried to manually set conservation modes but still its the same. 

Here are some logs of various utilities I tried.

- `tlp-stat -b`
```
--- TLP 1.7.0 --------------------------------------------

+++ Battery Care
Plugin: lenovo
Supported features: charge threshold
Driver usage:
* vendor (ideapad_laptop) = active (charge threshold)
Parameter value range:
* STOP_CHARGE_THRESH_BAT0: 0(off), 1(on) -- battery conservation mode

/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode = 1 (on)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = L22M4PF4
/sys/class/power_supply/BAT0/cycle_count                    =     11
/sys/class/power_supply/BAT0/energy_full_design             =  73000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  75810 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  58320 [mWh]
/sys/class/power_supply/BAT0/power_now                      =    752 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

Charge                                                      =   76.9 [%]
Capacity                                                    =  103.8 [%]
```

- `tlp-stat -c`
```
--- TLP 1.7.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0006: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_MSG_COLORS="91 93 1 92"
defaults.conf L0009: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0010: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0011: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0013: MAX_LOST_WORK_SECS_ON_BAT="60"
/etc/tlp.conf L0579: CPU_ENERGY_PERF_POLICY_ON_AC="performance"
/etc/tlp.conf L0580: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_performance"
defaults.conf L0016: NMI_WATCHDOG="0"
defaults.conf L0017: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0018: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0019: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0020: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0021: DISK_IOSCHED="keep keep"
defaults.conf L0022: SATA_LINKPWR_ON_AC="med_power_with_dipm"
defaults.conf L0023: SATA_LINKPWR_ON_BAT="med_power_with_dipm"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0025: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0026: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0027: PCIE_ASPM_ON_AC="default"
defaults.conf L0028: PCIE_ASPM_ON_BAT="default"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0030: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0032: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0033: AMDGPU_ABM_LEVEL_ON_AC="0"
defaults.conf L0034: AMDGPU_ABM_LEVEL_ON_BAT="1"
defaults.conf L0035: WIFI_PWR_ON_AC="off"
defaults.conf L0036: WIFI_PWR_ON_BAT="on"
defaults.conf L0037: WOL_DISABLE="Y"
defaults.conf L0038: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0039: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0040: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0041: BAY_POWEROFF_ON_AC="0"
defaults.conf L0042: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0043: BAY_DEVICE="sr0"
defaults.conf L0044: RUNTIME_PM_ON_AC="on"
defaults.conf L0045: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0046: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon xhci_hcd"
defaults.conf L0047: USB_AUTOSUSPEND="1"
defaults.conf L0048: USB_EXCLUDE_AUDIO="1"
defaults.conf L0049: USB_EXCLUDE_BTUSB="0"
defaults.conf L0050: USB_EXCLUDE_PHONE="0"
defaults.conf L0051: USB_EXCLUDE_PRINTER="1"
defaults.conf L0052: USB_EXCLUDE_WWAN="0"
defaults.conf L0053: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0054: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0055: NATACPI_ENABLE="1"
defaults.conf L0056: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0576: PLATFORM_PROFILE_ON_AC="performance"
/etc/tlp.conf L0577: PLATFORM_PROFILE_ON_BAT="balanced"
/etc/tlp.conf L0582: CPU_BOOST_ON_AC="1"
/etc/tlp.conf L0583: CPU_BOOST_ON_BAT="0"
```

- Output of `cat /sys/class/power_supply/BAT0/status`
[power_supply.txt](https://github.com/user-attachments/files/19684543/power_supply.txt)

- Output of `journalctl -b | grep -i "acpi\|power\|battery"`
[journalctl.log](https://github.com/user-attachments/files/19684478/journalctl.log)

- Output of `upower -i /org/freedesktop/UPower/devices/battery_BAT0`
[upower.txt](https://github.com/user-attachments/files/19684541/upower.txt)

linrunner, 2025-04-11:
TLP has nothing to do with the charging behavior you described, because the **hardware controls the charging process**. TLP just writes the threshold to the EC (embedded controller). It is activated as your output shows:

> /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/**conservation_mode = 1 (on)**

You can remove the threshold with the command

    sudo tlp fullcharge

The problem is not caused by TLP, there is nothing else I can do to help.

Have you tried using a different charger?

Rudraksh88, 2025-04-12:
Thanks for replying. After randomly stumbling upon upower's repo, found out that this is due to a [regression](https://gitlab.freedesktop.org/upower/upower/-/issues/306) in `upower` version 1.90.8. 

This has been fixed in [1.90.9](https://gitlab.freedesktop.org/upower/upower/-/commit/bdbd99b809cde47cfccbd24eea47c3235c043081)

closed 2025-04-12 by Rudraksh88

linrunner, 2025-04-13:
Thanks for your feedback.

If you want to see the uninterpreted battery readings (via kernel) you better look at **/sys/class/power_supply/BAT0/status** (in the `tlp-stat -b `output) instead of upower.

Btw: this is not the first time that upower has been unable to cope with the effects of charging thresholds.