linrunner/TLP #797: Radio Device Wizard no longer automatically toggles WiFi

count0x30, 2025-05-04:
**Describe the bug**
I'm on a ThinkPad T440p running Mint 22.1 Cinnamon.  I had set up an active-backup Ethernet-WiFi bond, so that I don't lose any connections when moving around with the laptop.  I used TLP 1.6 Radio Device Wizard from the default repos to enable/disable WiFi when the Ethernet cable was unplugged/plugged-in.  This worked for months, until I recently did a bunch of updates (probably including network-manager and netplan.io), and now WiFi is no longer automatically toggled.  I also tried TLP 1.8 from the PPA and have the same problem.

However, if I unplug the LAN, then click on the Network Manager icon in the taskbar and disable the Wired connection, Wireless will turn on by itself.  When I plug in the LAN, and manually enable the Wired connection, Wireless turns off. 

The only other thing I'm using TLP for is battery thresholds, and those are still working.

**Expected behavior**
Until this sudden change in behavior, TLP was working exactly as expected since I first set it up.  When I unplugged the LAN, the Wireless turned on automatically and immediately.  When I plugged it back in, Wireless turned off automatically and immediately.

**To Reproduce**
Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? both
2. Actions to reproduce the behaviour: unplug or plug in LAN, WiFi status does not change
4. Shell commands entered and their output *including error message(s)*: no errors
5. **Full output of `tlp-stat` via https://gist.github.com/ for *all*   matching cases of 1** (not as file attachment, no screenshots): [tlp-stat on AC](https://gist.github.com/count0x30/4bea3c5be4412bfc691d9ad8dd4e696f) and [tlp-stat on Battery](https://gist.github.com/count0x30/dc813b64d495d3134f87923c8ab4d4de)

**Additional context**
Problem persists after many restarts.

linrunner, 2025-05-07:
Hi, that's quite an unusual use case. I would never think of switching off WiFi via RDW when using a bond.

The cause probably lies in incompatible changes to NetworkManager or Netplan (no idea why Ubuntu needs this complexity ...).

Please activate the trace via configuration

    TLP_DEBUG="lock nm rf run”

Reproduce the problem and show the output of

    sudo journalctl -g 'tlp|NetworkManager' -b 0

ps. You left out something important in the Reproduce section: how exactly did you configure the bond?

count0x30, 2025-05-08:
>(no idea why Ubuntu needs this complexity ...)

Yeah, I didn't want to break anything, so I set up the bond through the Network Manager GUI.  It was an ordeal, and I was stopped for a while by [this bug](https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/2098093).  When it came time to making LAN/WLAN mutually exclusive, I was _so relieved_ that TLP just did the trick.  For a while, it was a thing of beauty to just unplug the Ethernet cable, take the laptop elsewhere, and maintain ssh connections, downloads, etc.

I activated the trace, restarted, and then used `journalctl --follow` to get the following output:

**unplug LAN**
```
May 07 19:50:30 Mint kernel: e1000e 0000:00:19.0 enp0s25: NIC Link is Down
May 07 19:50:30 Mint kernel: bond0: (slave enp0s25): link status definitely down, disabling slave
May 07 19:50:30 Mint kernel: bond0: now running without any active interface!
```
**plug in LAN**
```
May 07 19:50:55 Mint kernel: e1000e 0000:00:19.0 enp0s25: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
May 07 19:50:55 Mint NetworkManager[1440]: <info>  [1746661855.4772] device (enp0s25): carrier: link connected
May 07 19:50:55 Mint NetworkManager[1440]: <info>  [1746661855.4784] device (bond0): carrier: link connected
May 07 19:50:55 Mint kernel: bond0: (slave enp0s25): link status definitely up, 1000 Mbps full duplex
May 07 19:50:55 Mint kernel: bond0: (slave enp0s25): making interface the new active one
May 07 19:50:55 Mint kernel: bond0: active interface up!
```
That output is complete.  Seems like when it says `link status definitely down, disabling slave`, NM is not _actually_ disabling the wired connection, and TLP is never activated to do its thing.

When I manually disable/enable the wired connection, TLP still works as expected:

**[unplug LAN, manually disable Wired Connection](https://gist.github.com/count0x30/07bbd55166c68fc0629b2b7aa56a757c)**

**[plug in LAN, manually enable Wired Connection](https://gist.github.com/count0x30/9d36443bb80170049d57edaf12e85255)**

Probably, a new version of NM or Netplan or something else broke the disabling of the wired connection?  I hope I'm wrong, and that after so much time spent on this, the bond can be fixed with a tweak or two.   If it would help, I can post the bond's auto-generated .yaml Netplan file.  Thanks for taking a look at this.

linrunner, 2025-05-08:
> NM is not actually disabling the wired connection, and TLP is never activated to do its thing.

So this is the crux of the matter. RDW places a hook in **/usr/lib/NetworkManager/dispatcher.d/** in order to be called upon interface up or down. If NM never disables (enables) the wired connection, the hook is never called and RDW is never engaged. Works as designed.

The other possibilty is that **NetworkManager-dispatcher.service** isn't enabled on your system.

count0x30, 2025-05-11:
RDW is correctly engaged when I manually disable the wired connection.  So, I guess the next step is to test with standard non-bonded Ethernet.  Any suggestions on where I should report this?  Is this definitely an issue with Network Manager?  Or could it be the kernel itself, because there were also several kernel updates since this was working?

linrunner, 2025-05-11:
The next logical step would be an issue against NetworkManager as to why the hook is not called in the bonding situation. After all, it worked for you until some time ago.

count0x30, 2025-05-11:
I'll do that.  And for the maybe 1 other person on planet Earth with this issue: it's only for bonded interfaces.  Regular wired connections are still disabled when the cable is unplugged, and TLP is invoked.

Thanks for taking the time.  Much appreciated.

closed 2025-05-11 by count0x30

linrunner, 2025-05-12:
Please keep me posted and share the link to the NM issue.

count0x30, 2025-05-13:
Submitted [Bug #2110517](https://bugs.launchpad.net/ubuntu/+bug/2110517).